fairecasoimeme/Zlinky_TIC #84: Problème de portée du Zlinky TIC, de fait produit inutilisable !!!

domB54, 2022-08-29:
Bonsoir à tous,
Je reprends ce fil après 3 tentatives de connexion avec le Zlinky ... Pour intégrer la clé au réseau Zigbee je suis obligé de déplacer et connecter ma box Eedomus dehors à coté de mon Linky, il y a environ 10 m de l'emplacement normal (+ le mur en parpaing et l'isolation de la maison et le muret de la niche où se trouve le Linky), je fais l'opération d'appairage Zigbee avec mon Ipad en Wifi (que j'ai encore lui à cette distance).
A proximité ça semble bien se passer, ensuite je vais dans mon garage porte ouverte avec l'Ipad (à 8/9 m du compteur et là j'arrive à paramétrer les devices proposés dans la longue liste de la config. et, après modif du polling à 2 mn, je vois bien les valeurs des devices sélectionnés s'incrémenter ... Super :) !!
**Mais NON malheureusement**, quand je rebranche la box Eedomus dans le bureau (maison sur s/sol donc un peu surélevée) toujours à environ 10 m du Linky mais avec l'épaisseur des murs en plus les valeurs sont totalement figées et malheureusement ça ne se rafraichit plus :x Mais alors plus du tout ... Un vrai bout de bois à 50€ :( 
Je suis très déçu car je voulais pouvoir utiliser ces infos dans la Domotique bien sûr, surtout depuis que j'ai fait installer des panneaux photovoltaïques sur le toit, j'avais d'ailleurs fait modifier le mode TIC du Linky en STANDARD...
De votre coté avez-vous déjà rencontré aussi ce pb ??
Vraiment dommage que la portée soit si faible cela restreint vraiment l'utilisation du produit car on a pas tous le Linky au garage ou dans la maison ! Ca sert à quoi de développer un produit comme celui là et d'être obligé de le renvoyer? Pourquoi ne pas avoir intégrer directement une petite antenne pour augmenter sa portée ?? D'autant que j'ai des détecteurs d'ouverture AQARA, donc Zigbee aussi, derrière cette fois coté abri de jardin qui est au moins au double de distance et qui fonctionne très bien (pour environ 10 € voir moins) même si j'ai dû faire l'association à coter de la box comme souvent ...
**Très déçu !!** Vraiment dommage, je vais donc le retourner chez Domotique Store qui ont été très coopératifs pourtant !
Merci à vous tous.

SylvainPer, 2022-08-29:
Bonjour,
LQI de 80 à 9m en intérieur en direct sur une zzh avec antenne.
Prenez contact par mail avec le revendeur.
Le produit peut-être défectueux ou votre réseau mauvais/perturbé de ce côté là et vous auriez dû choisir le modèle avec antenne extérieur lors de votre achat.
Il y a eu d'autres sujets ouverts auparavant sur le même sujet.

domB54, 2022-08-29:
Bonsoir,
Merci pour votre prompt retour,
Je suis déjà en contact avec Domotique Store depuis qq jours et ils sont
très réactifs sans avoir malheureusement de vraie solution!
Bien sûr que j'aurai aimé choisir le modèle avec antenne ... *Mais où le
trouver !?? une idée de votre côté ?*
Oui j'ai déjà beaucoup lu sur le sujet mais *rien* ne m'a encore vraiment
aidé ...

Merci pour votre écoute.
Bien cordialement.

SylvainPer, 2022-08-29:
Boutique officielle : https://lixee.fr/produits/37-zigate-usb-ttl-3770014375148.html
Sujet similaire : #81 
pour le mail : contact (at) lixee.fr

domB54, 2022-08-29:
Re-merci pour votre support,
Mais le lien que vous avez joint renvoie vers un ZLinky SANS antenne visiblement et je le connaissais, mais c’était donc le même produit …
Aussi déjà vu le post #81, le problème est le même oui mais il n’y a pas vraiment de solution!?
Bien cordialement.
---
Dominique Bouet

domB54, 2022-08-29:
Oups pardon, pas vu que l’antenne était une option (en petit en bas)!!!
Désolé !

---
Dominique Bouet

SylvainPer, 2022-08-29:
Dans le lien vers le produit, on peut sélectionner antenne juste au dessus
du prix.
Comme vous avez déjà le Zlinky, je vous laisse avec Fred pour voir comment
faire.

domB54, 2022-08-29:
Bien merci mais qui est Fred. ??

---
Dominique Bouet

SylvainPer, 2022-08-29:
Pardon, Lixee, le fabricant, faites un mail.

domB54, 2022-08-29:
Ok merci très bien.
Je vais donc envoyer un mail à leur service commercial. 
Bonne soirée.
---
Dominique Bouet

fairecasoimeme, 2022-08-30:
Bonjour,

Comme pour tout appareil radio, il est difficile de savoir ce qui ne va pas. La distance, les perturbations / interférences. Sachez aussi, que les appareils chinois (notamment Aqara / Xiaomi) ne répondent pas à la même norme et qu'en général, utilisent plus de puissance pour émettre. Donc impossible à comparer.

Concernant le ZLinky_TIC avec l'antenne externe, il peut résoudre en partie le problème dans le sens où il limitera les perturbations et permettra de faire sortir l'antenne du cache du Linky.

La meilleure solution est de mailler votre réseau zigbee. C'est à dire de multiplier les appareils Zigbee qui ont la fonction de routeur/répéteur (en général les appareils sur secteur). Ce principe qui fait la force du zigbee vous permet non seulement d'étendre votre réseau mais aussi de le sécuriser.

Fred

closed 2022-08-30 by fairecasoimeme

Kuinox, 2022-08-30:
Bonjour,  
Lors de mes tests avec la version sans antenne, j'ai testé avec un adhésif en aluminium, ce qui permet de faire un réflécteur radio en renvoyant le signal des cotés opposé a ma maison, j'ai gagné quelques mètres comme cela, mais ce n'étais pas suffisant pour moi.  

Peut être que cela vous permettra d'atteindre la portée nécessaire pour vous.

domB54, 2022-08-30:
Merci pour ton retour Kuinox mais je crains aussi que ce ne soit pas
suffisant le Linky étant dans une niche EDF dos au mur et la maison
derrière!
Bien cordialement.

Kuinox, 2022-08-30:
Je suis a peu près dans le meme cas, sauf que ma maison est a 25 mètres et mur de vielle maison de 1mètre d'épaisseurs.  

Vus la distance tu es probablement au bord de la portée peut qu'etre qu'un bout d'aluminium permettra que ca marche :p.

Il ne suffit pas de grand chose pour que la qualité du signal change, ici l'antenne a été déplacé de 20 cm:  
[image]

domB54, 2022-08-30:
J’ai commandé la version avec antenne mais je note également ta suggestion si ça peut encore améliorer pourquoi pas !!😊
Merci

---
Dominique Bouet