fairecasoimeme/Zlinky_TIC #85: Limitiation of 20 Configur Reporting

pipiche38, 2022-08-30:
In Standard Mono they are about 22 Attributes which can be reportable
In Standard Tri Prod they are about 39 attributes reportable

Can we have this limit set to 40, in order to accomodate the correct setup in reporting mode fo 39 attributes

fairecasoimeme, 2022-08-30:
I have already increase to 50 with the v6 version. It should be available soon

fairecasoimeme, 2022-08-30:
v6 reduces the role of router to 1 route, routes are initiated by routers and coordinators. it doesn't prevent the ZLinky from being used on multiple routes, it just reduces the fact that ZLinky will only register one route in its table.
do not confuse being routed with routing

pipiche38, 2022-09-12:
Which means that you don't allow the ZLinky to have several routes toward the coordinator. This is no sense, and against the zigbee mesh. In case the route break, the ZLinky is dead for a while and will consume a lot of energy to retreive a new route.

fairecasoimeme, 2022-09-12:
I use the same config than an end device for the network layer config. I don't know if you have some informations about this but I believe that end device don't receive route informations (including many to one route requests). This is the parent (routers ou coordinator) which manage routing path. An end device can only send packet to its parent. If an end device must send a broadcast or a unicast transmission to other devices in the network, it sends the message directly to its parent and the parent performs any necessary route or address discoveries to route the packet to the final destination.

You are comparing the energy used to send one paquet( to find a new accessible route) when there is a route break with a permanent routing trafic due to other devices ?

pipiche38, 2022-09-12:
Simply configure this device as an End Device if you don't want it to route and use the Pooling feature to handle commands. 
Configuring a router which is not managing any Routing Table doesn't make sense

fairecasoimeme, 2022-09-12:
No because the linky acquisition, you can't use a sleepy device. This is not the only router type device which can't route. 
And there will be a routing version.
And there is no link with the issue

closed 2022-09-12 by fairecasoimeme