fairecasoimeme/Zlinky_TIC #87: Led clignote de temps en temps.

WarC0zes, 2022-09-01:
Bonjour,
j'ai mis a jour vers le firmware 6 et je viens de remarquer que j'ai la led bleu clignote 3 a 5 fois de temps en temps ( 5-10 min) et reviens fixe. Avec le firmaware 5 le led bleu rester fixe.  C'est une nouveauté du firmware 6 ou un bug du Zlinky?

j'ai débrancher le Zlinky et rebrancher sur le linky. 
Mon model a le condensateur ( acheter en avril) et est utiliser avec ZHA et un clef sonoff zigbee 3.0.

fairecasoimeme, 2022-09-01:
Quand la Led clignote, c'est que le ZLinky a eu une erreur dans le process de décodage du Linky (caractère perdu ou autre) à un moment donnée.
Ce n'est pas une nouveauté du firmware ni un bug.
Ceci peut-être dû à plusieurs choses.
- sollicitation spontanée trop importante entrainant un timeout dans l'acquisition
- Erreur transmission Linky (plutôt rare)
Dans tous les cas, ce n'est pas critique (pas de perte de données ou valeurs)

Cependant, si ça arrive très souvent, essayez de revenir au firmware v5 pour voir si ça se reproduit. Dans ce cas, il y a un biais inconnu à ce jour. Niveau firmware, il n'y a pas eu de modification sur l'acquisition du Linky.

WarC0zes, 2022-09-01:
Ok, je vais repasser sur le firmware v5 et voir.
Comme vous en parler, depuis le début que j'ai le Zlinky (en v5) je reçois des erreurs du genre:

```
2022-08-22 04:09:35.342 WARNING (MainThread) [homeassistant.helpers.entity] Update of sensor.lixee_zlinky_tic_cf99d205_electrical_measurement is taking over 10 seconds
2022-08-22 04:09:37.727 ERROR (MainThread) [homeassistant] Error doing job: Task exception was never retrieved
Traceback (most recent call last):
File "/usr/local/lib/python3.10/site-packages/zigpy_znp/api.py", line 984, in request
response = await response_future
asyncio.exceptions.CancelledError
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
File "/usr/local/lib/python3.10/site-packages/zigpy_znp/api.py", line 1023, in request_callback_rsp
await self.request(request, **response_params)
File "/usr/local/lib/python3.10/site-packages/zigpy_znp/api.py", line 982, in request
async with async_timeout.timeout(self._znp_config[conf.CONF_SREQ_TIMEOUT]):
File "/usr/local/lib/python3.10/site-packages/async_timeout/__init__.py", line 129, in __aexit__
self._do_exit(exc_type)
File "/usr/local/lib/python3.10/site-packages/async_timeout/__init__.py", line 212, in _do_exit
raise asyncio.TimeoutError
asyncio.exceptions.TimeoutError
```
ou

```
2022-08-22 18:40:08.220 ERROR (MainThread) [homeassistant.components.automation.lecture_zlinky_tic_hc_hp_every_minute] Read ZLinky_TIC HC + HP every minute: Error executing script. Unexpected error for call_service at pos 2:
Traceback (most recent call last):
File "/usr/local/lib/python3.10/site-packages/zigpy_znp/zigbee/application.py", line 1046, in _send_request_raw
response = await asyncio.shield(
asyncio.exceptions.CancelledError
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 447, in _async_step
await getattr(self, handler)()
File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 680, in _async_call_service_step
await service_task
File "/usr/src/homeassistant/homeassistant/core.py", line 1713, in async_call
task.result()
File "/usr/src/homeassistant/homeassistant/core.py", line 1750, in _execute_service
await cast(Callable[[ServiceCall], Awaitable[None]], handler.job.target)(
File "/config/custom_components/zha_toolkit/__init__.py", line 697, in toolkit_service
raise handler_exception
File "/config/custom_components/zha_toolkit/__init__.py", line 661, in toolkit_service
await handler(
File "/config/custom_components/zha_toolkit/__init__.py", line 746, in command_handler_default
await default.default(
File "/config/custom_components/zha_toolkit/default.py", line 33, in default
await handler(app, listener, ieee, cmd, data, service, params, event_data)
File "/config/custom_components/zha_toolkit/zcl_attr.py", line 273, in attr_read
await attr_write(*args, **kwargs)
File "/config/custom_components/zha_toolkit/zcl_attr.py", line 315, in attr_write
result_read = await u.cluster_read_attributes(
File "/config/custom_components/zha_toolkit/utils.py", line 739, in cluster_read_attributes
return await cluster.read_attributes(attrs, manufacturer=manufacturer)
File "/usr/local/lib/python3.10/site-packages/zigpy/zcl/__init__.py", line 482, in read_attributes
result = await self.read_attributes_raw(to_read, manufacturer=manufacturer)
File "/usr/local/lib/python3.10/site-packages/zigpy/quirks/__init__.py", line 233, in read_attributes_raw
results = await super().read_attributes_raw(
File "/usr/local/lib/python3.10/site-packages/zigpy/device.py", line 291, in request
radio_result, msg = await self._application.request(
File "/usr/local/lib/python3.10/site-packages/zigpy_znp/zigbee/application.py", line 302, in request
return await self._send_request(
File "/usr/local/lib/python3.10/site-packages/zigpy_znp/zigbee/application.py", line 1161, in _send_request
response = await self._send_request_raw(
File "/usr/local/lib/python3.10/site-packages/zigpy_znp/zigbee/application.py", line 1043, in _send_request_raw
async with async_timeout.timeout(DATA_CONFIRM_TIMEOUT):
File "/usr/local/lib/python3.10/site-packages/async_timeout/__init__.py", line 129, in __aexit__
self._do_exit(exc_type)
File "/usr/local/lib/python3.10/site-packages/async_timeout/__init__.py", line 212, in _do_exit
raise asyncio.TimeoutError
asyncio.exceptions.TimeoutError
```

Des problèmes de timeout comme vous me dite. Je sais pas d'ou vient ce problème.

fairecasoimeme, 2022-09-01:
Avez-vous vu l'errata : https://github.com/fairecasoimeme/Zlinky_TIC#errata-hardware
êtes-vous concerné ?

WarC0zes, 2022-09-01:
Non, j'ai le condensateur.

fairecasoimeme, 2022-09-01:
> Excusez mon ignorence, mais comment remettre la v5 ? Ca va pas me bloquer si je suis en v6 ?

je pense que la mise à jour par OTA va poser problème vu que c'est du downgrade. Pouvez-vous effectuer la mise à jour en suivant : https://github.com/fairecasoimeme/Zlinky_TIC#mise-%C3%A0-jour-du-firmware-non-ota ?

fairecasoimeme, 2022-09-01:
Vous avez globalement combien de timeout par jour ?

WarC0zes, 2022-09-01:
C'est aléatoire de 3 a 8 par jours , si c'est pas plus.
Pour la maj non ota, ca va être un peu compliquer pour moi. c'est pas trop de mon niveau et j'ai pas de module USB TTL.

fairecasoimeme, 2022-09-01:
ok je vais voir pour générer un OTA permettant de downgrader et je le posterai ici dés que réaliser.

WarC0zes, 2022-09-01:
D'accord, merci beaucoup.

fairecasoimeme, 2022-09-01:
[ZLinky_GpProxy_Ota_OM15081_V5.zip](https://github.com/fairecasoimeme/Zlinky_TIC/files/9471487/ZLinky_GpProxy_Ota_OM15081_V5.zip)
Normalement, cette version devrait faire l'affaire

WarC0zes, 2022-09-01:
ok, je test le downgrade.

fairecasoimeme, 2022-09-01:
[ZLinky_router_v5.0_OTA_v7.zip](https://github.com/fairecasoimeme/Zlinky_TIC/files/9471506/ZLinky_router_v5.0_OTA_v7.zip)
cette version devrait être ok

WarC0zes, 2022-09-01:
ok, ouf.

WarC0zes, 2022-09-01:
C'est bon ca downgrade 
```
2022-09-01 18:13:38.212 DEBUG (MainThread) [zigpy.zcl] [0xF1C9:1:0x0019] OTA upgrade progress: 1.7
```
je te tient au courant.

WarC0zes, 2022-09-01:
le downgrade en v5 est ok.
Depuis 1h la led n'a pas clignoté, elle reste fixe. Ca a l'air bon, mais je vais tester sur 24h voir.
merci beaucoup pour ton aide.

WarC0zes, 2022-09-02:
Bonjour,
la led bleu est rester fixe depuis le downgrade en v5. J'ai vu deux fois clignoté la led depuis le downgrade, rien d'alarmant a mon avis. Les infos remontent bien. 
j'ai les erreurs dans les logs comme avant ( celle du post au dessus). Je vais activer le debug pour zigpy_znp pour en savoir un peu plus.

Encore merci pour tous.

closed 2022-09-02 by WarC0zes