fairecasoimeme/Zlinky_TIC #89: Ne capte plus depuis mise à jour

antibill51, 2022-09-02:
Bonjour. Depuis la mise à jour, aucune remonté d'informations.
Impossible de ré apairer également, pourtant d'autres périphériques placés à côté fonctionnent très bien.

J'utilise z2m, y'a t'il possibilité de faire un downgrade en ota ? Histoire d'être certain que le problème vienne de là. 
En déplaçant provisoirement le coordinateur et en supprimant le routeur cc2531, je peux réapairer. 

Merci

fairecasoimeme, 2022-09-02:
Bonjour,
Normalement aucun rapport mais vous pouvez utilisez le fichier qui est sur le lien suivant : 
https://github.com/fairecasoimeme/Zlinky_TIC/issues/87#issuecomment-1234488515
tenez moi au courant.
Fred

ach-78, 2022-09-02:
Bonjour,
Soucis de remontée d'info intermittente depuis la v6. J'ai bien un condo, je n'ai ce soucis que depuis la montée en v6. Ça va ça vient, avec de longue périodes sans remontée (plusieurs heures) entrecoupées de périodes où j'ai mon sensor.
En ZHA et Clé Zigbee Sonoff 3.0.
Je suis en train de rollbacker en v5 (bien qu'avant d'avoir ces soucis j'étais en v1).

ach-78, 2022-09-02:
A noter : avec le fichier de downgrade en v5 qui se présente en pseudo "v7", une fois la Maj de downgrade réalisée, une nouvelle Maj firmware démarre aussitôt puisqu'il détecte une Maj v5 à v7 nécessaire...

J'ai donc du commenter la partie Maj OTA dans le .yaml puis redémarrer HA alors qu'une Maj était en cours de chargement. En espérant qu'il n'y a aucune conséquence à cette interruption.

WarC0zes, 2022-09-02:
@ach-78 j'ai eu le même soucis. J'ai eu peur de couper le downgrade, donc j'ai attendu la fin du 2eme downgrade et j'ai vite reboot au 100%.

tomj44300, 2022-09-03:
bonjour vous faites comment pour downgrade a zigbee2mqtt , inforlation figé depuis la maj de v5 a v6

ach-78, 2022-09-03:
Il faut faire comme un upgrade mais en utilisant la pseudo v7 (qui est en fait la v5) du lien donné plus haut. Attention l’upgrade va partir en boucle car détecté sans fin une v5 actuelle et une v7 proposée.

Depuis hier soir que j’ai viré la v6 tout semble revenu en ordre.

tomj44300, 2022-09-03:
j ai pas réussi a downgrade , je dois pas avoir compris la procédure https://github.com/Koenkk/zigbee2mqtt.io/blob/develop/docs/guide/usage/ota_updates.md#local-ota-index-and-firmware-files
par contre je doit avoir de la chance le lixee refonctionne après je ne sais cb de debranchage et suppression ajout

zen2, 2022-09-03:
je confirme aussi plus aucune remontée d'information après mise à jour OTA en v6.

L041S, 2022-09-03:
Pareil ici, depuis la V6 plus aucune remontée d’info  :(

L041S, 2022-09-04:
J’avoue n’avoir pas trop compris la procédure pour le downgrade, je n’ai jamais eu à faire cette manip.

git-greatguy, 2022-09-04:
idem ici, j'aimerais bien que l'on communique la procédure de mise a jour pour la pseudo v5.
J'ai tenté de procéder via z2m en poussant en local un fichier json avec la config et le fichier ota, retour : "c'est a jour".
idem en placant cette config sur un serveur web.

Quelque soit la configuration, pour ce device précis, il ne charge pas mon override et l'ota me répond qu'il n'y a pas de mise a jour disponible ?!?

 2022-09-04 16:13:07Received MQTT message on 'zigbee2mqtt/bridge/request/device/ota_update/check' with data ```
'{"id":"Lixee_zlinky","transaction":"pyt27-5"}'
Info 2022-09-04 16:13:07MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Checking if update available for 'Lixee_zlinky'","meta":{"device":"Lixee_zlinky","status":"checking_if_available"},"type":"ota_update"}'
Debug 2022-09-04 16:13:10Received Zigbee message from 'Lixee_zlinky', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":6,"imageType":1,"manufacturerCode":4151}' from endpoint 1 with groupID 0
Debug 2022-09-04 16:13:10Got OTA request '{"fieldControl":0,"manufacturerCode":4151,"imageType":1,"fileVersion":6}'
Info 2022-09-04 16:13:11Using firmware file ZLinky_router_v6.0.ota
Debug 2022-09-04 16:13:11Is new image available for '0x00158d0005ce7980', current '{"fieldControl":0,"manufacturerCode":4151,"imageType":1,"fileVersion":6}', latest meta '{"fileVersion":6,"fileSize":243758,"url":"https://github.com/fairecasoimeme/Zlinky_TIC/releases/download/v6.0/ZLinky_router_v6.0.ota"}'
Info 2022-09-04 16:13:11MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"No update available for 'Lixee_zlinky'","meta":{"device":"Lixee_zlinky","status":"not_available"},"type":"ota_update"}'
Info 2022-09-04 16:13:11MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/check', payload '{"data":{"id":"Lixee_zlinky","updateAvailable":false},"status":"ok","transaction":"pyt27-5"}'

```

Il doit y avoir une subtilité dans le process ota qui n'apparait qu'avec ce device dans mqtt.

Ex si je pousse le bouton "check for update" pour un autre appareil :

```
Debug 2022-09-04 16:18:44ZigbeeOTA: downloaded main index
Debug 2022-09-04 16:18:44ZigbeeOTA: Loading override index https://www.****.eu/index.json
```

On dirait que le device pousse sa propre url pour les mises a jour et ne permet pas d'être outrepassé. j'ai bon ?

git-greatguy, 2022-09-04:
je m'auto-répond en donnant une explication / une méthode potentiel pour downgrade le firmware.

Le système de distribution de l'ota pour le lixee zlinky est dit "propriétaire", il n'utilise pas le dépot "standard" de github pour la distribution de l'ota zibbee. Il faut donc surcharger le "converter" qui est livré avec zigbee2mqtt pour indiquer une mise à jour dite "ota.zigbeeOTA"

Les modifications a faire sont les suivantes :

1 - Recupérer le fichier /app/node_modules/zigbee-herdsman-converters/devices/lixee.js du container zigbee2mqtt

`docker cp addon_45df7312_zigbee2mqtt:/app/node_modules/zigbee-herdsman-converters/devices/lixee.js ./`

note : le nom du conteneur est peut-être a adapté a votre situation.

ensuite éditer le fichier et modifier le ligne suivante
`ota: ota.lixee,`
en
`ota: ota.zigbeeOTA,`

2 - Modifier le chemin des dépendances en début de fichier avec le chemin complet

`/app/node_modules/zigbee-herdsman-converters/lib/`
au lieu de
`../lib`

3 - Déposer le fichier modifié dans le répertoire /conf/zigbee2mqtt du container homeassistant

4 - Procéder à la création d'un fichier /conf/zigbee2mqtt/my_index.json avec le contenu suivant :
```
[
    {
        "manufacturerCode": 4151,
        "imageType": 1,
        "url": "ZLinky_router_v5.0.ota",
        "force": true
    }
]

```

5 - Déposer dans ce même répertoire, le fichier ZLinky_router_v5.0.ota

6 - Ajouter dans le fichier configuration.yaml de zigbee2mqtt la déclaration de votre fichier override
```
ota:
  zigbee_ota_override_index_location: my_index.json
```

7 - Redémarrer zigbee2mqtt et lancer le "check update".

Pour ma part l'update est en cours. il reste une vingtaine de minute. Je mettre a jour le post pour annoncer la réussite ou non  de l'opération.

git-greatguy, 2022-09-04:
Ca a fonctionné nickel !

`MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/update', payload '{"data":{"from":{"date_code":"20220830","software_build_id":"4000-0006"},"id":"Lixee_zlinky","to":{"date_code":"20220217","software_build_id":"4000-0005"}},"status":"ok","transaction":"0httf-1"}'`

L041S, 2022-09-04:
Merci pour ton retour, je commence à y voir un peu plus clair !
De mon côté zigbee2mqtt n’est pas installé via l’addon home assistant mais via docker (étant donné que j’ai home assistant en image docker je n’ai pas le store).
J’ai donc adapté ta commande en tapant
`docker cp zigbee2mqtt:/app/node_modules/zigbee-herdsman-converters/devices/lixee.js ./`

J’ai donc bien le fichier mais une fois ce dernier modifié je ne sais pas où le mettre, j’ai bien un dossier zigbee2mqtt/data/ dans lequel j’ai mon configuration.yaml que j’ai pu éditer pour ajouter ota: ota.zigbeeOTA mais en collant le fichier lixee.js edité + l’ota ZLinky_router_v5.0.ota dans ce dossier, le check update me dit que tout est encore à jour. Je ne suis pas dans le bon répertoire ?

git-greatguy, 2022-09-04:
si tu as bien le fichier de config de z2m, tu as dans le bon repertoire. 
As tu créé le fichier my_index.json avec la config pour lui indiquer l'ota a surcharger ? 
as tu vérifier, une fois ton image docker redémarré, que tes modifs ont bien été conservé ? (assures toi d'etre dans un volume)

L041S, 2022-09-04:
Le my_index.json avec la config que tu as indiqué est ok, il se trouve dans ce même répertoire à côté de l’ota, le configuration.yaml de z2mqtt et le lixee.js  modifié. Au cas ou, la commande que j’utilise pour lancer le container zigbee2mqtt :
`docker run -d    --name zigbee2mqtt    --restart=always    --device=/dev/ttyUSB0    -p 8080:8080    -v /home/user/zigbee2mqtt/data:/app/data    -v /run/udev:/run/udev:ro    -e TZ=Europe/Paris    koenkk/zigbee2mqtt `

git-greatguy, 2022-09-04:
Il faut regarder le log de z2m pour voir ce qu'il fait quand tu lance le check ota. normalement si il prend bien en compte les fichiers, tu doit voir une mention à l'override du dépot main.
Tu peux aussi créer une faute de syntaxe dans le fichier lixee.js pour valider qu'il le charge bien. cela devrait générer une erreur au démarrage.

L041S, 2022-09-04:
Je viens de vérifier les logs, il ne prend tout simplement pas en compte mes modifs et continue de vérifier sur le github. Pourtant le fichier lixee.js est bien là à côté du configuration.yaml, je vais creuser ça demain… Merci pour tes réponses en tous cas !

`2022-09-05 01:08:27 Is new image available for 'XXXX', current '{"fieldControl":0,"manufacturerCode":4151,"imageType":1,"fileVersion":6}', latest meta '{"fileVersion":6,"fileSize":243758,"url":"https://github.com/fairecasoimeme/Zlinky_TIC/releases/download/v6.0/ZLinky_router_v6.0.ota"}'`

git-greatguy, 2022-09-04:
au tamps pour moi... j'ai oublié une ligne de config : 
dans le fichier configuration.yaml de z2m il faut ajouter cette information de surcharge du lixee.js

```
external_converters:
  - lixee.js
```

Je vais updaté mon post précédent... désolé :O

L041S, 2022-09-04:
Victoire, la mise à jour s’effectue !!!
Merci beaucoup !🥳
Je n’ai plus qu’à le laisser terminer son update et ensuite je pourrais tout remettre comme avant au niveau du configuration.yaml ?

Par ailleurs j’ai lu sur les posts du dessus que certains restaient dans une mise à jour en boucle, il y a une manipulation spéciale à faire pour que cela n’arrive pas ?

git-greatguy, 2022-09-04:
il suffit de commenter les deux lignes de config dans le configuration.yaml de z2m pour supprimer le process de rollback forcé. 
Pour ma part, depuis le rollback en v5, il me propose a nouveau d'upgrade en v5, a cause du "force : true" mais il ne boucle pas. Pour supprimer la proposition de mise a jour, il suffit de supprimer et réappairer le device. Il te proposera a nouveau la v6 du dépot officiel si tu fais un check update.

tomj44300, 2022-09-05:
apres avoir essayé de downgrade durant 2h car la v6 ne marchais pas , çà c'est débloqué je ne sais comment et marche nickel depuis 48h non stop, par contre j'ai l'impression que çà ne fait aucun poling çà m'envoie les données dès changement, maximum 10s avant maj auto le plus souvent 2/3s max
[image]

Patoche781, 2022-09-06:
Bonjour. J'ai le même souci depuis le passage de V5 en V6: plus de remontées + impossibilité de se connecter à z2m depuis car ('ai supprimé le device Zlixee de z2m dans l'espoir de le reconnecter.... Dès lors impossible de l'appairer ! Le voyant bleu s'éteint et se rallume en bleu fixe à chaque reset... Comment faire pour le downgrader en V5 puisque je n'ai plus de lien de communication device-z2m ?
Merci pour votre aide.

gael1980, 2022-09-06:
Bonjour,
J'ai réussi à faire tourné la V6 en

1. Supprimer lixee de z2m (option force)
2. éteindre HA
3. déconnecter ma clé combi2 30s (j'ai lu une fois dans un forum que de déconnecter la clé, forcer la ré init)
4. reconnecter la clé et re démarrer HA
5. refaire l'asso du lixee

Depuis, cela fonctionne en v6. 
Le seul PB, est la réception qui est moins bonne, du au fait qu'il ne ce connecte que sur un seul routeur)

git-greatguy, 2022-09-06:
J'ai vu également que la v6 s'est vu sa connexion zigbee limitée en terme de node. le changelog fait référence a la consommation énergétique pour justifier ce changement.
Quel est le gain énergétique réel de cette modification ? Y'a t-il eu des cas ou le lixee charge plus que les 130mw de puissance dispo sur les bornes de la tic ?

fairecasoimeme, 2022-09-06:
Le ZLinky n'est pas censé se connecter sur un seul routeur mais ne gérer qu'une seule route. Il ne faut pas confondre la fonction de router avec être routé. Mais en aucun cas, le ZLinky ne peut appartenir qu'à une seule route. Le coordinateur ou les routeurs qui routent le ZLinky ne sont pas impactés...

De ce que j'ai vu de mes tests, le problème du firmware V6 n'a l'air d'impacter que Z2M. Au bout d'un moment, il n'y a plus de polling et je ne sais pas pourquoi. Concernant les reports, ça à l'air de passer.

Le fait de limiter la capacité de router d'autres devices permet de limiter le trafic radio passant par le ZLinky et donc économiser des transmissions/réceptions radios consommatrices d'énergie. 

Cette version a été développée car certains clients, ayant par exemple des ampoules philips Hue (Très bavardes) connectées au ZLinky ou plutôt routaient par le ZLinky, ont des coupures liées à un manque d'énergie délivré par le Linky.

git-greatguy, 2022-09-06:
Une question, toujours sur le routage via le z-linky : 
Ne peut-on pas gérer, via un paramètre, la possibilité d'activer ou désactiver la fonction de routage pour d'autres appareils ?

Personnellement, j'ai besoin qu'il route un contacteur de porte dans le sous sol. La v6 me faire perdre tout lien avec cet appareil et m'oblige a revoir la configuration de mon réseau, qui passerai sans doute par l'achat / la mise en place d'un nouveau routeur dans le sous sol !

Je comprend très bien maintenant l'origine de cette modification, et elle totalement fondé. Mais il ne faut pas perdre de vue que la fonction de routage peut aussi être un facteur important pour d'autres utilisateurs de la solution ;)

fairecasoimeme, 2022-09-06:
Pas de panique, je vais sortir une version avec routage, j'espère cette semaine ou semaine prochaine.

Patoche781, 2022-09-06:
Le Zlinky est UP!
 Opérations suivantes ont été faites : shutdown HA via Proxmox puis unplug device Sonoff USB Zigbee puis reboot HA . Suite au reboot -> reset Zlinky puis analyse des logs z2m -> payload sur device... Tiens tiens ! Restart z2m -> new device apparu: c'est lui en V6!  Remontées datas -> en cours . Reparamétrage des W précision, etc...   En attente remontées dans tableau "energie".

pdecat, 2022-09-06:
> De ce que j'ai vu de mes tests, le problème du firmware V6 n'a l'air d'impacter que Z2M.

Pour ma part, j'utilise Home Assistant avec ZHA (Zigpy ZNP) et j'ai également dû repasser de v6.0 en v5.0 car au bout de quelques heures après avoir été débranché/rebranché, mon ZLinky (modèle avec antenne et condensateur) clignotait fréquemment, perdait souvent la connectivité ZigBee, et ne répondait plus aux lectures d'attributs ou autres commandes. Aucun problème depuis le downgrade en v5.0 effectué dimanche matin.

Note : pour parvenir à déclencher l'OTA de downgrade, j'ai dû brancher mon ZLinky en USB avec un adaptateur TTL proche de mon coordinateur (mon compteur est à l'extérieur à ~20 mètres de la maison), et le réinitialiser. C'est la  première fois que ça me fait cela.

WarC0zes, 2022-09-07:
> Pour ma part, j'utilise Home Assistant avec ZHA (Zigpy ZNP) et j'ai également dû repasser de v6.0 en v5.0 car au bout de quelques heures après avoir été débranché/rebranché, mon ZLinky (modèle avec [...]

Ca me rassure, étant sous ZHA aussi j'ai eu les même soucis avec la v6 et du faire un downgrade en v5.

Parlendir, 2022-09-07:
N'est il pas possible de :
* pousser une nouvelle image estampillée v7 mais dont les données correspondent à la v5 pour que tout le monde puisse faire la màj ota facilement
* supprimer la v6 du dépôt contenant les mises à jour
* avoir une prochaine v8 correspondant à la correction de la v6

Cela simplifierait grandement la gestion de cette mise à jour qui est une régression pour les utilisateurs z2m.

pdecat, 2022-09-07:
> * pousser une nouvelle image estampillée v7 mais dont les données correspondent à la v5 pour que tout le monde puisse faire la màj ota facilement

C'est ce qui a été fait :
* https://github.com/fairecasoimeme/Zlinky_TIC/issues/89#issuecomment-1235530610
* https://github.com/fairecasoimeme/Zlinky_TIC/issues/87#issuecomment-1234488515

Parlendir, 2022-09-07:
> 

> * pousser une nouvelle image estampillée v7 mais dont les données correspondent à la v5 pour que tout le monde puisse faire la màj ota facilement

Mon z2m (v1.27.2) ne voit pas de mise à jour pour le lixee. Donc ce n'est pas ce que j'ai évoqué : la mise à jour n'est pas le dépôt consulté par défaut par z2m, selon la procédure basique.
Pas de souci, j'attendrai la mise à jour de correction.
Merci au dév, c'est un super produit ! J'ai mis beaucoup trop de temps à découvrir son existence :)

pdecat, 2022-09-07:
La mise à jour pour z2m est là : https://github.com/Koenkk/zigbee-OTA/pull/156

ach-78, 2022-09-07:
> > * pousser une nouvelle image estampillée v7 mais dont les données correspondent à la v5 pour que tout le monde puisse faire la màj ota facilement
> 
> C'est ce qui a été fait :
> 
> * [Ne capte plus depuis mise à jour #89 (comment)](https://github.com/fairecasoimeme/Zlinky_TIC/issues/89#issuecomment-1235530610)
> * [Led clignote de temps en temps. #87 (comment)](https://github.com/fairecasoimeme/Zlinky_TIC/issues/87#issuecomment-1234488515)
> * https://github.com/fairecasoimeme/Zlinky_TIC/files/9471506/ZLinky_router_v5.0_OTA_v7.zip

Ce n'est pas ce qui a été fait, car une fois installé, ce firmware pseudo-v7 apparait comme une v5 en terme de version courante.
C'est comme ça que mon downgrade (ZHA) est parti en boucle : à peine terminé, il redétectait une Maj à faire de v5 en v7. Sauf à modifier le yaml et redémarrer HA, ça peut boucler comme ça indéfinniment.

Je suis d'accord avec Parlendir que ça diminuerait le risque de complication de sortir une v7 (identique en contenu qu'une v5) mais qui reste apparente "v7" une fois installée comme si c'était une vraie v7, le temps qu'il y ait une vraie v8 (qui serait une v6 fixée en fait).

pdecat, 2022-09-07:
Ok, j'avais mal compris ce qui était demandé. Pour ma part avec ZHA, je gère les fichiers dans le dossier `zigpy_ota` à la main donc j'ai juste eu à supprimer les fichiers une fois le downgrade terminé.

antoinelibert, 2022-09-08:
Bonjour,

Pour ma part, je tourne avec la V6 (devenue "expérimentale") depuis sa sortie et pas de problème particulier. J'ai quand même du reset le zlinky après la màj, ré-appairer avec z2m et reconfigurer). Habituellement j'avais un plantage du zlinky plus ou moins toutes les 1-2 semaines. Pour l'instant depuis sa màj, rien à signaler.

fairecasoimeme, 2022-09-09:
Bonjour, 
Avant de mettre en release, est-ce possible pour ceux qui ont eu un problème de tester cette version de firmware qui je l'espère corrige les problèmes que vous avez eu.

[ZLinky_GpProxy_Ota_OM15081_V7.zip](https://github.com/fairecasoimeme/Zlinky_TIC/files/9536627/ZLinky_GpProxy_Ota_OM15081_V7.zip)

domoblog, 2022-09-10:
Bonjour a tous,
V6 fonctionnelle et stable sur z2m après les manipulations suivantes :

1- update vers 1.27.2
2- delete du Lixee
3- restart HA
4- nouvelle association du Lixee
5- reparametrage

Fait il y a 3 jours sans downgrade et plus de décrochage, ni de perte de mesure.

pdecat, 2022-09-10:
Bonjour @fairecasoimeme, upgrade fait à l'instant avec `ZLinky_GpProxy_Ota_OM15081_V7.ota`:

```
2022-09-10 10:56:42.851 DEBUG (MainThread) [zigpy.zcl] [0xB164:1:0x0019] OTA image_block handler for 'LiXee ZLinky_TIC': field_control=FieldControl.0, manufacturer_id=4151, image_type=1, file_version=7, file_offset=243726, max_data_size=48, request_node_addr=Noneblock_request_delay=None
2022-09-10 10:56:42.851 DEBUG (MainThread) [zigpy.zcl] [0xB164:1:0x0019] OTA upgrade progress: 100.0
2022-09-10 10:57:57.726 DEBUG (MainThread) [zigpy.zcl] [0xB164:1:0x0019] Decoded ZCL frame: Ota:upgrade_end(status=<Status.SUCCESS: 0>, manufacturer_code=4151, image_type=1, file_version=7)
2022-09-10 10:57:57.727 DEBUG (MainThread) [zigpy.zcl] [0xB164:1:0x0019] OTA upgrade_end handler for 'LiXee ZLinky_TIC': status=Status.SUCCESS, manufacturer_id=4151, image_type=1, file_version=7
```

Tests en cours...

pdecat, 2022-09-10:
J'ai eu quelques soucis suite à l'upgrade, le ZLinky_TIC ne captait plus comme avec la v6.0.
J'ai dû le rebrancher proche de mon coordinateur alimenté en USB TTL pour réussir à le réinitialiser, puis il s'est mis à répondre aux lecture d'attributs.
Je viens de le rebrancher sur mon compteur Linky et ça semble fonctionner à nouveau normalement.

ach-78, 2022-09-10:
J'ai mis la v7 hier soir, et de ce que je vois ce matin j'ai de nouveau une réception dégradée.

En v5 j'ai aussi régulièrement des "WARNING (MainThread) [homeassistant.helpers.entity] Update of sensor.compteur_linky_electricalmeasurement is taking over 10 seconds", plusieurs fois par jour.

Je doit donc être un peu limite en lien zigbee.
C'est juste que ce qui est plutôt ponctuel en v5 et ne m'empêche pas d'avoir des courbes qui tiennent la route s'aggrave visiblement en v7 où j'ai des graphiques avec une valeur non mise à jour pendant vraiment longtemps (j'ai vu des valeurs figées de plus de 2h plusieurs fois et ce matin cette nuit).

Je suis en ZHA, VM Synology, dongle Sonoff 3.0. HA avait été redémarré suite à la mise à jour v7 (pour enlever les logs DEBUG). Je continue d'observer pour voir s'il y a du changement.

WarC0zes, 2022-09-10:
@ach-78 Je suis comme toi, sous ZHA et dongle sonoff 3.0 et j'ai exactement la même erreur depuis que j'utilise le Zlinky. Je suis en v5, car la v6 fonctionnais pas. Mais sur les 2 versions j'ai eu ces erreurs et c'est 3-4 fois par jour.
```
2022-09-10 14:22:22.480 WARNING (MainThread) [homeassistant.helpers.entity] Update of sensor.lixee_zlinky_tic_electricalmeasurement is taking over 10 seconds
2022-09-10 14:22:27.496 ERROR (MainThread) [homeassistant] Error doing job: Task exception was never retrieved
Traceback (most recent call last):
File "/usr/local/lib/python3.10/site-packages/zigpy_znp/api.py", line 984, in request
response = await response_future
asyncio.exceptions.CancelledError
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
File "/usr/local/lib/python3.10/site-packages/zigpy_znp/api.py", line 1023, in request_callback_rsp
await self.request(request, **response_params)
File "/usr/local/lib/python3.10/site-packages/zigpy_znp/api.py", line 982, in request
async with async_timeout.timeout(self._znp_config[conf.CONF_SREQ_TIMEOUT]):
File "/usr/local/lib/python3.10/site-packages/async_timeout/__init__.py", line 129, in __aexit__
self._do_exit(exc_type)
File "/usr/local/lib/python3.10/site-packages/async_timeout/__init__.py", line 212, in _do_exit
raise asyncio.TimeoutError
asyncio.exceptions.TimeoutError
```
A lire ton message, je vais pas tester la v7.

warlordattack, 2022-09-10:
hello,
du coup il y a une maj OTA prévue pour fixer tout ça ? le mien remonte tout le temps les mêmes valeurs figées
et de temps en temps il clignote 6 fois d'affilée : à quoi ça correspond ?

ach-78, 2022-09-11:
24h plus tard, c'est comme si j'étais revenu à mon taux d'erreurs de "croisière" (i.e équivalent à la v5). J'en ai quelques séries dans les logs mais pas de quoi être visible dans les courbes.
Difficile à comprendre, mais c'est mon constat avec le peu de recul que j'ai. Il semble se passer quelque chose après la Maj qui, potentiellement, rentre dans l'ordre tout seul...

fairecasoimeme, 2022-09-12:
Merci pour tous les retours. J'observe qu'il y a plusieurs sujets dans les retours. En ce qui concerne la mise à jour, J'ai l'impression que le ZLinky n'arrive pas à traiter tout le flux d'informations qu'elle se prend au départ.
Pour ceux qui ont des pertes de données régulières, dans un premier temps, je vous conseille de limiter les demandes d'informations au polling (limiter aux données qui vous intéressent) et de préférer l'utilisation des rapports (config report).
J'ai pu tester à de nombreuses reprises que HA enclenche énormément de read attribute que le ZLinky ne peut pas traiter car il coupe pour faire une acquisition Linky.

De mon côté je vais regarder si je peux optimiser et décaler l'acquisition Linky s'il y a une activité radio, ce qui permettrait de ne pas couper et stopper les timeout. Juste cette modif pourrait corriger tous les problèmes.

En entendant restez sur la V5 ou la version dans laquelle vous êtes. Je vous tiendrai au courant.

warlordattack, 2022-09-12:
finalement, j'ai juste supprimé et reassocié le Zlinky_TIC dans Z2M et j'ai de nouveau les données qui arrivent bien

ach-78, 2022-09-12:
Si quelqu'un a un lien à partager pour configurer le Zlinky de manière à limiter les attributs à remonter, je suis preneur. Je suis un gros noob de HA !

Etant en ZHA et un compteur en mode historique, je n'ai de toute façon que 3 sensor qui marchent (VA, Ampères et kWh compteur) donc j'imagine que je pourrais drastiquement alléger la charge du Zlinky...

Sinon sur les dernières 24h, je n'ai eu plus qu'un seul warning "taking over 10 seconds" donc y'a pas photo, quelque chose cloque après la maj mais semble rentrer dans l'ordre par la suite !

antoinelibert, 2022-09-18:
Bonjour,

Une vingtaine de jours plus tard après màj en v6, le zlinky vient de crash (une amélioration par rapport à ce que j'avais avant (au mieux 2 semaines entre chaque crash).

Firmware build date 20220830
Firmware version 4000-0006
Z2M 1.27.2
ConBee2 (0x26720700)
7 reportings configurés (1 minute de délais minimum par reporting, les logs me donnent ~2 messages / min)
Linky en STANDARD BASE / Pas de production / Simple phase

ipogs, 2022-09-18:
Bonjour,

Chez moi avec la v6, pas de crach, mais valeurs PAPP et IINST sont figées depuis la mise à jour

> HA + Z2M + Clé Zigbee Sonoff
[image]
[image]

WarC0zes, 2022-09-21:
Des retour sur la v7 avec ZHA ?

@fairecasoimeme les nouveautées de la v7 sont sympa surtout le "Add 2 config files. With or without childs routing function", car j'ai l'impression qu'avec des enfants connecter au zlinky c'est pire. Je testerai "without childs routing function" qui je suppose est "[ZLinky_router_v7.0_no_route.ota](https://github.com/fairecasoimeme/Zlinky_TIC/releases/download/v7.0/ZLinky_router_v7.0_no_route.ota)" en OTA ?

fairecasoimeme, 2022-09-21:
@WarC0zes oui dans ton cas, il vaut mieux utiliser "[ZLinky_router_v7.0_no_route.ota](https://github.com/fairecasoimeme/Zlinky_TIC/releases/download/v7.0/ZLinky_router_v7.0_no_route.ota)"

pdecat, 2022-09-21:
Bonjour @fairecasoimeme, est-ce que `ZLinky_GpProxy_Ota_OM15081_V7.ota` est exactement la même chose que https://github.com/fairecasoimeme/Zlinky_TIC/releases/download/v7.0/ZLinky_router_v7.0.ota ? Si ça n'est pas le cas, comment je fais pour l'OTA étant donné la version qui est la même ?

WarC0zes, 2022-09-21:
J'ai maj en v7 no route, ca a l'air stable ( plus stable que la v6 en tout cas). Je verrai dans 24h, je vous tient au courant.

fairecasoimeme, 2022-09-21:
@pdecat la nouvelle version n'est pas la même. Je t'ai envoyé un OTA qui simule une v8 pour que la mise à jour se déclenche
Merci des retours qui me permettront de la passer en release.

WarC0zes, 2022-09-22:
@fairecasoimeme La v7 no route a pas l'air plus stable que la v5. Quand j'ai maj en v7 no route j'avais 6 routeurs sur le réseau zigbee (ZHA), la led bleu a clignoté (lent) quelques fois sur 12h. Ce matin j'ai voulu tester d'ajouter une multiprise, donc je suis passé a 7 routeurs et le Zlinky clignote (lent) un peu plus que avec 6 routeurs (sur la v5 s'était pareille).

J'ai remarquer que sur la v7 no route le Zlynky avait moins de route que les autres routeurs.
[image]

1/ Pouvez vous limitez le nombre de route au Zlinky?
2/ Ou est-il possible de faire un firmware Zlinky enfant, qui communiquerais  qu'avec le coordinateur.

antibill51, 2022-09-22:
J'ai mis à jour hier sur la V7 Routeur, et tout a l'air d'aller depuis.

pdecat, 2022-09-22:
Aucun soucis de mon côté avec l'OTA [ZLinky_router_v7.0.ota](https://github.com/fairecasoimeme/Zlinky_TIC/releases/download/v7.0/ZLinky_router_v7.0.ota).

WarC0zes, 2022-09-22:
@pdecat @antibill51 vous avez combien de routeurs sur votre réseau zigbee avec le Zlinky ?

[71 more replies]