fairecasoimeme/Zlinky_TIC #90: Decoding of STGE

pipiche38, 2022-09-09:
Looks like the decoding of STGE to get 
-Contact sec 
-Organe de coupure 
-État du cache-bornes distributeur

and plenty of other information is not done.

As STGE is more a bitmap type of information why coding it in a String which make the things much more complex to process ?

This should be a 32bits bitmap and if this doesn't exist in zigbee simply an uint32

fairecasoimeme, 2022-09-09:
I only retransmit what the teleinformation protocol provides. For example : 

LTARF : Heure pleine
EAST : 000000491
EASF01 : 000000478
EASF02 : 000000013
EASF03 : 000000000
EASF04 : 000000000
EASF05 : 000000000
EASF06 : 000000000
EASF07 : 000000000
EASF08 : 000000000
EASF09 : 000000000
EASF10 : 000000000
EASD01 : 000000491
EASD02 : 000000000
EASD03 : 000000000
EASD04 : 000000000
IRMS1 : 000
IRMS2 : 000
IRMS3 : 000
URMS1 : 000
URMS2 : 000
URMS3 : 227
PREF : 03
PCOUP : 03
SINSTS : 00000
SINSTS1 : 00000
SINSTS2 : 00000
SINSTS3 : 00000
SMAXSN : 00000
SMAXSN1 : 00000
SMAXSN2 : 00000
SMAXSN3 : 00000
SMAXSN-1 : 00000
SMAXSN1-1 : 00000
SMAXSN2-1 : 00000
SMAXSN3-1 : 00000
UMOY1 : 000
UMOY2 : 000
UMOY3 : 226
**STGE : 000B0411**
MSG1 : PAS DE          MESSAGE
PRM : 00000000000000
RELAIS : 000
NTARF : 02
NJOURF : 00
NJOURF+1 : 00
PJOURF+1 : 0000C001 0A008002 1000C001 NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE

From the Linky I receive only ASCII values. I could turn the value into uint32 (I have to dev a function to convert ascii to int) but it cost processor time during acquisition. If necessary, I can do it...

In aim to decoding STGE, it's only a conversion from string to hexadecimal. Then you have to follow the Enedis-NOI-CPT_54E documentation and doing bits comparisons

For example (compteur triphasé de test en standard) : 
STGE = 000B0411= 0x000B0411 = 10110000010000010001

Contact sec : bit 0 = 1 => ouvert
Organe de coupure : bit 1 à 3 = 000 => fermé
État du cache-bornes distributeur : bit 4 = 1 => ouvert 
Surtension sur une des phases : bit 6 =0 => pas de surtension
Dépassement de la puissance de référence : bit 7 = 0 => pas de dépassement
Fonctionnement producteur/consommateur : bit 8 = 0 => consommateur
Sens de l’énergie active : bit 9 =0 => énergie active positive
Tarif en cours sur le contrat fournisseur : bit 10 à 13 = 0001 => Energie ventilée sur index 2
Tarif en cours sur le contrat distributeur : bit 14 à 15 = 00 => Energie ventilée sur index 1
Mode dégradée de l’horloge (perte de l’horodate de l’horloge interne) : bit 16 = 1 => horloge en mode dégradée
État de la sortie télé-information : bit 17 = 1 => mode standard
État de la sortie communication Euridis : bit 19 à 20 = 01 => Activée sans sécurité
Status CPL  : bit 21 à 22 = 00 =>new/unlock
Synchronisation CPL : bit 23 =0 => compteur non synchronisé
etc ....

pipiche38, 2022-09-12:
This is just from a consistency standpoint that it should be send as an uint32. You are already sending a number of attributes in binary mode, so why not that one .

closed 2022-09-12 by pipiche38