fairecasoimeme/Zlinky_TIC #92: 0x0702/0x0110 is documented as RP how when Report Config we got an error

pipiche38, 2022-09-12:
[image]

[image]

fairecasoimeme, 2022-09-12:
I confirm that cluster 0x702 / 0x110 is in RP mode. 
your trace is about read reporting config (0x122 command) not a setting of a report config (0x120 command)
This kind of error could arrive when setting a config report is not passed.

pipiche38, 2022-09-12:
Juste after a fresh reset of the ZLinky
[image]

pipiche38, 2022-09-12:
[image]

[image]

fairecasoimeme, 2022-09-12:
After a reset or a flash erase ?

pipiche38, 2022-09-12:
A reset

fairecasoimeme, 2022-09-12:
No a reset is a reboot with a clear of RAM memory and registry . If you want to clear cfg binding table, pairing and other flash memory, it's a factory reset. So what did you do ?

closed 2022-09-12 by pipiche38