fairecasoimeme/Zlinky_TIC #93: Reporting Configuration of 0x0b04/0x050b.... result in Software Failure !

pipiche38, 2022-09-12:
[image]
[image]
[image]
[image]
[image]
[image]

fairecasoimeme, 2022-09-12:
This error message can occur if the config report table is full.
You can verify with a usb-ttl module connected to 115200 baud. when you use the reset button, the device reboot with the config report table

pipiche38, 2022-09-12:
How can Config Report table be full just after a RESET ? How can we clean it ?

- The ZLinky has been reset

        "ConfigureReporting": {
                "0702": {
                        "Attributes": {
                                "0100": { "DataType": "25", "MinInterval": "0046", "MaxInterval": "0E10", "TimeOut": "0000", "Change": "000000000005" },
                                "0102": { "DataType": "25", "MinInterval": "0046", "MaxInterval": "0E10", "TimeOut": "0000", "Change": "000000000005" },
                                "0104": { "DataType": "25", "MinInterval": "0046", "MaxInterval": "0E10", "TimeOut": "0000", "Change": "000000000005" },
                                "0106": { "DataType": "25", "MinInterval": "0046", "MaxInterval": "0E10", "TimeOut": "0000", "Change": "000000000005" },
                                "0108": { "DataType": "25", "MinInterval": "0046", "MaxInterval": "0E10", "TimeOut": "0000", "Change": "000000000005" },
                                "010a": { "DataType": "25", "MinInterval": "0046", "MaxInterval": "0E10", "TimeOut": "0000", "Change": "000000000005" },
                                "010c": { "DataType": "25", "MinInterval": "0046", "MaxInterval": "0E10", "TimeOut": "0000", "Change": "000000000005" },
                                "010e": { "DataType": "25", "MinInterval": "0046", "MaxInterval": "0E10", "TimeOut": "0000", "Change": "000000000005" },
                                "0110": { "DataType": "25", "MinInterval": "0046", "MaxInterval": "0E10", "TimeOut": "0000", "Change": "000000000005" },
                                "0112": { "DataType": "25", "MinInterval": "0046", "MaxInterval": "0E10", "TimeOut": "0000", "Change": "000000000005" }
                        }
                },
                "0b04": {
                        "Attributes": {
                                "0508": { "DataType": "21", "MinInterval": "000A", "MaxInterval": "0E10", "TimeOut": "0000", "Change": "0001" },
                                "050b": { "DataType": "29", "MinInterval": "000A", "MaxInterval": "0E10", "TimeOut": "0000", "Change": "0001" },
                                "050d": { "DataType": "29", "MinInterval": "000A", "MaxInterval": "0E10", "TimeOut": "0000", "Change": "0001" },
                                "050f": { "DataType": "21", "MinInterval": "000A", "MaxInterval": "0E10", "TimeOut": "0000", "Change": "0001" },
                                "090b": { "DataType": "29", "MinInterval": "000A", "MaxInterval": "0E10", "TimeOut": "0000", "Change": "0001" }
                        }
                },
                "ff66": {
                        "Attributes": {
                                "0203": { "DataType": "23", "MinInterval": "000A", "MaxInterval": "fffe", "TimeOut": "0000", "Change": "00000001" },
                                "0204": { "DataType": "23", "MinInterval": "000A", "MaxInterval": "fffe", "TimeOut": "0000", "Change": "00000001" },
                                "0205": { "DataType": "23", "MinInterval": "000A", "MaxInterval": "fffe", "TimeOut": "0000", "Change": "00000001" },
                                "0206": { "DataType": "23", "MinInterval": "000A", "MaxInterval": "fffe", "TimeOut": "0000", "Change": "00000001" },
                                "0210": { "DataType": "29", "MinInterval": "000A", "MaxInterval": "fffe", "TimeOut": "0000", "Change": "0001" }
                        }
                }
        },

fairecasoimeme, 2022-09-12:
Please share your debug console to see the config report table.

To clear all the table you have to do a factory reset --> press 10 sec to the link button.

closed 2022-09-12 by pipiche38