fairecasoimeme/Zlinky_TIC #96: Triphasé :SINSTS1, SINSTS2 et SINSTS3 valeur énorme

baptou60, 2022-09-14:
Bonjour,
Je possède le zlinky en version de FW : 4000-0006. 
Je suis sur une installation triphasé et je remonte les infos dans Jeedom via Zigbee2MQTT.

Je trouve que mes valeurs SINSTS1, SINSTS2 et SINSTS3 (puissance instantané par phase) ont des valeurs très importantes malgré qu'il y ait personne à la maison et pas de période chauffe.
Pour le même relevé de valeur, j'ai :
SINSTS1 = 9871VA
SINSTS2 = 46688,65VA
SINSTS3 = 25966VA
A votre avis, est ce l'unité VA qui n'est pas bonne. La virgule n'est elle pas à la bonne place. 
Comme (selon ce que j'ai lu) 1VA = 1Watt, il est impossible à mon sens d'avoir la phase 2 à 46Kw ou même la phase 3 à 25Kw.
Pour info, je consomme par jour entre 13Kw et 17Kw selon les remonté sur l'application EDF et moi.
J'interprète peut être mal les données, c'est pourquoi je vous sollicite pour essayer de comprendre.
D'avance merci.

BrunoSpy, 2022-09-21:
Attention, les puissances apparentes SINSTS sont des valeurs instantanées, pour comparer avec la consommation journalière, il faut multiplier par la durée de consommation.

Ceci dit, on voit effectivement un petit souci sur SINSTS3 qui ne descend jamais en dessous de 20 kVA, soit à la louche 480 kWh rien que pour cette phase. Certainement moins car VA est toujours supèrieur à W (il faut prendre en compte le déphasage) mais l'ordre de grandeur est surprenant et je serai surpris que vous ayez autant de puissance sur une seule phase (surtout sur du triphasé).
Regardez

baptou60, 2022-09-21:
Ce qui me rassure c'est que les valeurs vous semblent extrèmes également. 
Maintenant, je ne vois pas comment je pourrais identifier d'où vient ces informations érronné vu qu'elles sont extraite du compteur Linky..
C'est pour cela que je pensais à un problème d'unités dans les valeurs..

BrunoSpy, 2022-09-21:
Vous pouvez déjà afficher les valeurs directement sur le linky avec les boutons.

thieryal, 2022-09-21:
Est-ce que par hasard ton Linky est un sagemcom S34C3/LNE29080 ?

pipiche38, 2022-10-01:
Getting similar issues on a BASE mode with Firmware version8. Going back to version 6 which was more stable from that side
[image]