fairecasoimeme/Zlinky_TIC #98: confusion route / no_route

Albirew, 2022-09-20:
Latest beta release shows 2 .ota files with one's name with no_route appended.
Issue is that there's no documentation or comment in release notes or readme.md about it.

I understand i have to do with "*Add 2 config files. With or without childs routing function*", but there's no clear explanation about this "routing" thing. 
Is it some kind of linking to an hardware (to my conbee in my case)?
Does I need to do something more after installing release, like authorize linking to my conbee on the zlinky side?

So I'm asking here instead : What does this no_route means, witch one should i take, and is there additional steps after upgrade?

fairecasoimeme, 2022-09-21:
comment here : https://github.com/fairecasoimeme/Zlinky_TIC/blob/master/README.md#route-or-not-route-from-v7

closed 2022-09-21 by fairecasoimeme

pipiche38, 2022-09-23:
looks like they are more un documented changes between the 2  #99