fairecasoimeme/Zlinky_TIC #101: Energie produite PV

BrunoSpy, 2022-09-29:
Bonjour,
Quel est le champ qui permet de relever l'énergie produite par les panneaux solaires en autoconsommation.
Si je comprends bien EAIT permet de connaitre l'énergie injectée sur le réseau (le surplus non consommé) mais je souhaiterais relever la partie autoconsommée.
Merci d'avance

fairecasoimeme, 2022-09-30:
Bonjour,
je ne suis pas expert dans les systèmes d'autoconsommation mais vous pouvez retrouver toutes les étiquettes en rapport avec la production ici (colonne comment): https://github.com/fairecasoimeme/Zlinky_TIC#mode-standard

BrunoSpy, 2022-10-02:
Merci
J'ai donc ma réponse : on ne peut avoir que la puissance et énergie injectée dans le réseau, pas la production elle-même. Ce qui est fort logique au demeurant.

closed 2022-10-02 by BrunoSpy