fairecasoimeme/Zlinky_TIC #104: Ne recoit plus grand chose d'actuel

gromo69, 2022-10-03:
Hello
Depuis les mises a jour OTA 0006-0007-0008-0009 je ne reçois plus grand chose à jour comme valeur ! 
En effet seul TIC_current_date semble updater toutes les minutes
Dans les logs Zigbee2MQTT je vois cela toute les 60sec. : "zlinky config: standard, three_phase, true, ENGIE ELEC WE"
zha core 2022.9.1 Z2M 1.28.0-1 

restart zha ou Z2M n'y change rien, idem pour reconfigurer le LiXee...

Je ne sais pas comment revenir en arrière OTA sur le firmware 0005 par exemple

Merci de votre aide

Thierry

fairecasoimeme, 2022-10-03:
Bonjour,
Pour tester si le ZLinky répond, vous pouvez aller dans l'onglet console dev et faire une requête manuelle : 
[image]
Certains utilisateurs ont ce type de problème mais je ne sais pas d'où cela provient

antoinelibert, 2022-10-03:
Bonjour,

Après une mise à jour, il faut penser à le reset : supprimer de Z2M en forçant le unpair, reset le zlinky (10 secs sur le bouton) puis pair de nouveau . J'ai rarement eu un comportement normal sans faire un reset après une màj.

gromo69, 2022-10-04:
OK merci à tous... en effet supprimé, resetté, ré-appairé puis renommé.. c'est bon j'ai un update complet des valeurs... :-)

closed 2022-10-04 by gromo69