fairecasoimeme/Zlinky_TIC #105: optimiser le circuit de détection des trames TIC

christopheVia, 2022-10-03:
Le compteur Linky comporte deux modes de fonctionnements, historique et standard. Les données sont envoyées entre les contacts I1 et I2 à travers une modulation en amplitude sur une porteuse de 50kHz. En mode historique, le débit de données est de 2400bauds mais il monte à 9600bauds en standard. Sur ce dernier mode, la différence de fréquence entre les données et la porteuse n'est plus si grande que cela et cela peut parfois engendrer des mauvaises détections. 
Dans la pluspart des montages que l'on trouve sur le net, la démodulation est confiée à un optocoupleur. Mais tous les optocoupleurs ont des dispersions importantes de leur caractéristiques et en particulier les capacités parasites qui sont liées à la bande passante du circuit. Si on n'a pas de chance, l'optocoupleur XXX va mal fonctionner chez soi alors que le même circuit fonctionne sans problème chez son voisin. En se basant sur ce constat, je propose le schéma suivant (dessiné et simulé sous LTspice) :
[image]

Je me suis imposé le cahier des charge suivant :
- faire au plus simple = pas de détection AM synchrone
- le moins cher possible = composants standards et facilement trouvables
- consommation minimale tirée de l'alimentation 3,3V du module Zigbee
- Ne pas confier la démodulation AM à l'optocoupleur

Le circuit commence entre les noeuds I1-I2. Le reste à gauche tente de modéliser le signal qui sort du Linky (amplitude 12V ici mais c'est 25V à vide).

Détection d'enveloppe à diode : 
Le pont à diode D2-D3-D4-D5 commence par faire un redressement double alternance du signal pour faire un x2 à la fréquence de la porteuse. Des diodes Schottky sont indispensable ici pour éviter de perdre trop d'amplitude due à la tension de seuil. Cette partie est indispensable au bon fonctionnement de la démodulation parce qu'elle diminue l'amplitude des ondulations présentes sur le signal d'enveloppe. Cette détection d'enveloppe est opérée par C1 en parallèle avec une résistance constituée par la résistance dynamique de la LED de l'optocoupleur et R5. La constante de temps de ce circuit RC est choisie pour couper la porteuse tout en laissant passer le débit binaire de 9600bps. La détection d'enveloppe (=l e signal binaire qui nous intéresse) est faite avant l'optocoupleur. La LED de l'optocoupleur mesure et envoie un courant qui est l'image du signal numérique. Les résidus de porteuse ne nous embetteront pas. 
La LED de l'optocoupleur n'est pas utilisée pour faire la détection d'enveloppe, ça ne va pas assez vite (trop de capacité parasite).

Mise en forme des signaux :
Les optocoupleurs classiques comme le PC817 ne sont pas très rapides lorsqu'ils sont utilisés comme suggéré dans leur datasheet. Ils le sont beaucoup plus lorsque ils sont utilisés dans la configuration du schéma ci-dessus. Une grande bande passante est appréciable ici, même si la démodulation n'est pas confiée à l'optocoupleur parce que l'on cherche malgré tout à ce qu'il commute rapidement. L'émetteur du photo-transistor attaque la base de Q1. Pour Q1, un bipolaire est choisi, et surtout pas un MOS, parce que la faible impédance d'entrée permet de maintenir assez bas le temp de transition de l'état ON vers OFF du photo-transistor. La résistance R4 est aussi indispensable pour limiter le stockage de charges dans la base. Q1 inverse également le signal numérique : les 1 logiques sont transmis par l'état bas dans le signal modulé et il faut retourner tout ça à la fin.

(oublier R8 qui n'est là que pour éviter à LTspice de planter)

Voici quelques captures d'écran issus de la simulation avec une porteuse à 12V
[image]
Le signal issu du redresseur double alternance pour la détection d'enveloppe est en rouge. La porteuse est déjà bien atténuée. Le courant capté par la LED de l'optocoupleur est en bleu marine.
Le signal bleu clair est le signal de sortie. Plus aucun résidu de porteuse n'est perceptible.
L'ensemble tire 500µA sur le rail d'alimentation 3,3V (en rose, coruant négatif parce que convention géné). 
[image]
idem mais avec une porteuse réglée à 25V. Jje ne sais pas vraiment ce que le Linky sort réellement. La doc Enedis n'est pas d'une clarté époustouflante.

Enfin, je suggère de remplacer les BAV21 ( @fairecasoimeme j'ai bon pour la ref ?) par des diodes Schottky 1A car la tension de seuil tombe à 0,3 - 0,35V au lieu de 0,7V dans le cas de diodes silicum classiques. Lorsque l'alim doit fournir trop de courant, la tension issue du générateur entre les contacts A et I1 chute. Tout s'arrête lorsque la tension appliquée à l'entrée du régulateur n'est pas suffisante pour le faire fonctionner. Avec des diodes Schottky la limite en puissance est repoussée un petit peu plus haut.

fairecasoimeme, 2022-10-04:
Merci beaucoup pour la contribution et l'analyse.  
`En mode historique, le débit de données est de 2400bauds` Attention, la vitesse est de **1200bauds** me semble t-il.

Je vais, du coup, tester ton schéma en réel pour pouvoir faire des tests avec les différents Linky que j'ai et voir la différence avec le schéma originel.

Pour les BAV21W, oui c'est la bonne réf mais à l'origine je les avais choisi car Enedis demandais que le module soit 230V compliant (pas pour que ça fonctionne à cette tension mais que ça ne dégrade pas) mais dans la discussion avec eux ça n'était pas applicable au ZLinky (le module empêche le branchement d'autres équipements par erreur) et je me suis arrêté là puis je ne suis pas revenu en arrière sur la réf. 

Du coup, ce point me semble important et je vais repasser sur des Schottky à faible chute de tension.

Malgré tout, je n'ai quasi jamais eu de retour sur un problème de décodage du Linky. La majorité des problèmes sont sur le blocage du ZLinky lié à des problèmes de tensions:
- soit une grosse chute qui a été corrigé en grande partie par le condensateur (voir [errata](https://github.com/fairecasoimeme/Zlinky_TIC#errata-hardware))
- soit une baisse de tension progressive sans tomber à zéro qui met l'oscillateur en PLS que je corrige normalement avec un superviseur de tension BD48L26G-TL

Enfin, Enedis a reconnu à travers les remontées clients que certains Linky ne respecte pas le tableau : 
[image]

Certains utilisateurs ont :
- Tension à vide : 2.97 V
- Tension quelques secondes après branchement (connexion Zigbee rétablie) : 1.87-1.90 V

La mesure est prise entre A et i1, du coup, les diodes + les régulateurs, le micro démarre normalement à partir de 1.9V. Difficile de faire quelque chose avec ça. Avec le superviseur à 2.6V, au moins ça permet de ne pas démarrer plutôt que d'avoir un système bancal. Peut-être qu'un buck boost peut faire l'affaire ... mais pas magique non plus.

Je vous tiendrai au courant sur ce fil de mes tests avec ces modifications.

Merci encore pour la contribution

christopheVia, 2022-10-04:
Je corrige pour 1200bauds. Si ça marche pour 9600, ça marchera pour 1200. Dans tes expérimentations, il faudra peut être retoucher R6 et R2, voire ne mettre qu'une seule résistance avant le pont de diode. Il ne faut pas dépasser 0,6mA dans la LED de l'optocoupleur. 
J'ai cru comprendre aussi que l'impédance d'entrée ne doit pas dépasser 2kohm ? mais la Doc Enedis n'est pas clair et je ne suis pas sûr que ce soit critique.

Pour ce qui est des BAV21, j'ai pensé que c'était pour ça que tu avais choisi des 250V, par contre, ce que je comprends de la doc Linky c'est que la table 1 donne les caractéristiques de leur sortie. Mais ce n'est pas très clair. En outre, je ne vois pas pourquoi un module autonome sans aucun contact avec l'extérieur devrait tenir des surtensions de 250V.

L'idée du superviseur d'alim est bonne, il faut la garder parce qu'elle permet de ré-initialiser proprement le module. Par contre, pour les Linky qui ne présentent qu'1,9V en charge, c'est compliqué. Il existe bien des convertisseurs DC-DC SEPIC low-power mais la difficulté va être de trouver un ref capable de fonctionner avec une très basse tension d'entrée et qui en même temps soit capable de gérer des tensions d'entrée de 16V. Il y a le LT1613 ou le LT1703 mais ils n'acceptent pas plus de 10 à 12V d'entrée. Chez TI, je ne vois pas de circuit adapté.

fairecasoimeme, 2022-10-04:
> Mais ce n'est pas très clair. En outre, je ne vois pas pourquoi un module autonome sans aucun contact avec l'extérieur devrait tenir des surtensions de 250V.

 En fait, Enedis pensaient a des ERL avec des connecteurs accessibles et pensant à des utilisateurs qui branchent du 220V par erreur.

Et oui pas facile de trouver des reguls qui gèrent bien à la fois des tensions faibles et des tensions à 16V tout en étant performant.

> Est-ce qu'il ne faudrait pas qu'ENEDIS considère les linky qui ne délivrent que 3V comme défectueux ? Les utilisateurs dans ce cas demandent à le faire changer. Clairement, on ne peut pas faire grand chose en auto-alimenté avec des Linky comme ça.

Je leur ai posé la question pour les clients dans ce cas et pour le moment, il n'existe pas de procédure pour changer un Linky avec la TIC défectueuse. Rien n'est prévu. On devrait en parler encore à la prochaine réunion avec eux. Ils sont très à l'écoute mais étant une grande "institution" ça ne bouge pas trop vite.

Tolriq, 2022-10-04:
Surement une question bête vu que j'y connais rien, mais pour ceux qui ont les Linky pourris (probablement mon cas) il n'y a pas moyen de créer un petit module d'alimentation externe à insérer entre les 2?

christopheVia, 2022-10-04:
> En fait, Enedis pensaient a des ERL avec des connecteurs accessibles et pensant à des utilisateurs qui branchent du 220V par erreur.
>
> Et oui pas facile de trouver des reguls qui gèrent bien à la [...]

Par contre, pour les Linky "normaux", un régulateur SEPIC peut aussi être une bonne idée. En améliorant le rendement global de l'alimentation, tu pourrais peut être éviter d'avoir à maintenir deux codes informatiques sans et avec le mode routeur.

fairecasoimeme, 2022-10-04:
@christopheVia il est évident que si j'arrive à améliorer la partie alimentation, il n'y aurait plus qu'un mode.

@Tolriq oui, il y a moyen d'alimenter le zlinky sans passer par l'autoalimentation du Linky. Tu peux suivre ce schéma et prendre que le 3.3v et GND https://github.com/fairecasoimeme/Zlinky_TIC#mise-%C3%A0-jour-du-firmware-non-ota

fairecasoimeme, 2022-10-10:
Bonjour,
J'ai commencé par faire le changement des diodes par des Schottky (B5819W-TP). Malheureusement, ce changement ne change rien sur la tension.
J'ai fait des tests sur la LED pour baisser la conso. La version du circuit actuelle fait que la LED n'est pas sur une broche PWM compatible. Mais j'ai utilisé un timer (simulant un peu le PWM). Du coup, j'ai réussi, avec cette méthode, à gagner 0.3V. C'est pas un miracle mais ça peut aider. Sur la nouvelle version, ce sera piloter par un vrai PWM. Je vous tiendrai au courant des avancements.

christopheVia, 2022-10-12:
@fairecasoimeme 
En cherchant pour une autre appli au boulot, je suis tombé sur le TPS560430. Il s'agit d'un régulateur DC-DC buck en boitier SOT23-6, pas cher, simple à régler, rendement excellent qui pourrait sérieusement accroitre le courant disponible pour le module zigbee.
Une alim 15V donnée pour 120mW peut délivrer 8mA max. en continu à la charge (en considérant que la tension ne bouge pas jusqu'à la limite en courant). Abaissée à 3,3V avec un régulateur linéaire, on pourra éventuellement se rapprocher de ces 8mA sur la ligne 3,3V si la conso interne du régulateur n'est pas trop élevée. La même alim cablée sur un régulateur buck pour descendre à 3,3V va pouvoir délivrer jusqu'à 32mA en sortie, en considérant un rendement de 90%. ça change tout, non ?

christopheVia, 2022-11-04:
Je viens de voir dans le wiki les photos de la clé en version "antenne intégrée". Je dispose pour ma part de la version à antenne déportée.

Petite remarque pour @fairecasoimeme, toujours dans l'idée de proposer des pistes d'amélioration du produit : 
L'antenne zigzag du module radio se trouve au dessus du PCB du Zlinky. Même si aucun plan de masse ne s'y trouve pour éviter les problèmes de rayonnement, il se peut que la seule présence d'un PCB à cet endroit ait pour effet de décaler la fréquence de résonance de l'antenne, dégradant par là même la portée de la radio.
En général, une ouverture est pratiquée sous l'antenne pour éviter d'augmenter la capacité parasite de l'élément rayonnant sous l'effet de la constante diélectrique de l'époxy FR4 plus élevée que celle de l'air (entre 4 et 5, vs 1 pour l'air). Cette [doc](https://www.infineon.com/dgdl/Infineon-AN91445_Antenna_Design_and_RF_Layout_Guidelines-ApplicationNotes-v09_00-EN.pdf) explique pas mal de choses sur ces antennes.