fairecasoimeme/Zlinky_TIC #108: Mise à jour de l'index (smart energy consumption)

Kiyozz, 2022-10-07:
Hello,

J'utilise le ZLinky TIC sur Home Assistant via ZHA.

Je rapporte un problème : l'index Linky ne se met pas à jour (home assistant ou l'interval d'update est large ?)

Il peut se mettre à jour parfois quand je redémarre Home Assistant

stadros83, 2022-10-08:
J'ai le même soucis :)
Par contre la consommation instantanée fonctionne bien.

Kiyozz, 2022-10-08:
Ma consommation dans la page Énergie est complètement fausse également

fairecasoimeme, 2022-10-10:
Bonjour,
Malheureusement, il n'y a pas encore de compatibilité native avec ZHA. Il faut suivre ce tuto pour paramétrer et faire remonter les données que l'on souhaite gérer.  https://forum.hacf.fr/t/zlinky-tic-ou-connecter-simplement-linky-en-zigbee/7112/154?u=max

stadros83, 2022-10-10:
Merci pour le retour.
Est-ce qu'une compatibilité avec ZHA est prévue ?
Si j'en crois ce qui est écrit ici il y a des correctifs à apporter au firmware : https://github.com/home-assistant/core/issues/78984
Après je n'y connais rien :(

Kiyozz, 2022-10-10:
C'est quand même bien dommage de se faire livrer un module non mis à jour à sa dernière version

stadros83, 2022-10-10:
J'ai pu faire la mise à jour OTA en version 5 via Home Assistant donc pas de problème là-dessus pour moi.
Par contre ça ne corrige pas le problème de remontée d'information (pour l'instant).
J'ai vu qu'il y avait de nouveaux firmware mais en mode pre release.

fairecasoimeme, 2022-10-10:
> C'est quand même bien dommage de se faire livrer un module non mis à jour à sa dernière version

Le module est livré dans sa dernière version officiel au moment de sa livraison ...

fairecasoimeme, 2022-10-10:
Je pense que c'est la remontée de la version qui n'est pas ok. Essayez de supprimer puis réappairer. La version 01 n'a pas été produite depuis Février 2022.

stadros83, 2022-10-10:
Désolé si j'insiste mais est-il prévu de corriger les "erreurs" dans le firmware pour que celui-ci soit compatible avec ZHA (et le standard Zigbee)  ?

https://github.com/home-assistant/core/issues/78984

fairecasoimeme, 2022-10-10:
> Désolé si j'insiste mais est-il prévu de corriger les "erreurs" dans le firmware pour que celui-ci soit compatible avec ZHA (et le standard Zigbee) ?
> 
> [home-assistant/core#78984](https://github.com/home-assistant/core/issues/78984)

Oui ça été corrigé côté firmware. https://github.com/home-assistant/core/issues/78984#issuecomment-1264272508

stadros83, 2022-10-10:
Ah cool merci.
On peut l'installer sans "craindre de soucis" ?
Ou mieux vaut-il attendre encore (je ne suis pas pressé).

Kiyozz, 2022-10-10:
J'ai migré sur Z2MQTT, ça marche x100 mieux. Je dispose maintenant de beaucoup plus d'informations.

Hedda, 2022-10-17:
> Oh cool thank you. Can we install it without "fear of worries"? Or is it better to wait some more (I'm in no hurry).

Assuming that you have already updated the [ZLinky_TIC Router v9.0 firmware pre-release image](https://github.com/fairecasoimeme/Zlinky_TIC/releases) on your Zlinky TIC then you will still need to wait for next ZHA Device Handlers (zha-quirks) package to get released and bumped in Home Assistant core, see:

https://github.com/home-assistant/core/issues/78984

stadros83, 2022-11-03:
J'ai fait les deux dernières mises à jour, v10 et v11.
Aucune ne fonctionne convenablement avec ZHA.
Même l'affichage des watts instantané n'a fonctionné que quelques heures, maintenant le Zlinky ne renvoit plus rien ...

piitaya, 2022-11-26:
J'ai commencé l'intégration des différentes consommation (heures creuses/pleines) dans ZHA : https://github.com/home-assistant/core/pull/82602

Par contre je ne comprend pas trop pourquoi les valeurs des consommations ne sont pas mises à jour automatiquement. Je dois aller dans les attributs des clusters et faire "lire". Quelqu'un aurait une idée ? 😅

WarC0zes, 2022-11-27:
@piitaya 
Content de voir des développeurs francais qui s'occupe du Zlinky sous ZHA. Car depuis ca sortie ( 1 ans ) c'est du bricolage sous ZHA pour récupérer les infos  HP et HC.