fairecasoimeme/Zlinky_TIC #110: Firmware no_router OTA

bemble, 2022-10-09:
Bonjour,

J'aimerai mettre le firmware no_router sur le zlinky (j'ai plus de 60 devices zigbee et je pense que mon zlinky est victime de son succès sur mon réseau) mais je n'ai que l'update OTA dispo.

Je suis sous zigbee2mqtt. J'ai essayé de mettre une location custom :
```yaml
  ota:
    zigbee_ota_override_index_location: custom_index.json
```

Et mon fichier custom_index.json :
```json
[
    {
        "manufacturerCode": 4151,
        "imageType": 1,
        "url": "ZLinky_router_v9.0_no_route.ota",
        "force": true
    }
]
```

Sauf qu'on ne me propose pas la MAJ, j'ai toujours 0.9 standard. Comment faire ?

antoinelibert, 2022-10-09:
Il faut suivre cette procédure : https://github.com/fairecasoimeme/Zlinky_TIC/issues/89#issuecomment-1236422701

(Le converter lixee doit être remplacé par un custom qui spécifie une autre manière de faire l'OTA)

bemble, 2022-10-10:
Merci j’étais passé à côté.

closed 2022-10-10 by bemble