fairecasoimeme/Zlinky_TIC #111: ZLinky not working / no report attributes

pipiche38, 2022-10-09:
Because I detected that the ZLinky was sending its repoort to a router and not the coordinator 0x0000, I decided to redo the bind. And systématically as I already mentioned here the bind restult in Table full message, and when trying to unbind the response is entry do not exist

See the unbind command screenshot
[image]
[image]

And then  the bind command

[image]
[image]

pipiche38, 2022-10-09:
Here is a frame where the ZLinky 0xef8c sending apparent power to a Router !!!
[image]

fairecasoimeme, 2022-10-10:
This function has been tested and re tested and re re tested and it's work perfectly.
As you can see in your trace, you tried to unbind an unknow bind.

As I can see with the report to a router, I think you binded (and config reported) with a router and you tried to unbind with the coordinator mac. (which doesn't exist)

As I already said in https://github.com/fairecasoimeme/Zlinky_TIC/issues/93#issuecomment-1243816164 you can verify with debug console. plug your usb TTL module to the Zlinky and use putty with 115200baud. When you reset (reboot) you can get report table and bind table

here is an ZLinky bind table example : 
[image]

pipiche38, 2022-10-10:
Your function tested and re-tested and work perfectly but in your lab .
As your assumption, you are totaly wrong because the reporting works and suddenly stop working, and it is just by putting the sniffer I was able to see the ZLinky sending reports to a router .

It is up to you to give this level of support and do no trust people reporting issues.

closed 2022-10-10 by pipiche38

fairecasoimeme, 2022-10-10:
Copy here your debug console ...

In your side, trust the function which is well-tested and don't draw any quickly conclusions.

pipiche38, 2022-10-12:
here is the binding tables which show the correct bindings between the ZLinky and the coordinator
[image]

fairecasoimeme, 2022-10-12:
This is not that I asked ....
I repeat : plug your usb TTL module to the Zlinky and use putty with 115200 bauds. Push the ZLinky reset button (reboot) and copy/paste the report and bind table.

pipiche38, 2022-10-12:
at least what I'm showing is the binding exists in the bidding table, and the unbind/bin function doesn't work.
I do not have access to the ZLinky, so I cannot do more