fairecasoimeme/Zlinky_TIC #112: Puissance Apparente totale non retournée

arenoux, 2022-10-14:
Bonjour,
Je possède un lixee et un linky en mode triphasé.

Dans Zigbee2MQTT, je recupere bien les "Puissance appararente immediates delivrées" sur mes 3 phases 
| Commande TIC ?  | attribut          |Groupe de fonctionnalités |Titre Compteur|
| :--------------- |:---------------:|:-----|:-----|
| SINSTS1  |   apparentPower        |  haElectricalMeasurement | PUIS APP SOUTIR 1 |
| SINSTS2  | apparentPowerPhB             |   haElectricalMeasurement | PUIS APP SOUTIR 2 |
| SINSTS3  | apparentPowerPhC          |    haElectricalMeasurement | PUIS APP SOUTIR 3 |

Il me semble qu'il existe un attribut totalApparentPower qui permettrait d'avoir la somme de ses 3 attributs (d'ailleurs mon compteur me l'affiche)

| Commande TIC ?  | attribut          |Groupe de fonctionnalités |Titre Compteur|
| :--------------- |:---------------:|:-----|:-----|
| ????  |   totalApparentPower        |  haElectricalMeasurement | PUIS APP SOUTIR |

**Comportement observé** :

Lorsque je tente d'ajouter cet attribut dans les rapport de mon Lixee sur Zigbee2MQTT, j'obtiens l'erreur suivante : 

> Request 'zigbee2mqtt/bridge/request/device/configure_reporting' failed with error: 
'ConfigureReporting 0x00158d0005d2a634/1 haElectricalMeasurement([{"attribute":"totalApparentPower","minimumReportInterval":60,"maximumReportInterval":"900","reportableChange":0}], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'

**Comportement attendu** : 
On devrait pouvoir ajouter cet attribut et obtenir la valeur correpondante

fairecasoimeme, 2022-10-17:
Bonjour,
En effet, Il manque l'information de la puissance apparente totale.
l'étiquette SINSTS est gérée de la même manière que SINTS1 (même cluster / attribut) car je pensais que c'était juste l'équivalent du compteur monophasé mais pour le triphasé l'étiquette semble correspondre au total.

Je vais donc mettre à jour le firmware pour gérer SINSTS indépendamment pour récupérer la puissance apparente total.

D'ailleurs même situation pour SMAXS et SMAXS1. à vérifier.

Dès que ce sera OK, je vous tiendrai au courant.

arenoux, 2022-10-17:
Super !
Je confirme que c'est pareil pour la puissance MAX du coté de mon compteur :)

TROM83, 2022-10-31:
Bonjour,

Pour ce même problème, je viens de passer sur le firmware 10. Mais je n'ai toujours pas de valeur SINSTS qui gère la puissance apparente totale.
Pour l'instant, j'ai créé un sensor qui est la somme de SINSTS1, SINSTS2 et SINSTS3 pour géréer cette valeur.
Je suis sous HA avec Zigbee2MQTT.

Merci pour le travail réalisé sur le ZLinky

Tonylolo, 2023-05-08:
Bonjour 

D'ailleurs est-ce la puissance apparente ou active qui est facturée ?
Car les 2 sont retournées par le linky.

Merci

arenoux, 2023-08-12:
Je confirme que c'est ok du coté du firmware

| Commande TIC ?  | attribut          |Groupe de fonctionnalités |Titre Compteur|
| :--------------- |:---------------:|:-----|:-----|
| SINSTS   |   totalApparentPower        |  haElectricalMeasurement | PUIS APP SOUTIR |
| SINSTS1  |   apparentPower        |  haElectricalMeasurement | PUIS APP SOUTIR 1 |
| SINSTS2  | apparentPowerPhB             |   haElectricalMeasurement | PUIS APP SOUTIR 2 |
| SINSTS3  | apparentPowerPhC          |    haElectricalMeasurement | PUIS APP SOUTIR 3 |

Reste plus qu'a mettre a jour https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/src/devices/lixee.js pour avoir les valeurs correctement retournées dans mon ZIgbee2MQTT/Home Assistant :)
Je vais essayer de leur proposer une pull request

closed 2023-08-12 by arenoux

SilentT-FR, 2023-08-12:
@arenoux  https://github.com/Koenkk/zigbee-herdsman-converters/compare/master...SilentT-FR:zigbee-herdsman-converters:FIX-112 ?

SilentT-FR, 2023-08-12:
ah oui ta presque tout refait :D

arenoux, 2023-08-12:
@SilentT-FR , j'ai proposé une version un peu différente de la tienne pour conserver leur "logique" avec les phases...
Phase 1: xxx
Phase 2: xxxPhB
Phase 3: xxxPhC

https://github.com/Koenkk/zigbee-herdsman-converters/pull/6070/commits/d77964e946d7b25432390275e193a4eb3aab9049

En réalité je préfère ton approche, mais si on veut être cohérent, il faudrait reprendre tous leurs noms de data pour les triphasées...

SilentT-FR, 2023-08-12:
c'est vrais que l'approche permet de bien avoir le nommage de toute les phase 
mais cela demande de rename apparent_power donc certainement perte de data

SilentT-FR, 2023-08-12:
Mais dans la logique il faudrait que ce soit fait sur tout attribut a triphase 
active_power_max par exemple

SilentT-FR, 2023-08-12:
> Bonjour
> 
> D'ailleurs est-ce la puissance apparente ou active qui est facturée ? Car les 2 sont retournées par le linky.
> 
> Merci

Tres bonne question, comme les valeur EASD0X qui pour moi ne me concerne pas

arenoux, 2023-08-12:
> Mais dans la logique il faudrait que ce soit fait sur tout attribut a triphase active_power_max par exemple

Oui :) Il faudrait le rajouter mais pour l'instant j'ai pas pu tester cette data.
Je n'utilise que ApparentPower chez moi...