fairecasoimeme/Zlinky_TIC #114: PAPP et IINST ne remontent plus depuis MAJ 0.5 -> 0.9 no_route

bemble, 2022-10-18:
Bonjour,

J'avais des soucis de zLinky surchargé, du coup je suis passé sur la version 0.9 no_route du firmware. Cependant depuis cette migration, les valeurs de PAPP et IINST ne remontent plus :
[image]

Sur les graphs c'est vonstant car c'est la dernière donnée envoyée qui est utilisée, je n'en ai pas reçu depuis la MAJ 0.9 no_route. J'ai rehardé sur mon compteur, le PAPP est bien rempli.

A noter que j'ai également ceci qui remonte dans mes logs zigbee2mqtt :
```
Zigbee2MQTT:error 2022-10-18 14:06:01: Failed to configure 'ZLinky', attempt 2 (Error: Bind 0x00158d0005d2a16e/1 seMetering from '0x00212effff073f6f/1' failed (waiting for response TIMEOUT)
    at DeconzAdapter.bind (/app/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:768:19)
    at Endpoint.bind (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:567:13)
    at Object.bind (/app/node_modules/zigbee-herdsman-converters/lib/reporting.js:40:9)
    at Object.configure (/app/dist/util/externally-loaded.js:605:9)
    at Configure.configure (/app/lib/extension/configure.ts:115:13))
```

Une idée ?

fairecasoimeme, 2022-10-19:
Bonjour,
le message d'erreur correspond à un problème de portée. (Timeout sur la réponse)
Avez-vous essayé de faire sortir le ZLinky du réseau puis réappairer ?

pipiche38, 2022-10-19:
https://github.com/fairecasoimeme/Zlinky_TIC/issues/65#issuecomment-1267362376

bemble, 2022-10-20:
Curieux, parce que d'autres données remontent pourtant :
[image]

Enfin, j'ai essayé de sortir de zlinky du réseau et réappairer, mais impossible, je pense avoir un problème avec mon zlinky, j'ai peur de l'avoir brické en tentant de remettre le firmware v5 :
[image]

bemble, 2022-10-28:
Après reflashage complet via usb avec reset mémoire , tout est rentré dans l’ordre.

closed 2022-10-28 by bemble