fairecasoimeme/Zlinky_TIC #115: Firmware verso

mospaeda, 2022-10-19:
Hello
In addition, the firmware seems reported wrong in the display

ZLinky_TIC
par LiXee
Connecté via [Zigbee Coordinator]
Firmware: 0x00000001

BUT From the cluster details it seems that it is different (in my case firmware 9 as I received last week)
reference from offcial doc : https://github.com/fairecasoimeme/Zlinky_TIC#clusters

Basic (Endpoint id: 1, Id: 0x0000, Type: in)
Valeur 9

sw_build_id (id: 0x4000)
4000-0009

date_code (id: 0x0006)
20220930

I don't know from where the value 0x00000001 is coming but it is probably not pointing to right register

mospaeda, 2022-10-21:
For record
I received the LiXee ZLinky_TIC already with the firmware 9 installed 
I didn't update the received Zlinky by OTA to version 9

Basic -Endpoint id: 1, Id: 0x0000 - Value is 9
Basic - sw_build_id (id: 0x4000) - Value is 4000-0009
Basic - date_code (id: 0x0006) - Value is 20220930.

For some reason, the OTA - current_file_version is reported as 1

Question to @fairecasoimeme 
- is OTA current file version set to 1 , if no OTA was done or should it be 9 (or any other number) ?
- is firmware version in Basic or in OTA ?
- do I need to reload version 9 by OTA?

fairecasoimeme, 2022-10-21:
I didn't know that on ZHA the firmware version was related to the OTA cluster
For me the version firmware is in Basic not in OTA
I'll see to update also the version in OTA cluster

mospaeda, 2022-10-21:
Thanks 
I will check /confirm with codyhackw which cluster is used
This is related to https://github.com/zigpy/zha-device-handlers/issues/1844

nerthazrim, 2023-04-05:
Hi,

Same thing here with a Zlinky received with firmware 4000-0013.
Should I write the attribute "OTA current_file_version" to 13? Is it mandatory to be able to update OTA when the next firmware is available?

Thanks!

nerthazrim, 2023-04-16:
Hi @fairecasoimeme, hope you won't mind the ping. Just making sure you've seen the questions, please  :)