fairecasoimeme/Zlinky_TIC #116: Increase default reporting frequency for LiXee ZLinky_TIC Smartenergysummation

MayeulC, 2022-10-20:
Default reporting frequency is very slow. It's possible to increase it via ZHA Toolkit though:

Dev tools -> Service -> yaml mode 

```yaml
service: zha_toolkit.conf_report
data:
  ieee: sensor.lixee_zlinky_tic_electricalmeasurement
  cluster: 1794
  attribute: 0
  max_interval: 200
  reportable_change: 1
  min_interval: 40
```

This asks an arbitrary min reporting period of 40s, max of 200s, and min change of 1Wh. It would be nice to have it by default in the FW.

(_Originally posted in https://github.com/fairecasoimeme/Zlinky_TIC/issues/18#issuecomment-1285215566_)

The difference is pretty clear:

fairecasoimeme, 2022-10-20:
There aren't no default reporting in ZLinky at the begining. That is the coordinator accros a plugin/box which send config report to the ZLinky.

I think for this value It's ZHA which send a default config report with a low frequency

MayeulC, 2022-10-20:
Thank you for clarifying. It would probably be a good idea to set one; that's not an issue I've had with any other device, including connected plugs, etc.

On the other hand, it's OK to leave it like that, as long as it's documented, IMO.
It's quite hard to find good documentation on how to use the ZLinky (with zha). It might be worth opening this github repo's wiki, hosting a wiki somewhere else, or making a github pages-based website.

Ozaroth, 2022-10-27:
Hi,

It doesn't work for me, any idea why ?

my log: 

`event_data {'zha_toolkit_version': 'v0.8.20', 'zigpy_version': '0.51.3', 'zigpy_rf_version': '0.34.2', 'ieee_org': 'sensor.lixee_zlinky_smartenergysummation', 'ieee': '00:15:8d:00:05:00:6f:fe', 'command': 'conf_report', 'command_data': None, 'start_time': '2022-10-27T20:29:42.613552+00:00', 'errors': [], 'params': {'cluster_id': 1794, 'attr_id': 0, 'min_interval': 40, 'max_interval': 59, 'reportable_change': 1, 'dir': 0, 'tries': 1, 'expect_reply': True, 'args': [], 'read_before_write': True, 'read_after_write': True}, 'success': False, 'result_conf': Configure_Reporting_rsp(status_records=[ConfigureReportingResponseRecord(status=<Status.INVALID_VALUE: 135>, direction=<ReportingDirection.SendReports: 0>, attrid=0)])}`

Thanks !

MayeulC, 2022-10-28:
@Ozaroth did you change some options? I see "expect_reply", "read_before_write", "max_interval" among others, that I didn't specify when doing this.

Ozaroth, 2022-10-28:
@MayeulC only the max_interval.

service: zha_toolkit.conf_report
data:
   ieee: 00:15:8d:00:05:00:6f:fe
  cluster: 1794
  attribute: 0
  max_interval: 59
  reportable_change: 1
  min_interval: 40
  event_done: zha_done

and the result :

`event_type: zha_done
data:
  zha_toolkit_version: v0.8.20
  zigpy_version: 0.51.3
  zigpy_rf_version: 0.34.2
  ieee_org:
    - 254
    - 111
    - 0
    - 5
    - 0
    - 141
    - 21
    - 0
  ieee: 00:15:8d:00:05:00:6f:fe
  command: conf_report
  command_data: null
  start_time: "2022-10-28T15:15:27.528736+00:00"
  errors: []
  params:
    cluster_id: 1794
    attr_id: 0
    min_interval: 40
    max_interval: 59
    reportable_change: 1
    dir: 0
    tries: 1
    expect_reply: true
    args: []
    event_done: zha_done
    read_before_write: true
    read_after_write: true
  success: false
  result_conf:
    - - status: 135
        direction: 0
        attrid: 0
origin: LOCAL
time_fired: "2022-10-28T15:15:28.354774+00:00"
context:
  id: 01GGFJK3Z263W44HA9TM772QDY
  parent_id: null
  user_id: null
`

fairecasoimeme, 2022-10-28:
Hi, max_interval must be greater than 60 for reporting
`#define MAX_REPORT_INTERVAL       0x3d`

Ozaroth, 2022-10-28:
Thank you @fairecasoimeme it's working !

bachy, 2022-11-26:
Hi,
I'm trying to increase the active_power report frequence like you did (for now reporting every 30min, unusable)

the conf_report seems to works, but active_power does not come more frequently
```yaml
service: zha_toolkit.conf_report
data:
  ieee: 00:15:8d:00:05:ce:77:e0
  cluster: 2820
  attribute: 1291
  min_interval: 10
  max_interval: 61
  reportable_change: 1
  event_done: zha_done
```

```log
DEBUG (MainThread) [custom_components.zha_toolkit] event_data {
'zha_toolkit_version': 'v0.8.26', 
'zigpy_version': '0.51.5', 
'zigpy_rf_version': '0.19.0', 
'ieee_org': 00:15:8d:00:05:ce:77:e0, 
'ieee': '00:15:8d:00:05:ce:77:e0', 
'command': 'conf_report', 
'command_data': None, 
'start_time': '2022-11-26T16:32:41.630726+00:00', 
'errors': [], 
'params': {
  'cluster_id': 2820,
  'attr_id': 1291,
  'min_interval': 10,
  'max_interval': 61,
  'reportable_change': 1,
  'dir': 0,
  'tries': 1,
  'expect_reply': True,
  'args': [],
  'event_done': 'zha_done',
  'read_before_write': True,
  'read_after_write': True
}, 
'success': True, 
'result_conf': Configure_Reporting_rsp(status_records=[ConfigureReportingResponseRecord(status=<Status.SUCCESS: 0>)])}

```

```yaml
service: zha_toolkit.conf_report_read
data:
  ieee: 00:15:8d:00:05:ce:77:e0
  cluster: 2820
  attribute: 1291
```

```
DEBUG (MainThread) [custom_components.zha_toolkit] event_data {
'zha_toolkit_version': 'v0.8.26', 
'zigpy_version': '0.51.5', 
'zigpy_rf_version': '0.19.0', 
'ieee_org': 00:15:8d:00:05:ce:77:e0, 
'ieee': '00:15:8d:00:05:ce:77:e0', 
'command': 'conf_report_read', 
'command_data': None, 
'start_time': '2022-11-26T16:36:26.184368+00:00', 
'errors': [], 
'params': {
  'cluster_id': 2820, 
  'attr_id': 1291, 
  'dir': 0, 'tries': 1, 
  'expect_reply': True, 
  'args': [],
  'read_before_write': True,
  'read_after_write': True
}, 
'success': True, 
'result_conf': [
 {
  'cluster': 'Electrical Measurement', 
  'cluster_id': '0x0B04', 
  'ep': 1, 
  'attr_id': '0x050B', 
  'direction': <ReportingDirection.SendReports: 0>, 
  'status': <Status.SUCCESS: 0>, 
  'type': '0x29', 
  'min_interval': (10,), 
  'max_interval': (61,), 
  'attr': 'active_power'}
]}

any idea ?

misterbenj34, 2023-01-03:
Thanks @MayeulC , you're a life saver !
My ZLinky was working fine yesterday, but stopped reporting the "Summation Delivered" suddenly. Only the Apparent Power was reported every 40 seconds.
I used your trick with ZHA toolkit, and it restored the values

I had to migrate to ZHA yesterday, because my integration in deCONZ was also failing after 1 day..

nerthazrim, 2023-04-28:
> Hi, I'm trying to increase the active_power report frequence like you did (for now reporting every 30min, unusable)
> 
> the conf_report seems to works, but active_power does not come more frequently
> 
> ```yaml
> service: zha_toolkit.conf_report
> data:
>   ieee: 00:15:8d:00:05:ce:77:e0
>   cluster: 2820
>   attribute: 1291
>   min_interval: 10
>   max_interval: 61
>   reportable_change: 1
>   event_done: zha_done
> ```
> 
> ```
> DEBUG (MainThread) [custom_components.zha_toolkit] event_data {
> 'zha_toolkit_version': 'v0.8.26', 
> 'zigpy_version': '0.51.5', 
> 'zigpy_rf_version': '0.19.0', 
> 'ieee_org': 00:15:8d:00:05:ce:77:e0, 
> 'ieee': '00:15:8d:00:05:ce:77:e0', 
> 'command': 'conf_report', 
> 'command_data': None, 
> 'start_time': '2022-11-26T16:32:41.630726+00:00', 
> 'errors': [], 
> 'params': {
>   'cluster_id': 2820,
>   'attr_id': 1291,
>   'min_interval': 10,
>   'max_interval': 61,
>   'reportable_change': 1,
>   'dir': 0,
>   'tries': 1,
>   'expect_reply': True,
>   'args': [],
>   'event_done': 'zha_done',
>   'read_before_write': True,
>   'read_after_write': True
> }, 
> 'success': True, 
> 'result_conf': Configure_Reporting_rsp(status_records=[ConfigureReportingResponseRecord(status=<Status.SUCCESS: 0>)])}
> ```
> 
> ```yaml
> service: zha_toolkit.conf_report_read
> data:
>   ieee: 00:15:8d:00:05:ce:77:e0
>   cluster: 2820
>   attribute: 1291
> ```
> 
> ```
> DEBUG (MainThread) [custom_components.zha_toolkit] event_data {
> 'zha_toolkit_version': 'v0.8.26', 
> 'zigpy_version': '0.51.5', 
> 'zigpy_rf_version': '0.19.0', 
> 'ieee_org': 00:15:8d:00:05:ce:77:e0, 
> 'ieee': '00:15:8d:00:05:ce:77:e0', 
> 'command': 'conf_report_read', 
> 'command_data': None, 
> 'start_time': '2022-11-26T16:36:26.184368+00:00', 
> 'errors': [], 
> 'params': {
>   'cluster_id': 2820, 
>   'attr_id': 1291, 
>   'dir': 0, 'tries': 1, 
>   'expect_reply': True, 
>   'args': [],
>   'read_before_write': True,
>   'read_after_write': True
> }, 
> 'success': True, 
> 'result_conf': [
>  {
>   'cluster': 'Electrical Measurement', 
>   'cluster_id': '0x0B04', 
>   'ep': 1, 
>   'attr_id': '0x050B', 
>   'direction': <ReportingDirection.SendReports: 0>, 
>   'status': <Status.SUCCESS: 0>, 
>   'type': '0x29', 
>   'min_interval': (10,), 
>   'max_interval': (61,), 
>   'attr': 'active_power'}
> ]}
> ```
> 
> any idea ?

Hi @bachy, did you find an answer to your questions, by any chance?
I'm trying to do the same: increase the recording frequency of active_power by ZHA. Any help/guide would be appreciated :)

Thanks in advance

bachy, 2023-04-29:
Hi @nerthazrim 

No i never get to make it work.
I gave up on zlinky-tic.
I swictched to zigbee2mqtt, and now instead of zlinky i'm using multiple custom ptvo + pzem device which are working smoothly

MayeulC, 2023-04-29:
That cluster (2820/1291 aka b04/50b) sounds a bit weird:
> Point n de la courbe de charge active soutirée

And it's only available in standard mode. Are you sure it's the right one? I suggest you try to read a clister from the UI on the device page to confirm it returns a value, and the right one, before configuring the reporting period.

Maybe you are looking for `IINST | 0x0B04 | 0x0508` and you just mistyped/misread 8 as B?, Or `PAPP | 0x0B04 | 0x050F`?

Of course, another option is to differentiate the energy, as E=P*t  -> P=E/t (dE/dt, the derivative, is more technically correct). E in joules (=1 W.s), t in seconds, P in watts.

nerthazrim, 2023-04-30:
> Thank you @fairecasoimeme it's working !

Hi @Ozaroth, would you mind sharing the conf snippet you're using, please? :)

Ozaroth, 2023-04-30:
Hi @nerthazrim you want my conf for active_power ?
If yes my config for active_power is the same as you and not working for me too, the value is reported every 30 minutes.
but for current_summ_delivered it's working perfectly.

nerthazrim, 2023-05-02:
> Hi @nerthazrim you want my conf for active_power ? If yes my config for active_power is the same as you and not working for me too, the value is reported every 30 minutes. but for current_summ_delivered it's working perfectly.
Damn it :(
I thought you managed to find how to increase the frequency :(

I'll open a ticket with in the ZHA project. I guess that's the way to go.

nerthazrim, 2023-05-02:
Well, actually, quick one for you @fairecasoimeme: we can read here (on a similar issue)
https://github.com/home-assistant/core/issues/88166#issuecomment-1434857328

> We don't pull date from the device, the device push data upon changes, we don't control the update rate.

Would it be possible that the ZLinky does not advertise properly a change in the "active_power" value to ZHA? Which then does not record a new value?

MayeulC, 2023-05-03:
> Well, actually, quick one for you @fairecasoimeme: we can read here (on a similar issue) [home-assistant/core#88166 (comment)](https://github.com/home-assistant/core/issues/88166#issuecomment-1434857328)
> 
> > We don't pull date from the device, the device push data upon changes, we don't control the update rate.
> 
> Would it be possible that the ZLinky does not advertise properly a change in the "active_power" value to ZHA? Which then does not record a new value?

The device pushes updates, but the coordinator can configure how often, which is what `conf_report` does. You can specify a minimum time in between updates, as well as a maximum time, and a change after which an update should be sent, for each property (though I think there is a maximum, perhaps about 10, properties that can be reported this way, probably dependent on the device.

iamdabe, 2024-12-03:
This is still an outstanding issue I feel. I've just setup my zlinky and was disappointed at the 30min refresh rate of **sensor.lixee_zlinky_tic_power**  which makes it unusable for me. It's strange when **sensor.lixee_zlinky_tic_apparent_power** is updated near-real time but it reports in VA and without a power factor it's also useless for calculating the real time watts. 

For now (as I have sent back my ecojoko) I have setup a template trigger sensor which takes the **lixee_zlinky_tic_summation_delivered** and calculates the watts based on the time differential as it gives me something usable for now as it's a TEMPO RED day today and the children are due home soon!

The template sensor code:

`#LiXee ZLinky derived current power consumption
  - trigger:
    - platform: state
      entity_id: sensor.lixee_zlinky_tic_summation_delivered
    sensor:
      - name: "LiXee ZLinky Current Power Consumption"
        unique_id: "lixee_zlinky_current_power_consumption"
        device_class: power
        unit_of_measurement: W
        state_class: measurement
        state: >
          {% set prev_value = trigger.from_state.state | float(0) %}
          {% set curr_value = trigger.to_state.state | float(0) %}
          {% set prev_time = as_datetime(trigger.from_state.last_changed) %}
          {% set curr_time = as_datetime(trigger.to_state.last_changed) %}
          
          {% if prev_time and curr_time and curr_value > prev_value %}
            {% set time_diff = (curr_time - prev_time).total_seconds() %}
            {% set energy_diff = curr_value - prev_value %}
            {% if time_diff > 0 %}
              {% set result = ((energy_diff * 3600) / time_diff) * 1000 %}
              {{ result | round(0) }}               
            {% else %}
              0
            {% endif %}
          {% else %}
            0
          {% endif %}
        availability: >
          {{ trigger.from_state is not none and
             trigger.to_state is not none and
             trigger.to_state.state | is_number and
             trigger.from_state.state | is_number }}
`

nerthazrim, 2024-12-03:
My understanding was that it was the Linky that was updating the value no more than once every 30min, not a Zlinky's limitation.
Did I get this wrong?

iamdabe, 2024-12-03:
If that is the case then this is bizarre as ecojoko provides the value in watts and quite accurate it was. they must have no more info than ZLinky?

fairecasoimeme, 2024-12-04:
Ecojoko have 2 solutions : 
* with Linky sensor. only apparent power (VA) from Linky and active power (W) calculated from ecojoko
* with circuit breaker sensor. Approximative active power (W)

The Linky provides only apparent power (VA)

benriou, 2025-03-18:
Hello

I'm surprised of your concerns, because my linky with lixee tic reports an increment for *every* single watt consumed.

[image]

[image]

Cannot you deduct the consumed power from this metric ?

iamdabe, 2025-03-18:
> Hello
> 
> I'm surprised of your concerns, because my linky with lixee tic reports an increment for _every_ single watt consumed.
> 
> Cannot you deduct the consumed power from this metric ?

Yes, this is exactly what I did in the end (see my template sensor code above). It's proven to be pretty accurate over the last couple of months so I am happy!

Hopefully it helps others who come across this thread researching the same problem. Just takes a little manual work but that's ok!