fairecasoimeme/Zlinky_TIC #117: Reported CCASN doesn't change for 15 minutes

renaiku, 2022-10-20:
Hello,

Using Home Assitant + Zigbee2mqtt with the following configuration:
[image]
[image]
The measurement_poll_chunk is "0" by default not "1", I never changed it.

The used converter has:
```
    {cluster: clustersDef._0x0B04, att: 'activePower', reportable: true, onlyProducer: false, linkyPhase: linkyPhaseDef.all, linkyMode: linkyModeDef.standard, exposes: exposes.numeric('CCASN', ea.STATE).withUnit('W').withProperty('active_power').withDescription('Current point of the active load curve drawn')},
```

I get the following values in the logger every 60-61 seconds:
```
2022-10-20 12:54:14 Received Zigbee message from 'Linky', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":384}' from endpoint 1 with groupID 0
2022-10-20 12:55:16 Received Zigbee message from 'Linky', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":384}' from endpoint 1 with groupID 0
2022-10-20 12:56:17 Received Zigbee message from 'Linky', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":384}' from endpoint 1 with groupID 0

...

2022-10-20 13:04:26 Received Zigbee message from 'Linky', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":532}' from endpoint 1 with groupID 0
```
But the reported value is not changing. CCASN only changes every 15 minutes:
[image]

The CCASN value is updated only every 15 minutes. I'm looking to get my live consumption at least every minutes (I think zlinky is capable every 10s ?). Am I having a wrong configuration ? Or is it a converter issue ?

CladStriffe, 2022-10-26:
Same here, but it doesn’t seem to be a ZLinky limitation. Instead it seems to be an average value only provided every 15min by the Linky terminal. 
you should instead have a look at SINSTS to have a value in almost real-time. It is the apparent power in VA, so the real power used in your house as seen by the provider.

renaiku, 2022-10-28:
@CladStriffe J'ai jeté un oeuil à la documentation du linky dans le pdf de ENEDIS, y'a rien qui indique que CCASN est une valeur changeante toute les 15 minutes. Donc tant qu'on a pas mesuré, je me méfie ^^ !

Si SINSTS est en temps réel mais en VA, comment je le convertis en puissance active en W ?

CladStriffe, 2022-10-28:
> @CladStriffe J'ai jeté un oeuil à la documentation du linky dans le pdf de ENEDIS, y'a rien qui indique que CCASN est une valeur changeante toute les 15 minutes. Donc tant qu'on a pas mesuré, je me méfie ^^ !

Tu as raison, ils ne parlent pas de 15 minutes. 
Mais CCASN s'appelle dans la doc Enedis : "Point n de la courbe de charge active soutirée". 
Et la définition par Enedis de la courbe de charge : "Ensemble de valeurs **moyennes** horodatées de la puissance active ou réactive injectée ou soutirée, sur des périodes d’intégration consécutives et de même durée."
C'est donc bien une valeur moyenne sur une durée qui peut (si j'ai bien compris) a priori varier selon le paramétrage demandé au Linky par un fournisseur, mais d'au moins 10 minutes.

Une puissance c'est une tension (V) multipliée par un courant (A). Donc en terme d'unité, VA c'est équivalent à W.
Pour un usage domestique, la puissance apparente est à quelques % près égale à la puissance active.

renaiku, 2022-10-28:
> Et la définition par Enedis de la courbe de charge : "Ensemble de valeurs moyennes horodatées de la puissance active ou réactive injectée ou soutirée, sur des périodes d’intégration consécutives et de même durée."

Punaise bien vu !

Ok top merci ben je vais switcher sur SINSTS !

closed 2022-10-29 by renaiku