fairecasoimeme/Zlinky_TIC #118: Lixee installation firmware no route

lendormi054, 2022-10-22:
Subissant régulièrement des déconnexion de mon lixee, je voudrais essayer avec la version no_route. Mais je n'ai aucune idée de comment faire!!

Je suis sous home assistant et lixee est intégré via mqtt.
Un pti tuto de noob serait le bienvenue ;-)

lecourtb, 2022-10-22:
See https://forum.hacf.fr/t/zha-ota-update-firmware/9693/4

lendormi054, 2022-10-22:
@lecourtb 
Je peux utiliser 
`zha:
  zigpy_config:
    ota:
      otau_directory: /config/zigpy_ota`
alors que mon appareil est sous z2m ????

MayeulC, 2022-10-24:
@lendormi054 non, il faut que tu suives la procédure ici: https://www.zigbee2mqtt.io/guide/usage/ota_updates.html#local-ota-index-and-firmware-files

Tu crées un fichier json index local pour les ota:

```json
[
    {
        "url": "fichier.ota"
    }
]
```

Et tu indiques sa localisation dans le fichier de configuration.yaml:

```yaml
ota:
    zigbee_ota_override_index_location: my_index.json
```
(tous les chemins sont relatifs au dossier de `configuration.yaml`)

Enfin, il faut [envoyer une commande pour lancer l'OTA](https://www.zigbee2mqtt.io/guide/usage/ota_updates.html#update-to-latest-firmware). Chemin: `zigbee2mqtt/bridge/request/device/ota_update/update` donnée: `{"id": "deviceID"}` avec la MAC ou le nom de l'appareil.

Bonne chance (je n'utilise pas z2m, j'ai juste essayé d'expliciter un peu plus la documentation dont j'ai fourni les liens).

lendormi054, 2022-10-25:
@MayeulC 
La localisation du ficher dans configuration.yaml (qui est celui dans le répertoire zigbee2mqtt, c'est bon. Le ficher my_index.json, ca semble bon.
Je coince pour la suite. Pas de Maj trouvée et je ne sais pas ou verifier si tout est bon....
Mais ca progresse.... enfin je crois :-)

Tolriq, 2022-10-25:
Y'a un trick :)

https://github.com/fairecasoimeme/Zlinky_TIC/issues/89#issuecomment-1236422701

Par default ca utilise pas la conf mais un fichier a part, faut bien suivre le commentaire ^^

lendormi054, 2022-10-25:
@Tolriq 
Les étapes 3-4-5-6 du post https://github.com/fairecasoimeme/Zlinky_TIC/issues/89#issuecomment-1236422701, je vois comment faire.
En revanche l'étape 1 me pose probleme. J ai pas de commande docker (du moins dans le terminal...)

Tolriq, 2022-10-25:
Tu peut prendre le fichier directement à la source :
https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices/lixee.js

lendormi054, 2022-10-25:
@Tolriq 
Etape 1 ok. (merci pour le lien du fichier lixee.js)

Etape 2 : probleme :

`[16:52:44] INFO: Preparing to start...
[16:52:45] INFO: Socat not enabled
[16:52:47] INFO: Starting Zigbee2MQTT...
node:internal/modules/cjs/loader:936
  throw err;
  ^
Error: Cannot find module '../converters/fromZigbee'
Require stack:
- /app/dist/util/utils.js
- /app/dist/util/settings.js
- /app/index.js
    at Function.Module._resolveFilename (node:internal/modules/cjs/loader:933:15)
    at Function.Module._load (node:internal/modules/cjs/loader:778:27)
    at Module.require (node:internal/modules/cjs/loader:1005:19)
    at require (node:internal/modules/cjs/helpers:102:18)
    at /app/dist/util/externally-loaded.js:8:12
    at Script.runInContext (node:vm:139:12)
    at Script.runInNewContext (node:vm:144:17)
    at Object.runInNewContext (node:vm:298:38)
    at loadModuleFromText (/app/lib/util/utils.ts:148:8)
    at loadModuleFromFile (/app/lib/util/utils.ts:155:12)`

lendormi054, 2022-10-25:
Alors, avec ton fichier, j ai reussi à relancer z2m (j ai juste modifié le firmware (je veux  ZLinky_router_v9.0_no_route et pas le v5).

Il me dit qu'il y a une MAJ !!!! (sans etre certain que ce soit la bonne, mais "update in progress".
Et....
[image]

Peut etre que c'est du au fait que ce soit la meme version ?

MayeulC, 2022-10-25:
@lendormi054 tu as mis `"force": true` dans le json?

Merci @Tolriq pour les précisions (et au passage merci pour Yatse, que j'ai énormément utilisé en ~2009-2012 :P)

Tolriq, 2022-10-25:
Faut tester Symfonium alors ;)

lendormi054, 2022-10-25:
@MayeulC Oui oui, il y a l'option.
La question est bete mais dans le ficher my_index.json, il faut les [] ou juste les {}
J ai mis [{...}] pour info

MayeulC, 2022-10-26:
@lendormi054 il faut les deux, c'est une liste `[...]` de valeurs `{...}` qui peuvent correspondre à différents fabricants, etc. Je suis désolé, je ne vois pas mieux, ce serait peut-être intéressant de s'intéresser aux logs, et/ou de poser la question à la communauté z2m.

@Tolriq ok, lol, je testerai, merci (mais la musique ce n'est pas trop mon truc, même si tu as Finamp et Gelli en concurrents sur F-Droid pour Jellyfin). Enfin je ferai tester plutôt, je suis 100% FLOSS :P

Tolriq, 2022-10-26:
C'est pas le bon endroit pour en parler mais c'est pas vraiment des concurrents, ils sont open source et joue de la musique après on joue pas vraiment dans la même catégorie. Mais vu que c'est une app récente, si elle te plait je suis pas contre un petit commentaire Play Store.

lendormi054, 2022-10-26:
@MayeulC C'est déjà très bien. Je suis parti de rien ne marche à la mise à jour manuelle est disponible. Faudrait que j'essaye avec un firmware superieur mais ca ce sera pour un autre sujet. Merci à tous pour le coup de main et de m'avoir permit de progresser :-)

lendormi054, 2022-10-26:
L'installation manuelle du firmware lixee est disponible sous z2m

closed 2022-10-26 by lendormi054