fairecasoimeme/Zlinky_TIC #120: tic_command_whitelist non prise en compte

gauthier12, 2022-10-23:
Bonjour, 
j'essaie configurer mon nouveau Zlinky mais il y a des points un peu obscurs.
Le Zlinky est en mode standard en accord avec le compteur Linky. Il est connecté à mon réseau Zigbee et je gère l'ensemble via zigbee2mqtt.
La liste des valeurs renvoyé semble un peu aléatoire. Au fil des configurations, j'ai des valeurs qui apparaissent, disparaissent.
Je pense que c'est lié au tarif choisi.
J'ai un tarif HP/HC sem/WE qui utilise 4 index mais je ne trouve pas de tarif équivalent alors j'essaye de configurer le paramètre 
tic_command_whitelist avec 
``` HCHC,HCHP,BBRHCJW,BBRHPJW,PAPP,PTEC,NTARF,LTARF```
mais la liste que j'obtiens ne correspond pas  et j'obtiens deux fois les valeurs à chaque fois (voir plus bas).
Est-ce qu'il y a quelque chose que je ne comprends pas dans la configuration pour que ce soit pris en compte ?
Bonne journée

```
Oct 23 09:59:06 hermes npm[1029335]: zlinky config: standard, single_phase, false, HC SEM ET HC WE
Oct 23 09:59:06 hermes npm[1029335]: Zigbee2MQTT:info  2022-10-23 09:59:06: MQTT publish: topic 'zigbee2mqtt/linky/current_tarif', payload 'HC SEM ET HC WE'
Oct 23 09:59:06 hermes npm[1029335]: Zigbee2MQTT:info  2022-10-23 09:59:06: MQTT publish: topic 'zigbee2mqtt/linky/current_price', payload 'HP WEEK END'
Oct 23 09:59:06 hermes npm[1029335]: Zigbee2MQTT:info  2022-10-23 09:59:06: MQTT publish: topic 'zigbee2mqtt/linky/current_index_tarif', payload '2'
Oct 23 09:59:06 hermes npm[1029335]: Zigbee2MQTT:info  2022-10-23 09:59:06: MQTT publish: topic 'zigbee2mqtt/linky/rms_voltage', payload '236'
Oct 23 09:59:06 hermes npm[1029335]: Zigbee2MQTT:info  2022-10-23 09:59:06: MQTT publish: topic 'zigbee2mqtt/linky/apparent_power', payload '494'
Oct 23 09:59:06 hermes npm[1029335]: Zigbee2MQTT:info  2022-10-23 09:59:06: MQTT publish: topic 'zigbee2mqtt/linky/current_summ_delivered', payload '26263'
Oct 23 09:59:06 hermes npm[1029335]: Zigbee2MQTT:info  2022-10-23 09:59:06: MQTT publish: topic 'zigbee2mqtt/linky/current_tier1_summ_delivered', payload '4533'
Oct 23 09:59:06 hermes npm[1029335]: Zigbee2MQTT:info  2022-10-23 09:59:06: MQTT publish: topic 'zigbee2mqtt/linky/current_tier2_summ_delivered', payload '14653'
Oct 23 09:59:06 hermes npm[1029335]: Zigbee2MQTT:info  2022-10-23 09:59:06: MQTT publish: topic 'zigbee2mqtt/linky/current_tier3_summ_delivered', payload '2680'
Oct 23 09:59:06 hermes npm[1029335]: Zigbee2MQTT:info  2022-10-23 09:59:06: MQTT publish: topic 'zigbee2mqtt/linky/current_date', payload 'E221023093456'
Oct 23 09:59:06 hermes npm[1029335]: Zigbee2MQTT:info  2022-10-23 09:59:06: MQTT publish: topic 'zigbee2mqtt/linky/active_enerfy_out_d02', payload '15211'
Oct 23 09:59:06 hermes npm[1029335]: Zigbee2MQTT:info  2022-10-23 09:59:06: MQTT publish: topic 'zigbee2mqtt/linky/active_enerfy_out_d03', payload '1927'
Oct 23 09:59:06 hermes npm[1029335]: Zigbee2MQTT:info  2022-10-23 09:59:06: MQTT publish: topic 'zigbee2mqtt/linky/active_enerfy_out_d04', payload '3839'
Oct 23 09:59:06 hermes npm[1029335]: Zigbee2MQTT:info  2022-10-23 09:59:06: MQTT publish: topic 'zigbee2mqtt/linky/status_register', payload '00DA4401'
Oct 23 09:59:06 hermes npm[1029335]: Zigbee2MQTT:info  2022-10-23 09:59:06: MQTT publish: topic 'zigbee2mqtt/linky/drawn_v_a_max_n1', payload '3422'
Oct 23 09:59:06 hermes npm[1029335]: Zigbee2MQTT:info  2022-10-23 09:59:06: MQTT publish: topic 'zigbee2mqtt/linky/message1', payload 'PAS DE MESSAGE'
Oct 23 09:59:06 hermes npm[1029335]: Zigbee2MQTT:info  2022-10-23 09:59:06: MQTT publish: topic 'zigbee2mqtt/linky/message2', payload ''
Oct 23 09:59:06 hermes npm[1029335]: Zigbee2MQTT:info  2022-10-23 09:59:06: MQTT publish: topic 'zigbee2mqtt/linky/active_power', payload '420'
Oct 23 09:59:06 hermes npm[1029335]: Zigbee2MQTT:info  2022-10-23 09:59:06: MQTT publish: topic 'zigbee2mqtt/linky/average_rms_voltage_meas_period', payload '237'
Oct 23 09:59:06 hermes npm[1029335]: Zigbee2MQTT:info  2022-10-23 09:59:06: MQTT publish: topic 'zigbee2mqtt/linky/active_power_ph_b', payload '426'
Oct 23 09:59:06 hermes npm[1029335]: Zigbee2MQTT:info  2022-10-23 09:59:06: MQTT publish: topic 'zigbee2mqtt/linky/days_number_current_calendar', payload '1'
Oct 23 09:59:06 hermes npm[1029335]: Zigbee2MQTT:info  2022-10-23 09:59:06: MQTT publish: topic 'zigbee2mqtt/linky/linkquality', payload '36'
Oct 23 09:59:06 hermes npm[1029335]: Zigbee2MQTT:info  2022-10-23 09:59:06: MQTT publish: topic 'zigbee2mqtt/linky/update-state', payload 'idle'
Oct 23 09:59:06 hermes npm[1029335]: Zigbee2MQTT:info  2022-10-23 09:59:06: MQTT publish: topic 'zigbee2mqtt/linky/current_tarif', payload 'HC SEM ET HC WE'
Oct 23 09:59:06 hermes npm[1029335]: Zigbee2MQTT:info  2022-10-23 09:59:06: MQTT publish: topic 'zigbee2mqtt/linky/current_price', payload 'HP WEEK END'
Oct 23 09:59:06 hermes npm[1029335]: Zigbee2MQTT:info  2022-10-23 09:59:06: MQTT publish: topic 'zigbee2mqtt/linky/current_index_tarif', payload '2'
Oct 23 09:59:06 hermes npm[1029335]: Zigbee2MQTT:info  2022-10-23 09:59:06: MQTT publish: topic 'zigbee2mqtt/linky/rms_voltage', payload '236'
Oct 23 09:59:06 hermes npm[1029335]: Zigbee2MQTT:info  2022-10-23 09:59:06: MQTT publish: topic 'zigbee2mqtt/linky/apparent_power', payload '494'
Oct 23 09:59:06 hermes npm[1029335]: Zigbee2MQTT:info  2022-10-23 09:59:06: MQTT publish: topic 'zigbee2mqtt/linky/current_summ_delivered', payload '26263'
Oct 23 09:59:06 hermes npm[1029335]: Zigbee2MQTT:info  2022-10-23 09:59:06: MQTT publish: topic 'zigbee2mqtt/linky/current_tier1_summ_delivered', payload '4533'
Oct 23 09:59:06 hermes npm[1029335]: Zigbee2MQTT:info  2022-10-23 09:59:06: MQTT publish: topic 'zigbee2mqtt/linky/current_tier2_summ_delivered', payload '14653'
Oct 23 09:59:06 hermes npm[1029335]: Zigbee2MQTT:info  2022-10-23 09:59:06: MQTT publish: topic 'zigbee2mqtt/linky/current_tier3_summ_delivered', payload '2680'
Oct 23 09:59:06 hermes npm[1029335]: Zigbee2MQTT:info  2022-10-23 09:59:06: MQTT publish: topic 'zigbee2mqtt/linky/current_date', payload 'E221023093456'
Oct 23 09:59:06 hermes npm[1029335]: Zigbee2MQTT:info  2022-10-23 09:59:06: MQTT publish: topic 'zigbee2mqtt/linky/active_enerfy_out_d02', payload '15211'
Oct 23 09:59:06 hermes npm[1029335]: Zigbee2MQTT:info  2022-10-23 09:59:06: MQTT publish: topic 'zigbee2mqtt/linky/active_enerfy_out_d03', payload '1927'
Oct 23 09:59:06 hermes npm[1029335]: Zigbee2MQTT:info  2022-10-23 09:59:06: MQTT publish: topic 'zigbee2mqtt/linky/active_enerfy_out_d04', payload '3839'
Oct 23 09:59:06 hermes npm[1029335]: Zigbee2MQTT:info  2022-10-23 09:59:06: MQTT publish: topic 'zigbee2mqtt/linky/status_register', payload '00DA4401'
Oct 23 09:59:06 hermes npm[1029335]: Zigbee2MQTT:info  2022-10-23 09:59:06: MQTT publish: topic 'zigbee2mqtt/linky/drawn_v_a_max_n1', payload '3422'
Oct 23 09:59:06 hermes npm[1029335]: Zigbee2MQTT:info  2022-10-23 09:59:06: MQTT publish: topic 'zigbee2mqtt/linky/message1', payload 'PAS DE MESSAGE'
Oct 23 09:59:06 hermes npm[1029335]: Zigbee2MQTT:info  2022-10-23 09:59:06: MQTT publish: topic 'zigbee2mqtt/linky/message2', payload ''
Oct 23 09:59:06 hermes npm[1029335]: Zigbee2MQTT:info  2022-10-23 09:59:06: MQTT publish: topic 'zigbee2mqtt/linky/active_power', payload '420'
Oct 23 09:59:06 hermes npm[1029335]: Zigbee2MQTT:info  2022-10-23 09:59:06: MQTT publish: topic 'zigbee2mqtt/linky/average_rms_voltage_meas_period', payload '237'
Oct 23 09:59:06 hermes npm[1029335]: Zigbee2MQTT:info  2022-10-23 09:59:06: MQTT publish: topic 'zigbee2mqtt/linky/active_power_ph_b', payload '426'
Oct 23 09:59:06 hermes npm[1029335]: Zigbee2MQTT:info  2022-10-23 09:59:06: MQTT publish: topic 'zigbee2mqtt/linky/days_number_current_calendar', payload '1'
Oct 23 09:59:06 hermes npm[1029335]: Zigbee2MQTT:info  2022-10-23 09:59:06: MQTT publish: topic 'zigbee2mqtt/linky/linkquality', payload '36'
Oct 23 09:59:06 hermes npm[1029335]: Zigbee2MQTT:info  2022-10-23 09:59:06: MQTT publish: topic 'zigbee2mqtt/linky/update-state', payload 'idle'
```

gauthier12, 2022-10-23:
Je réponds à une partie de mes questions : 
Déjà, les commandes TIC sont des commandes pour TIC historiques et non pas standard

gauthier12, 2022-10-23:
je discute tout seul :
J'ai changé le TIC white_liste pour 
NTARF,LTARF,EASF01,EASF02,EASF03,EASF04,EASD01,EASD02,EASD03,EASD04,EAST,SINSTS

Je ne vois pas de différence dans les log zigbee2mqtt par contre, je vois les valeurs exposés dans le frontend zigbee2mqtt
[image]

En fait, dans les données Enedis qu'on peut télécharger sur le site, je retrouve bien les mêmes valeurs sauf qu'il manque dans les données Zlinky les données EASF04 et EASD01 ? 
Est-ce que quelqu'un aurait une idée pour "débloquer" ces valeurs ?

gauthier12, 2022-10-23:
Bon, je n'ai rien changé est la valeur "EASF04" est apparue.... mystère...
Je ne comprends pas trop comment cela fonctionne

CladStriffe, 2022-10-26:
Oui j’avais eu ce genre de comportement en passant du mode historique à standard, il a fallut entre 1 et 2 jours pour que le zlinky décode correctement les infos.

gauthier12, 2022-10-28:
En fait, au bout de 48H ça s'est stabilisé...
C'est quand même bizarre

gauthier12, 2022-11-03:
Bonjour
Je reviens quand même sur la question. 
Si je ne mets rien dans tic_command_whitelist, je ne vois plus de message dans les log zigbee de valeur venant du Zlinky. Je pensais que par défaut, c'était tout.
Ensuite, je remplis l'option tic_command_whitelist avec une seule valeur, disons EAST
[image]
Pour être sur parce que je ne comprends pas bien, je reconfigure le zlinky, je redémarre zigbee2mqtt. Dans l'interface zigbee2mqtt, dans la section "exposes" du linky, je vois bien 1 seule valeur.
[image]
Par contre, si je regarde les log zigbee2mqtt, j'ai quand même toutes les valeurs et avec un fréquence qui ne correspond pas à celle que j'ai réglée (120 secondes) 
```
Nov 03 09:28:40 hermes npm[1016128]: Zigbee2MQTT:info  2022-11-03 09:28:40: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/options', payload '{"data":{"from":{"energy_phase":"auto","kWh_precision":3,"legacy":false,"linky_mode":"auto","measurement_poll_chunk":1,"measurement_poll_interval":120,"production":"auto","tarif":"auto","tic_command_whitelist":"NTARF,LTARF,EAST,EASF01,EASF02,EASF03,EASF04,SINSTS"},"id":"0x00158d0005ce7750","to":{"energy_phase":"auto","kWh_precision":3,"legacy":false,"linky_mode":"auto","measurement_poll_chunk":1,"measurement_poll_interval":120,"production":"auto","tarif":"auto","tic_command_whitelist":"EAST"}},"status":"ok","transaction":"n5r6a-14"}'
Nov 03 09:28:56 hermes npm[1016128]: Zigbee2MQTT:info  2022-11-03 09:28:56: MQTT publish: topic 'zigbee2mqtt/presence_chambre_grise/battery', payload '43.5'
Nov 03 09:28:56 hermes npm[1016128]: Zigbee2MQTT:info  2022-11-03 09:28:56: MQTT publish: topic 'zigbee2mqtt/presence_chambre_grise/occupancy', payload 'false'
Nov 03 09:28:56 hermes npm[1016128]: Zigbee2MQTT:info  2022-11-03 09:28:56: MQTT publish: topic 'zigbee2mqtt/presence_chambre_grise/temperature', payload '21.2'
Nov 03 09:28:56 hermes npm[1016128]: Zigbee2MQTT:info  2022-11-03 09:28:56: MQTT publish: topic 'zigbee2mqtt/presence_chambre_grise/occupancy_timeout', payload '0'
Nov 03 09:28:56 hermes npm[1016128]: Zigbee2MQTT:info  2022-11-03 09:28:56: MQTT publish: topic 'zigbee2mqtt/presence_chambre_grise/illuminance', payload '12184'
Nov 03 09:28:56 hermes npm[1016128]: Zigbee2MQTT:info  2022-11-03 09:28:56: MQTT publish: topic 'zigbee2mqtt/presence_chambre_grise/illuminance_lux', payload '17'
Nov 03 09:28:56 hermes npm[1016128]: Zigbee2MQTT:info  2022-11-03 09:28:56: MQTT publish: topic 'zigbee2mqtt/presence_chambre_grise/motion_sensitivity', payload 'high'
Nov 03 09:28:56 hermes npm[1016128]: Zigbee2MQTT:info  2022-11-03 09:28:56: MQTT publish: topic 'zigbee2mqtt/presence_chambre_grise/update-state', payload 'idle'
Nov 03 09:28:56 hermes npm[1016128]: Zigbee2MQTT:info  2022-11-03 09:28:56: MQTT publish: topic 'zigbee2mqtt/presence_chambre_grise/linkquality', payload '131'
Nov 03 09:29:05 hermes npm[1016128]: Zigbee2MQTT:info  2022-11-03 09:29:05: MQTT publish: topic 'zigbee2mqtt/linky/current_tarif', payload 'HC SEM ET HC WE'
Nov 03 09:29:05 hermes npm[1016128]: Zigbee2MQTT:info  2022-11-03 09:29:05: MQTT publish: topic 'zigbee2mqtt/linky/current_price', payload 'HP SEMAINE'
Nov 03 09:29:05 hermes npm[1016128]: Zigbee2MQTT:info  2022-11-03 09:29:05: MQTT publish: topic 'zigbee2mqtt/linky/current_index_tarif', payload '4'
Nov 03 09:29:05 hermes npm[1016128]: Zigbee2MQTT:info  2022-11-03 09:29:05: MQTT publish: topic 'zigbee2mqtt/linky/rms_voltage', payload '236'
Nov 03 09:29:05 hermes npm[1016128]: Zigbee2MQTT:info  2022-11-03 09:29:05: MQTT publish: topic 'zigbee2mqtt/linky/apparent_power', payload '499'
Nov 03 09:29:05 hermes npm[1016128]: Zigbee2MQTT:info  2022-11-03 09:29:05: MQTT publish: topic 'zigbee2mqtt/linky/current_summ_delivered', payload '26397.802'
Nov 03 09:29:05 hermes npm[1016128]: Zigbee2MQTT:info  2022-11-03 09:29:05: MQTT publish: topic 'zigbee2mqtt/linky/current_tier1_summ_delivered', payload '4543.262'
Nov 03 09:29:05 hermes npm[1016128]: Zigbee2MQTT:info  2022-11-03 09:29:05: MQTT publish: topic 'zigbee2mqtt/linky/current_tier2_summ_delivered', payload '14696.337'
Nov 03 09:29:05 hermes npm[1016128]: Zigbee2MQTT:info  2022-11-03 09:29:05: MQTT publish: topic 'zigbee2mqtt/linky/current_tier3_summ_delivered', payload '2706.892'
Nov 03 09:29:05 hermes npm[1016128]: Zigbee2MQTT:info  2022-11-03 09:29:05: MQTT publish: topic 'zigbee2mqtt/linky/current_date', payload 'E221023093456'
Nov 03 09:29:05 hermes npm[1016128]: Zigbee2MQTT:info  2022-11-03 09:29:05: MQTT publish: topic 'zigbee2mqtt/linky/active_enerfy_out_d02', payload '15287.781'
Nov 03 09:29:05 hermes npm[1016128]: Zigbee2MQTT:info  2022-11-03 09:29:05: MQTT publish: topic 'zigbee2mqtt/linky/active_enerfy_out_d03', payload '1936.377'
Nov 03 09:29:05 hermes npm[1016128]: Zigbee2MQTT:info  2022-11-03 09:29:05: MQTT publish: topic 'zigbee2mqtt/linky/active_enerfy_out_d04', payload '3859.786'
Nov 03 09:29:05 hermes npm[1016128]: Zigbee2MQTT:info  2022-11-03 09:29:05: MQTT publish: topic 'zigbee2mqtt/linky/status_register', payload '00DACC01'
Nov 03 09:29:05 hermes npm[1016128]: Zigbee2MQTT:info  2022-11-03 09:29:05: MQTT publish: topic 'zigbee2mqtt/linky/drawn_v_a_max_n1', payload '3422'
Nov 03 09:29:05 hermes npm[1016128]: Zigbee2MQTT:info  2022-11-03 09:29:05: MQTT publish: topic 'zigbee2mqtt/linky/message1', payload 'PAS DE MESSAGE'
Nov 03 09:29:05 hermes npm[1016128]: Zigbee2MQTT:info  2022-11-03 09:29:05: MQTT publish: topic 'zigbee2mqtt/linky/message2', payload ''
[... cut ...]
Nov 03 09:31:21 hermes npm[1017074]: Zigbee2MQTT:info  2022-11-03 09:31:21: MQTT publish: topic 'zigbee2mqtt/linky/current_tier4_summ_delivered', payload '4451.31'
Nov 03 09:31:21 hermes npm[1017074]: Zigbee2MQTT:info  2022-11-03 09:31:21: MQTT publish: topic 'zigbee2mqtt/linky/active_enerfy_out_d01', payload '5313.781'
Nov 03 09:31:40 hermes npm[1017074]: zlinky config: standard, single_phase, false, HC SEM ET HC WE
```
Est-ce qu'il y a quelque chose dans le fonctionnement que j'ai mal compris/fait ? Qu'est ce que je peux donner comme info en plus ?

CladStriffe, 2022-11-03:
Quelques éléments de réflexion:
- Essaye en écrivant all dans le TIC whitelist. Il me semble que par défaut ce champ est vide, mais quand on commence à écrire des choses dedans, si on efface tout il filtre tout. En remettant all chez moi tout est revenu. 
- De ce que j'ai compris, ce champ n'a pas d'impact sur les données envoyées par le lixee. Dans tous les cas, il envoie tout.
- Par contre, ça va filtrer les données qui vont être "suivies" par Z2m et donc sauvegardées par home assistant derrière. Comme le lixee envoie beaucoup d'infos, ça permet de ne pas surcharger la base de données avec des données dont on a pas besoin.
- Sur la fréquence de rafraîchissement des infos, j'ai jamais compris comment régler la durée. 
- Pour éviter les multiples envois sur une même seconde, j'ai mis un denounce a 1 ou 2 dans les paramètres sur Z2M.

gauthier12, 2022-11-04:
@CladStriffe merci pour tes explications.
En fait, concernant le tic white list, je comprends qu'il n'y a rien à comprendre. Les attributs exposés n'ont rien à voir avec les remontées.
Merci pour l'astuce de debounce, ça nettoie pas mal le traffic MQTT. J'ai complété avec un filtre sur les attributs dans z2m
Par contre, je me dis que ça doit pas mal encombrer le réseau zigbee non ?

gauthier12, 2022-11-04:
J'ai l'impression que même si toutes les valeurs sont envoyées par le ZLinky, il n'y a que celle qui sont dans la TIC white list qui sont mises à jour. Les autres sont envoyées avec la dernière valeur connue

CladStriffe, 2022-11-04:
Ah possible. J'ai dû repasser en all quelques jours suite à un changement de fournisseur et les soucis de V10, comme j'ai maintenant des données cohérentes je vais commencer à refiltrer ce qui m'intéresse et je vérifierai ce point.

JPV84, 2023-01-11:
Avec "ALL" voici ce que j'obtient
mais pour HCHP et HCHC il n'y a pas de valeur ( ainsi que d'autres champs )
pour info cela a fonctionné 
pb depuis une raz
[image]

fairecasoimeme, 2023-01-11:
vu le LQI, vous avez peut-être un problème de portée.
Vous pouvez aussi redémarrer Z2M et relancer une configuration du ZLinky (bouton orange dans Z2M à côté du device)

JPV84, 2023-01-11:
effectivement, j'ai rapproché  le RPI du module Lixee et cela fonctionne correctement maintenant
les infos remontent bien
comment améliorer cette portée