fairecasoimeme/Zlinky_TIC #122: some sensors are lost since v10 update

Albirew, 2022-10-27:
It looks like update to v10 (Firmware: 0x0000000a) made me loose a whole lot of sensors:
- LiXee ZLinky_TIC Electricalmeasurement (-32 768 W)
- LiXee ZLinky_TIC Electricalmeasurementapparentpower (65 535 VA)
- LiXee ZLinky_TIC Electricalmeasurementpowerfactor (unknown)
- lixee_zlinky_tic_infos_PTEC
- and more...

among these, what I use the most are Electricalmeasurementapparentpower and PTEC
PTEC automation is the following:
```yml
  - service: zha_toolkit.execute
    data:
      command: attr_read
      ieee: REDACTED
      cluster: 1794
      attribute: 32
      state_id: sensor.lixee_zlinky_tic_infos_PTEC
```
[image]
[image]

bdien, 2022-10-27:
Same issue with V10, SINSTS 65535 and SMAXN -32768.

Using zigbee2mqtt 1.28.0, was working before OTA.
Tried to remove the ZLinky and rejoin, without any improvement.

nbufnoir, 2022-10-28:
Same here,  SINSTS at 65535 and not updating since I went to v10

CladStriffe, 2022-10-28:
This issue has already been raised (see #89 )
According to the developer, in V10 the way SINSTS and other values are managed has changed to cover both single-phase and tri-phase. This requires an update of zigbee2mqtt (no date provided).

bdien, 2022-10-28:
I've modified the file from zigbee2mqtt (Latest rev 5011671) to make SINSTS and SMAXN work (SMAXN-1 is more difficult, being a ZLinky special attibute).
https://gist.github.com/bdien/260fea4b4f82295a2613f85184ee332f

To use it, put the content as `lixee.js`, next to `configuration.yaml`.
And add the following lines to `configuration.yaml`:
~~~
external_converters:
  - lixee.js
~~~

Only tested with the standard mode, single phase, zigbee2mqtt 1.28.0.

fairecasoimeme, 2022-10-28:
I'll try to fix this issue with a new firmware version, to be compatible with plugins.

pipiche38, 2022-10-29:
same issue of Domoticz

navycrow, 2022-10-30:
same issue. :(

fairecasoimeme, 2022-10-31:
You can try the new version to fix it
https://github.com/fairecasoimeme/Zlinky_TIC/releases/tag/v11.0

CladStriffe, 2022-10-31:
> Thanks !
> 
> I've got a weird behaviour from zigbee2mqtt OTA: It is detecting the new file v11 but telling me that no update is available.
> 
> Could it be related to the `fileVersion` (10) field from the metadata that it is receiving (i am not sure from which URL) ?
> 
> 
> 
> ~~~
> 
> Is new image available for '0x00158d0006a40287',
> 
>    current '{"fieldControl":0,"manufacturerCode":4151,"imageType":1,"fileVersion":10}',
> 
>    latest meta '{"fileVersion":10,"fileSize":244142,"url":"https://github.com/fairecasoimeme/Zlinky_TIC/releases/download/v11.0/ZLinky_router_limited_v11.ota"}'
> 
> ~~~

Same here

fairecasoimeme, 2022-10-31:
Sorry I'm just reupload

CladStriffe, 2022-10-31:
Hello, thank you for the update. 
However I cannot update to v11 using OTA in Z2m. 
I have an error "image type mismatch". 
It is maybe related to the fact that z2m is trying to use the router_limited version, even if it should use the regular one (I never manually updated the ZLinky to use a no-router version). 
Am I missing something ?

bdien, 2022-10-31:
Managed to update to 4000-0011 with Z2M, SINSTS and SMAXN are working (SMAXN-1 seems too low (410VA), will check tomorrow).

For Z2M update, there is indeed an issue, Z2M will select the first .ota file from the release. I "fixed" it by modifying the file `node_modules/zigbee-herdsman-converters/lib/ota/lixee.js` in my docker image within the following section:
~~~js

    // Find the most recent OTA file available
    for (const e of releasesLIST.sort((a, b) => a.published_at - a.published_at)) {
        if (e.assets) {
            const targetObj = e.assets
                .find((a) => a.name.endsWith('router_v11.ota')); // <-- Patched here (Ugly)
            if (targetObj && targetObj.browser_download_url) {
                firmURL = targetObj;
                break;
            }
        }
    }
~~~

fairecasoimeme, 2022-10-31:
I'll try to rename to put the "legacy" first

plouf34, 2022-11-01:
Hello
Is it possible to upgrade from v10 router OTA to v11 limited OTA via zigbee2mqtt OTA update?

Albirew, 2022-11-01:
Thanks, I got back almost all semsors
What still doesn't work are:
- those that never worked in first place (Electricalmeasurement and Electricalmeasurementpowerfactor)
- the one I'd like it to work (PTEC witch shows current tariff, if someone with sensor knowledge can help)

Albirew, 2022-11-01:
Ah, i switched from "historic" to "standard" recently, but kept the HP/HC tariff.
How do you keep watch of currently applied tariff in standard mode then?

Albirew, 2022-11-01:
Thanks for pointing me to the right direction: cluster ff66 attr 200.
I edited automation like this to keep track of history:
```yml      
  - service: zha_toolkit.execute
    data:
      command: attr_read
      ieee: REDACTED
      cluster: 65382
      attribute: 512
      state_id: sensor.lixee_zlinky_tic_infos_PTEC
```

I will close this issue since all the sensors i use are back.
The 2 remaining sensors were not working from the start so i don't miss them (^_^; )

renaiku, 2022-11-10:
> I've modified the file from zigbee2mqtt (Latest rev 5011671) to make SINSTS and SMAXN work (SMAXN-1 is more difficult, being a ZLinky special attibute). https://gist.github.com/bdien/260fea4b4f82295a2613f85184ee332f

@bdien is it PRed in z2m ?
In zlinky OTA V11 & z2m 1.28.2, my SINSTS is stuck at 438VA.

bdien, 2022-11-10:
> @bdien is it PRed in z2m ? In zlinky OTA V11 & z2m 1.28.2, my SINSTS is stuck at 438VA.

Not PRed as it is working on my side (Z2M 1.28.2 without any modification, ZLinky 4000-0011 (V11), Standard Mode)