fairecasoimeme/Zlinky_TIC #124: Demande feature: RE - Pairing automatique

renaiku, 2022-10-29:
Bonjour,

J'ai un cas particulier à vous soumettre.

Entre mon coordinateur et mon zlinky_tic il y a une cage d'ascenseur.
l'attribut linkquality est en général de mauvais (60) à plutot bon (+200).
Cela m'arrive en firmware "no route" 0.9 (cela m'arrivait aussi en firmware normal en 0.5), je me retrouve avec parfois le zlinky_tic qui est en "hors ligne" ou "time out" sur z2m.

Il suffit re pair le device avec z2m pour que tout revienne dans l'ordre.

Est il possible de rajouter la fonctionnalitée suivante: avec un ping (ou autre), le zlinkty_tic detecte que le coordinateur ne lui répond plus, et qu'il passe automatiquement en mode pairing ?

CladStriffe, 2022-10-29:
Quand tu vas sur le zlinky pour relancer l'appairage dans ces cas là, est-ce que la led bleue est allumée ?
Avec le mien je n'ai qu'un linkquality d'environ 30 et je n'ai jamais eu de déconnexion, enfin que je sache.

Donc il faudrait d'abord identifier si c'est bien un soucis de connexion, ou bien d'alimentation du Zlinky par le Linky (ce qu'est en train d'analyser le développeur, car ça a l'air variable selon le fabricant du Linky)

renaiku, 2022-10-29:
J'en ai profité pour passer le pollingbde zéro (par défaut dans z2m) à 10, et l'interval de 10s à 60s.

Si j'ai encore des soucis, je ferais bien attention à la couleur de la led et à la méthodologie.

renaiku, 2022-10-30:
@CladStriffe Je suis apssé en V10 hier dans la matinée (donc je suppose que même si j'étais en no route, je suis deretour sur le firware de base).

Le zlinky s'est déconnecté peux après mais j'ai pas eu le temps d'aller le reboot avant maintenant.

**La LED bleue est bien allumée**, donc ca n'a pas l'air d'être un problème d'alimentation.

renaiku, 2022-11-10:
Je suis passé en v11.

Ce matin encore le zlinky est déconnecté du réseau zigbee. Je mets z2m en mode appairage, je vais voir le device. La led est bleue, elle ne clignote pas. Je reste appuyé sur le bouton 10seconde. L'appareil se re appaire bien. Jusqu'à la prochaine fois.

renaiku, 2022-11-10:
Le zlinky s'est encore déconnecté. La led était bleue.

Je l'ai débranché du compteur et rebranché (sans activer le pairing de z2m) et le zlinky s'est instantanément remis en ligne dans z2m.

Le linky est un modèle Sagemcom

CladStriffe, 2022-11-10:
Il faudrait un retour du développeur alors. 
De toute façon, chez moi même si je débranche et rebranche le zlinky je n'ai pas besoin d'appairer de nouveau, Z2M le reconnaît. 
Par ailleurs, passer le zlinky en appairage automatique comme tu suggères ne résoudrait pas vraiment ton problème, car il faut aussi relancer l'appairage dans Z2M.

CladStriffe, 2022-11-10:
Je viens de voir la question [129](https://github.com/fairecasoimeme/Zlinky_TIC/issues/129)
Est-ce que ton Zlinky se connecte directement à ta clé Zigbee ? Ou est-ce qu'il se connecte à un appareil relai qui pourrait se déconnecter parfois ?

renaiku, 2022-11-10:
Pour le moment il a l'air connecté à la Philips hue la plus proche de lui, et elle ne passe jamais offline.
Je pourrais essayer de réappairer uniquement sur le coordinateur.

Les valeurs ne changent plus depuis que je suis passé en V11, je reçois plus rien dans les logs de z2m sur le zlinky. La led est bleue solide.

Dès que j'ai un peu de temps j'essaye de le débrancher quelques heures, et je repair direct sur le coordinateur.

fairecasoimeme, 2022-11-15:
En effet, après une mise à jour sur Z2M, je ne sais pas pourquoi encore, mais il faut supprimer et re appairer quand les données ne remontent plus.