fairecasoimeme/Zlinky_TIC #125: Récupération de la puissance active

navycrow, 2022-11-02:
Afin de récupérer la puissance active que le linky ne sait pas retourner nativement, je me base sur l'index totaliseur EAST.
J'ai fait un petit automatisation sous HA qui me calcul cela.
Afin d'affiner le calcul, je souhaite changer les valeurs de reporting minimum_report_interval, maximum_report_interval et reportable change par défaut (60/900/100) à une valeur plus faible (15/60/10).
Je parviens à changer la valeur mini et j'obtiens bien un rafraichissement de la valeur au minimum toutes les 15 s.
mais je ne peux pas changer la valeur max. Ca me retourne une erreur INVALID VALUE.
Concernant le reportable change, dés que je le modifie. Il n'y a plus aucune rafraichissement. Quelle valeur dois-je mettre pour avoir un rafraichissement dès que 10 watts sont ajoutés ?
[image]
[image]

navycrow, 2022-11-02:
effectivement 61 c'est ok. merci
Mon petit souci concernant le reportable_change était lié à une petite coquille dans mon script.
Tout fonctionne parfaitement maintenant. Merci

closed 2022-11-02 by navycrow

waako, 2023-02-14:
@navycrow Tu peux partager ton script HA pour récupérer la puissance active instantanée ?

navycrow, 2023-02-20:
Voici l'automatisation à associer avec une entrée numérique
Je précise qu'il ne s'agit pas d'un récupération mais d'un calcul entre 2 périodes. Il y a peut-être une meilleur solution mais cela reste, dans mon cas du moins, assez proche de la réalité.

`alias: "Linky : Calcul de la puissance active V2"
description: >-
  A chaque changement de valeur de l'index totaliseur (EAST), la puissance
  active moyenne est calculé sur la dernière période

  Mise à jour du 02/11/2022 :

  - suppression de la condition d'augmentation (il faut que ça s'actualise
  lorsque les panneaux solaires produisent et que l'index du compteur n'augmente
  pas)

  Mise à jour du 03/11/2022 :

  - affichage du mini entre la puissance active calculée et la puissance
  apparente ou entre la puissance active existante et la puissance apparente
  selon l'augmentation d'énergie, la puissance active ne devant pas être
  supérieure à la puissance apparente
trigger:
  - platform: state
    entity_id:
      - sensor.compteur_linky_east
condition: []
action:
  - service: notify.persistent_notification
    data:
      message: |
        trigger : {{ trigger }}
      title: Puisance active
    enabled: false
  - variables:
      apparent_power: |
        {{ trigger.to_state.attributes.apparent_power }}
      active_power_old: |
        {{ states("input_number.compteur_linky_puissance_active") }}
      energy_diff_wh: >
        {{ ((trigger.to_state.state|float - trigger.from_state.state|float) *
        1000)|int }}
      time_diff_s: >
        {{ (as_timestamp(trigger.to_state.last_changed) -
        as_timestamp(trigger.from_state.last_changed))|round(3) }}
      active_power_new: |
        {% if energy_diff_wh > 0 %}
          {{ min(apparent_power,(energy_diff_wh * 3600 / time_diff_s)|int) }}
        {% else %}
          {{ min(apparent_power, active_power_old) }}
        {% endif %}
  - service: input_number.set_value
    data:
      value: |
        {{ active_power_new|int }}
    target:
      entity_id: input_number.compteur_linky_puissance_active
    enabled: true
  - service: notify.persistent_notification
    data:
      message: >
        apparent_power : {{ apparent_power }} / active_power_old : {{
        active_power_old }} / energy_diff_wh : {{energy_diff_wh}} / time_diff_s
        : {{time_diff_s}} / active_power_new : {{ active_power_new }}
      title: Puisance active
    enabled: false
mode: single
`