fairecasoimeme/Zlinky_TIC #126: How to rollback the firmware ?

Dephi78, 2022-11-03:
Hello,

J'ai intégré un ZLinky TIC (1.4) sous Home Assistant 
HA : 2022.11.1
via une TI CC2531 USB en ZHA

A reception, j'ai updaté le firmware en version 5, j'avais des remontées d'infos partiellement correctes.
J'ai décidé de migrer vers la version 10.
Firmware updaté avec succès, depuis j'ai des valeurs totalement incohérentes

Je n'ai pas trouvé la procédure pour rollbacker en version 5 par exemple

J'ai mis cette conf, mais je ne vois aucun update de firmware dans les logs
(le fichier ZLinky_router_v5.0.ota a été placé dans le repertoire /config/zigpy_ota)

zha:
  zigpy_config:
    ota:
      otau_directory: /config/zigpy_ota
      ikea_provider: false
      ledvance_provider: false
      force: true
      salus_provider: false

Merci d'avance pour l'aide :)

Dephi

PEGARifrouf, 2022-11-07:
Bonjour,

J'ai réussi à rollback depuis la v9 vers la v5 avec ce post :
https://github.com/fairecasoimeme/Zlinky_TIC/issues/89#issuecomment-1236422701

Bonne manip !

plouf34, 2022-11-07:
Toujours aucun moyen de passer d'un firmware V10 ou V11 routed à une version non routed (version legacy) sans devoir passer par le mode flash vua USB TTL?

HermesHonshappo, 2022-11-14:
Pareil. J'ai upgradé en v9 no-router en utilisant l'OTA, mais n'ayant pas le programmateur USB, je suis maintenant bloqué

Si j'ai bien compris les messages, il n'y a plus d'upgrade OTA pour ces releases, et ce n'était pas annoncé lorsque j'ai fait l'upgrade en v9 no-router.

J'avoue que ça m'ennuie profondément.