fairecasoimeme/Zlinky_TIC #128: Seulement 3 commandes

Naglefourbe, 2022-11-09:
Hello,
ZlinkyTIC à jour, sur une Zigate, sur un Jeedom avec plugin Zigate.
Je n'ai que 4 commandes de créées : manufacturer, refresh, type, current_delivred. La dernière commande affiche 1, pas de bouton de test dans les paramètres.
Est-ce que c'est une limite de la clé Zigate, du plugin ? J'ai retenté plusieurs fois l'inclusion avec le même résultat.

fairecasoimeme, 2022-11-15:
le ZLinky n'est pas compatible avec le plugin ZiGate (je ne suis pas le créateur du plugin). Sous jeedom, il faut utiliser le plugin zigbee ou le plugin Abeille ou encore zigbeelinker (z2m)

Naglefourbe, 2022-11-15:
how merci je comprends mieux :)
j'ai tenté le plugin Zigbee, mais la Zigate n'est pas reconnue je cumule beaucoup d'erreurs de port, 'fin bref retour sous le plugin Zigate c'est ce qui marche le mieux. Sauf le Zlinky T_T