fairecasoimeme/Zlinky_TIC #129: Reconstruction du réseau

bemble, 2022-11-10:
Bonjour,

J'ai l'impression que mon Zlinky souffre des même soucis que les devices Xioami/Aqara : il reste connecté au premier device sur lequel il est connecté et si celui-ci est offline, il ne cherche pas d'autre point pour se connecter. Et dans mon cas c'est dommage, il s'est connecté à des ampoules Hue qui ne sont pas constamment allumées (dommage, il y a juste à côté de lui des modules pour mes volets, mais non, il a choisi de passer par ces lumières).

Voilà un partie du schéma de mon réseau :
[image]

 Et ce matin, sans surprise :
[image]
[image]

Une idée ? Est-ce un bug ou moi qui ait mal configuré quelque chose ?

Firmware : v10 limited
Clé Zigbee : Conbee II
Soft : Zigbee2mqtt

fairecasoimeme, 2022-11-15:
Je pense qu'il est d'abord important de ne pas avoir des appareils Zigbee qui ne sont pas constamment alimentés car cela va provoquer une instabilité de votre réseau zigbee. Ensuite, c'est le réseau zigbee qui choisit quel est le meilleur parent d'un appareil selon des critères qui ne sont pas accessibles (couches basses zigbee). 
Quand le ZLinky perd le lien avec le coordinateur, il fait une demande "broadcast" pour trouver un nouveau chemin (s'il existe ou s'il y arrive). Je pense donc que vous avez un souci avec votre réseau zigbee que vous devez régler

bemble, 2022-11-17:
Merci de votre réponse. Pour l'instant j'ai gardé les ampoules constamment alimentées, avant de changer de manière de faire.

closed 2022-11-17 by bemble