fairecasoimeme/Zlinky_TIC #131: Jeedom OTA v5 to v11

ovander, 2022-11-12:
Bonjour à tous,
Merci d’avance pour votre aide.
J’ai un Lixee ZLinky en version 5, ZCL version 3, APP Version 5, Stack version 2, HW version 1, Date code: 20220217, Software version 4000-0005.
Jeedom: 4.3.10, zigbee 2022-10-20 01:03:41
Je pensais naïvement que je pouvais facilement mettre à jour le firmware vers la dernière version disponible en utilisant la procédure manuel du plugin Zigbee. J’imagine que j’ai loupé une étape car ça ne marche pas.
Ce que j’ai fait:

activé autoriser les mises à jour Over the air (OTA) dans la config du plugin zigbee.
Mis à jour les fichiers de modules
Relancé le daemon du plugin zigbee
inspecté les fichiers OTA dans le directory plugins/zigbee/data/ota ou je trouve la version v5 (ZLinky_router_v5.ota)
Ajout du fichier ZLinky_router_v11.ota trouvé sur le GitHub de ZLinky_TIC [Release ZLinky_TIC Router v11.0 · fairecasoimeme/Zlinky_TIC · GitHub](https://github.com/fairecasoimeme/Zlinky_TIC/releases/tag/v11.0)
Relancé le daemon du plugin zigbee
Dans la config du module sous l’onglet action, je force la mise à jour du module.
Pas de mise à jour.
Dans les log(zigbee_1), je trouve:

[2022-11-11 22:23:02][DEBUG] : [0x010F:1:0x0019] No OTA image is available

Je conclu que le fichier n’est pas dans le bon directory ou qu’il n’a pas le bon format (nom, contenu)?
J’ai essayé des variations: laisser que le fichier v11.0 en supprimant le v5.0
Avez-vous des suggestions ?

Merci,
Cordialement,
Olivier

mospaeda, 2022-11-12:
Bonjour 
Je ne sais pas si sur Jeedom, la MAJ via OTA fonctionne mais sur HA via zigate+ ça ne marche pas
Si tu as une Zigate c'est peut être la source 
C'est connu apparemment de @fairecasoimeme 
https://github.com/fairecasoimeme/Zlinky_TIC/issues/107#issuecomment-1272861384

ovander, 2022-11-13:
Bonjour,
J'ai en effet omis de spécifier que le dongle Zigbee que j'utilise est un POPP.
J'avais lu #107 mais sans trouver d'éléments qui pouvait m'aider.

fairecasoimeme, 2022-11-15:
Je pense que vous aurez plus de réponses si vous demandiez sur le forum de Jeedom

ovander, 2022-11-15:
Merci pour votre retour, j'ai posté aussi ce cas sur le forum de Jeedom. Sans réponse pour l'instant.

closed 2022-12-18 by ovander