fairecasoimeme/Zlinky_TIC #132: Production PV

chryss44, 2022-11-13:
Bonjour, 
Je suis l'heureux possesseur d'un ZLinky_TIC et je suis en mode production avec un CRAE mais je n'ai aucune info qui remonte.
J'utilise Zigbee2MQTT et j'ai fait le paramètrage pour être en mode Standard afin de voir remonter la production mais la valeurs EAIT reste à zéro alors que j'ai quelque-chose comme 1700Kwh sur le compteur en injecté.
Pareil, j'ai aussi un Wifinfo et ca m'a permis de constater que lorsque je produit et injecte tout en couvrant ma consommation, le wifinfo affiche bien 0 en consommation (ce qui est normal car je ne consomme rien) mais que côté TIC il ne descend pas à zéro mais reste à la dernière valeur (bizarre).
Donc quand j'ai 500W d'instantané, j'ai bien le TIC et le Wifinfo qui remonte la valeur et dès que le soleil donne, le Wifinfo tombe à 0 alors que le TIC reste bloqué sur la dernière valeur à 500. Du coup quand le soleil se cache et que j'ai par exemple un instantané qui remonte a 750W, le Wifinfo remonte bien de 0 à 750W alors que le TIC passe de 500W à 750W.
C'est un pli peu embêtant et pas très fiable du coup, est-ce que ca vient de mon paramètrage ?
En image Linky -->Wifinfo et Linky 2 -->TIC
[image]
Pour le paramétrage du TIC
[image]
Merci pour vos lumières

fairecasoimeme, 2022-11-15:
Pour voir si vous etes bien en mode production, il faut récupérer la valeur STGE ? cluster 0xff66 attribut 0x217 ?
[image]

chryss44, 2022-11-15:
J'ai ça : 
2022-11-15 21:05:14Read result of 'liXeePrivate': {"statusRegister":{"type":"Buffer","data":[48,48,68,65,52,49,48,49]}}
Mais franchement, je sais pas lire les données.

chryss44, 2022-11-16:
Semble t'il oui
Date du firmware : 20221031
Version du firmware : 4000-0011

fairecasoimeme, 2022-11-16:
Je parlais de la version de Z2M, pas du firmware du zlinky.

chryss44, 2022-11-16:
Ah oui, pardon, lecture en diagonal....
La version de Z2M est [1.28.0](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.28.0) commit: [03ba647d].

fairecasoimeme, 2022-11-17:
il faut passer semble-t-il sur la dernière : [1.28.2](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.28.2)

c'est corrigé depuis la 1.28.1 : https://github.com/Koenkk/zigbee2mqtt/issues/13991

chryss44, 2022-11-19:
Salut, passage en 1.28.2 et je récupère bien 0 quand la production est en route.
Par contre, surprise hier soir, il a fallut que je débranche et rebranche le module car la led était éteinte et il ne répondait n'y n'envoyait rien. C'est reparti mais à surveiller.
Merci en tout cas, c'est nickel

fairecasoimeme, 2022-11-21:
Bonjour, lorsque vous n'avez plus de remontée, pouvez-vous faire une requête sous Z2M comme sur le screenshot : 

[image]

chryss44, 2022-11-21:
The result
`Error 2022-11-21 20:43:34Publish 'set' 'read' to '0x00158d0005ce77f1' failed: 'Error: Read 0x00158d0005ce77f1/1 genBasic(["zclVersion"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received)'`

technarf, 2022-11-21:
Alors de mon côté le résultat est très similaire, seul le motif du failed semble différent chez moi (le test est fait alors que le Zlinky est injoignable) : 
`Error 2022-11-21 21:02:05Publish 'set' 'read' to 'Zlinky' failed: 'Error: Read 0x00158d0005ce76c6/1 genBasic(["zclVersion"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'MAC no ack' (233))'`

chryss44, 2022-11-21:
Après débranchement et re branchement, voila le retour
`Info 2022-11-21 21:08:07Read result of 'genBasic': {"zclVersion":3}`

fairecasoimeme, 2022-11-24:
Étrange, pouvez-vous passer sur la version 12 ?

chryss44, 2022-11-25:
Bonjour, depuis le passage en version 12, plus rien, j'ai ré appairé le TIC mais rien, il n'y a plus rien qui remonte.
C'est décidément très compliqué, que souhaitez-vous que je fasse, qu'est ce qui peut vous aider a diagnostiquer le problème ?

fairecasoimeme, 2022-11-25:
Bonjour,
C'est un problème qui apparait souvent sur Z2M et que je ne maitrise pas. La plupart du temps après une mise à jour du ZLinky sur Z2M, il faut supprimer avec le forçage de la suppression, puis re-appairer. vous pouvez aussi relancer la configuration. 
[image]

Vous pouvez aussi tenter de relancer Z2M.

antoinelibert, 2022-11-25:
Bonjour,

Je confirme que le passage en V12 a nécessité de nouveau un reset complet du ZLinky comme indiqué au dessus. Bizarrement, certaines mise à jour nécessite un reset, et d'autre, pas du tout....

chryss44, 2022-11-25:
Désolé, mais ca ne marche pas chez moi, j'ai bien la trace d'appairage mais après plus rien `Accepting joining not in blocklist device '0x00158d0005ce77f1'`

antoinelibert, 2022-11-26:
Essayer de débrancher/rebrancher si ça coince juste après l'apparaillage (notamment si Z2M indique "unknown device").

chryss44, 2022-11-26:
J'ai tout essayé, le supprimer et le ré appairer, le débrancher et le rebrancher, rien n'y fait, il n'y a que 'appairage qui fonctionne et après plus rien. je consolide le log et je vous l'envoie mais c'est pas réussite. Il y a moyen de faire un retour arrière en V11.

WarC0zes, 2022-11-27:
> Bonjour,
> 
> Je confirme que le passage en V12 a nécessité de nouveau un reset complet du ZLinky comme indiqué au dessus. Bizarrement, certaines mise à jour nécessite un reset, et d'autre, pas du tout....

Je confirme, même sous ZHA. Suivant les maj de firmware; il faut supprimer et ré-appairer le Zlinky pour que ca fonctionne bien. Ce que j'ai fait au passage de la v12.

chryss44, 2022-11-27:
Bonjour,,

Voci le log à partir du moment ou j'ai supprimé de Z2M, fait un reset du PIC puis une fois appairé débrancher puis rebrancher pour voir si il y avait un changement.
Je n'ai plus de message en provenance du TIC derrière la dernière ligne de log.
`info  2022-11-26 18:00:16: Removing device '0x00158d0005ce77f1' (block: false, force: true)
info  2022-11-26 18:00:16: MQTT publish: topic 'zigbee2mqtt/0x00158d0005ce77f1', payload ''
info  2022-11-26 18:00:16: Successfully removed device '0x00158d0005ce77f1' (block: false, force: true)
info  2022-11-26 18:00:16: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/remove', payload '{"data":{"block":false,"force":true,"id":"0x00158d0005ce77f1"},"status":"ok","transaction":"6jd7g-9"}'
info  2022-11-26 18:00:35: Accepting joining not in blocklist device '0x00158d0005ce77f1'
info  2022-11-26 18:00:35: Device '0x00158d0005ce77f1' joined
info  2022-11-26 18:00:35: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x00158d0005ce77f1","ieee_address":"0x00158d0005ce77f1"},"type":"device_joined"}'
info  2022-11-26 18:00:35: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0x00158d0005ce77f1"},"type":"device_connected"}'
info  2022-11-26 18:00:35: Starting interview of '0x00158d0005ce77f1'
info  2022-11-26 18:00:35: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x00158d0005ce77f1","ieee_address":"0x00158d0005ce77f1","status":"started"},"type":"device_interview"}'
info  2022-11-26 18:00:35: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x00158d0005ce77f1"},"type":"pairing"}'
info  2022-11-26 18:00:35: Successfully interviewed '0x00158d0005ce77f1', device has successfully been paired
info  2022-11-26 18:00:35: Device '0x00158d0005ce77f1' is supported, identified as: LiXee Lixee ZLinky (ZLinky_TIC)
info  2022-11-26 18:00:36: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"Lixee ZLinky","exposes":[{"access":1,"description":"Serial Number","name":"ADCO","property":"meter_serial_number","type":"text"},{"access":1,"description":"Tarif option","name":"OPTARIF","property":"current_tarif","type":"text"},{"access":1,"description":"Subscribed intensity level","name":"ISOUSC","property":"available_power","type":"numeric","unit":"A"},{"access":1,"description":"BBRHCJB index","name":"BBRHCJB","property":"current_tier1_summ_delivered","type":"numeric","unit":"kWh"},{"access":1,"description":"BBRHPJB index","name":"BBRHPJB","property":"current_tier2_summ_delivered","type":"numeric","unit":"kWh"},{"access":1,"description":"BBRHCJW index","name":"BBRHCJW","property":"current_tier3_summ_delivered","type":"numeric","unit":"kWh"},{"access":1,"description":"BBRHPJW index","name":"BBRHPJW","property":"current_tier4_summ_delivered","type":"numeric","unit":"kWh"},{"access":1,"description":"BBRHCJR index","name":"BBRHCJR","property":"current_tier5_summ_delivered","type":"numeric","unit":"kWh"},{"access":1,"description":"BBRHPJR index","name":"BBRHPJR","property":"current_tier6_summ_delivered","type":"numeric","unit":"kWh"},{"access":1,"description":"RMS current","name":"IINST","property":"rms_current","type":"numeric","unit":"A"},{"access":1,"description":"RMS current peak","name":"IMAX","property":"rms_current_max","type":"numeric","unit":"A"},{"access":1,"description":"Apparent power","name":"PAPP","property":"apparent_power","type":"numeric","unit":"VA"},{"access":1,"description":"Current pricing period","name":"PTEC","property":"active_register_tier_delivered","type":"text"},{"access":1,"description":"Tomorrow color","name":"DEMAIN","property":"tomorrow_color","type":"text"},{"access":1,"description":"Schedule HPHC","name":"HHPHC","property":"schedule_h_p_h_c","type":"numeric"},{"access":1,"description":"Subscribed Power Exceeded Warning","name":"ADPS","property":"warn_d_p_s","type":"numeric","unit":"A"},{"access":1,"description":"Link quality (signal strength)","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"ZLinky_TIC","options":[{"access":2,"description":"This device does not support reporting electric measurements so it is polled instead. The default poll interval is 60 seconds, set to -1 to disable.","name":"measurement_poll_interval","property":"measurement_poll_interval","type":"numeric","value_min":-1},{"access":2,"description":"Counter with TIC in mode standard or historique. May require restart (default: auto)","name":"linky_mode","property":"linky_mode","type":"enum","values":["auto","historique","standard"]},{"access":2,"description":"Power with single or three phase. May require restart (default: auto)","name":"energy_phase","property":"energy_phase","type":"enum","values":["auto","single_phase","three_phase"]},{"access":2,"description":"If you produce energy back to the grid (works ONLY when linky_mode: standard, default: auto)","name":"production","property":"production","type":"enum","values":["auto","true","false"]},{"access":2,"description":"Overrides the automatic current tarif. This option will exclude unnecesary attributes. Open a issue to support more of them. Default: auto","name":"tarif","property":"tarif","type":"enum","values":["Historique - BASE","Historique - HCHP","Historique - EJP","Historique - BBR","Standard - Sem WE Mercredi","Standard - BASE","Standard - Heure  [... cut ...]
debug 2022-11-26 18:01:54: Received MQTT message on 'zigbee2mqtt/bridge/request/device/options' with data '{"id":"0x00158d0005ce77f1","options":{"production":"true"},"transaction":"6jd7g-13"}'
info  2022-11-26 18:01:54: Changed config for device 0x00158d0005ce77f1
info  2022-11-26 18:01:55: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/options', payload '{"data":{"from":{"energy_phase":"single_phase","linky_mode":"standard"},"id":"0x00158d0005ce77f1","to":{"energy_phase":"single_phase","linky_mode":"standard","production":"true"}},"status":"ok","transaction":"6jd7g-13"}'`

chryss44, 2022-11-27:
Bon alors, sans vraiment le vouloir, le problème est résolu (pour le moment) car j'ai redémarrer Z2M pour justement forcer l'OTA du TIC en V11 avec un fichier index, mais du coup pas besoin car d'un coup ca s'est mis a discuter....pas bien compris mais je vais surveiller.
Donc si je comprends, il faut supprimer puis ré appairer le TIC mais aussi redémarrer Z2M pour que ca reparte....faire gaffe on est pas loin de Microsoft et du redémarrer pour la prise en comte :-) .

chryss44, 2023-01-03:
Bonjour,
Me concernant, à part le fait que ca cause beaucoup, c'est bon, je récupère bien ma production.
Merci

fairecasoimeme, 2023-01-12:
bonjour, tu as la réponse ici : https://github.com/fairecasoimeme/Zlinky_TIC/issues/78#issuecomment-1221533698

MichaelBitard, 2024-07-14:
Bonjour, j'ai un souci similaire, production photovoltaïque avec EDF-OA (contrat actif depuis plus d'un an, j'ai la remontée sur le compteur linky, tout ça tout ça) mais j'ai toujours 0 sur EAIT (avec Zigbee2MQTT).

J'ai tenté ce que vous aviez mis là https://github.com/fairecasoimeme/Zlinky_TIC/issues/132#issuecomment-1314981246 et j'ai comme résultat `2024-07-14 19:01:15Read result of 'liXeePrivate': {"statusRegister":{"type":"Buffer","data":[0,0,0,0,0,0,0,0]}}`, étrange non ? 

Si je tente d'exécuter la commande 0x217 sur le cluster 0xff66 j'ai une erreur me disant que LixeePrivate ne supporte pas la commande.

Tout fonctionne, sauf la remontée de la production ( j'ai bien configuré le linky en mode standard, single phase et production à true )