fairecasoimeme/Zlinky_TIC #133: Récupération CCASN retourne -32768

HermesHonshappo, 2022-11-14:
Bonjour

Je cherche à comprendre pourquoi, aussi bien sur ZHA que Z2M, le Lixee ne remonte pas le champ CCASN (active power) correctement? Chaque fois, il lit la valeur -32768

Exemple sur Z2M, avec la dernière version du firmware:

[image]

chryss44, 2022-11-14:
Si j'ai tout bien compris, tu es paramètre en Standard pour ton tic et ton compteur est en historique, j'ai ce problème aussi de 32768.

chryss44, 2022-11-14:
Non non, je pense pas que ce soit bancal mais j'ai le même symptôme que toi sauf qu'il semble que ce soit ce que j'ai décrit le problème....me suis dit bêtement que c'était pareil pour toii.
t'as essayer en inversant le paramétrage côté Zinky en mode standard pour voir si tu as le même résultat ?

fairecasoimeme, 2022-11-15:
Attention le mode du Linky (historique ou standard) n'est pas paramétrable au travers du ZLinky. C'est Enedis qui paramètre le Linky selon l'ancienneté ou la demande de l'habitant.

Avant de faire le paramétrage sur Z2M, il faut d'abord savoir dans quel mode est votre Linky. (en faisant défilé sur l'écran du Linky par exemple)

Si vous êtes en mode historique, l'information CCASN n'existe pas. Ce qui explique la valeur par défaut. pour connaitre les étiquettes valident, il faut consulter le [README
](https://github.com/fairecasoimeme/Zlinky_TIC#mode-historique)

chryss44, 2022-11-15:
Je viens de faire la demande en passant par Ennemis et ca ne pose aucun problème, il faut juste demander à basculer le TIC d'historique à Standard. La personne a qui tu vas faire la demande remplie un formulaire et il y a un batch qui bascule ton compteur à minuit en mode souhaité.

HermesHonshappo, 2022-11-16:
Tu as eu plus de bol que moi. La personne que j'ai eu chez Enedis n'a rien voulu faire si ça ne passait pas par mon fournsseur d'énergie avant.
Peu-être que ça dépend des accords entre Enedis et le fournisseur.

HermesHonshappo, 2022-11-17:
Je confirme qu'une fois le Linky basculé en mode standard, on récupère bien le CCASN

[image]

BRSS73, 2022-11-19:
[image]

Pas toujours malheureusement

HermesHonshappo, 2022-11-19:
Etrange. Si tu vas sur ton compteur Linky, et tu appuie sur la flèche vers le bas jusqu'à ce qu'il affiche TIC, il dit quoi? Historique ou Standard?

BRSS73, 2022-11-19:
Mon compteur est en mode Standard et mon Zlinky est forcé sur les paramètres (standard, single_phase, false, etc.) pour limiter les erreurs d'interprétation.

Je ne suis globalement pas satisfait du Zlinky. J'ai de fréquentes déconnexions, il produit un gros bavardage sur le réseau (firmware v11 "limited") en envoyant 5-6 messages en rafale avec juste DATE qui change et de temps en temps, il flood mon coordinateur au point que je suis obligé de le retirer de force du réseau et de refaire l'appairage.

Aujourd'hui, je tente le Debounce sur Z2M pour réduire le flood sur MQTT.

fairecasoimeme, 2022-11-19:
> J'ai de fréquentes déconnexions, il produit un gros bavardage sur le réseau (firmware v11 "limited") en envoyant 5-6 messages en rafale avec juste DATE qui change et de temps en temps, il flood mon coordinateur au point que je suis obligé de le retirer de force du réseau et de refaire l'appairage.

ça n'a rien à  voir avec le zlinky. C'est selon comment votre plugin est paramétré. Si vous "poll" le Zlinky réduisez le polling, si vous avez créé des rapports, vérifiez les paramètres de vos rapports pour ne pas flood...

BRSS73, 2022-11-19:
Poll à -1.
Le reporting avec les valeurs par défaut.

fairecasoimeme, 2022-11-19:
Le ZLinky n'envoie rien de lui même. S'il envoie 6 messages DATE en rafale, c'est qu'on lui a demandé

BRSS73, 2022-11-19:
Alors je vais revoir encore une fois le paramétrage dans Z2M.

closed 2023-01-03 by HermesHonshappo