fairecasoimeme/Zlinky_TIC #135: Rapprocher le module de la box domotique

fulljackz, 2022-11-20:
Bonjour,

Ceci est plus une question d'usage qu'un bug.

Mon linky est situé à environ 30m du garage ou mon tableau est installé. 
Mon serveur home assistant se situe quant à lui dans la maison.

J'ai passé un cable rj45 cat6 et je souhaite utiliser une des paires pour la téléinformation Linky jusqu'au garage.
Je souhaiterais à l'extrémité de cette paire, savoir s'il est possible de souder ou de démonter le module pour y brancher la paire branchée à l'autre bout au linky ?

Je souhaite avoir l'information par le cable jusqu'au garage pour prévenir la future installation d'une borne de recharge pour mon ve, ainsi je pourrais utiliser la téléinformation en direct sur la borne, et aussi récupérer ma consommation instantanée sur ma domotique.

Merci pour vos avis et réponses.

fairecasoimeme, 2022-11-22:
Bonjour,
Voici un exemple d'un client : 
[image]

Cependant, je ne sais pas du tout si sur 30m ça fonctionne
Fred

fulljackz, 2022-11-22:
Excellent, merci pour la réponse !

D'après mon électricien, la distance n'est, à priori, pas un problème.
Il faudrait que je trouve un système type wago pour aller sur ma paire en RJ45.

fairecasoimeme, 2022-11-22:
pas un problème pour la partie datas, pour l'autoconsommation, je ne saurai pas vous dire ...

Fred

fulljackz, 2022-12-23:
Bonjour,

J'ai fais un test au linky, le module s'allume bien. J'ai tenté un appairage depuis l'extérieur sans succès.
J'ai dans la foulée effectué un essai de connexion sur 3 brins du rj45 tiré jusqu'au compteur, sans succès. 

Je pense donc que la distance est un problème pour l'alimentation. Auriez vous des idées sur comment alimenter le module depuis une autre source que le linky ?
Est il possible de commander l'antenne à part ?

Merci.

fairecasoimeme, 2022-12-27:
Bonjour,
Merci pour le retour. il est possible d'alimenter le ZLinky en 3.3v avec les PIN de programmation.
On ne peut pas commander l'antenne à part car la version avec antenne nécessite des modifications matériels.
Cependant, on peut faire un échange. il suffit de me contacter par mail.