fairecasoimeme/Zlinky_TIC #136: Zigbee2MQTT flood

BRSS73, 2022-11-21:
Après quelques jours ou parfois quelques heures, Zibee2MQTT ne répond quasiment plus (utilisation CPU à 98%) et la log présente les traces suivantes :

`info  2022-11-21 19:58:07: MQTT publish: topic 'zigbee2mqtt/ZLinky', payload '{"active_enerfy_out_d01":4102,"active_power":-32768,"active_power_max":410,"active_power_ph_b":-32768,"apparent_power":1670,"available_power":6,"average_rms_voltage_meas_period":236,"current_date":"H221121195754","current_index_tarif":1,"current_price":"BASE","current_summ_delivered":4096,"current_tarif":"BASE","current_tier1_summ_delivered":4096,"current_tier2_summ_delivered":0,"device":{"applicationVersion":11,"dateCode":"20221031","friendlyName":"ZLinky","hardwareVersion":1,"ieeeAddr":"0x00158d0005be9744","manufacturerID":4151,"manufacturerName":"LiXee","model":"ZLinky_TIC","networkAddress":62437,"powerSource":"Mains (single phase)","softwareBuildID":"4001-0011","stackVersion":2,"type":"Router","zclVersion":3},"drawn_v_a_max_n1":1960,"last_seen":"2022-11-21T18:58:06.560Z","linkquality":120,"message1":"PAS DE MESSAGE","message2":"","meter_serial_number":"812064634057","power_threshold":6,"relais":null,"rms_current":2,"rms_current_max":65535,"rms_voltage":236,"site_id":"21337337086612","software_revision":2,"status_register":"003A0001","update":{"state":"available"},"update_available":true}'
debug 2022-11-21 19:58:07: Received Zigbee message from 'ZLinky', type 'readResponse', cluster 'liXeePrivate', data '{"currentTarif":"      BASE\u0000     "}' from endpoint 1 with groupID 0
debug 2022-11-21 19:58:07: Received Zigbee message from 'ZLinky', type 'readResponse', cluster 'liXeePrivate', data '{"currentTarif":"      BASE\u0000     "}' from endpoint 1 with groupID 0
debug 2022-11-21 19:58:07: Received Zigbee message from 'ZLinky', type 'readResponse', cluster 'liXeePrivate', data '{"currentTarif":"      BASE\u0000     "}' from endpoint 1 with groupID 0
debug 2022-11-21 19:58:07: Received Zigbee message from 'ZLinky', type 'readResponse', cluster 'liXeePrivate', data '{"currentPrice":"      BASE\u0000     "}' from endpoint 1 with groupID 0
debug 2022-11-21 19:58:07: Received Zigbee message from 'ZLinky', type 'readResponse', cluster 'liXeePrivate', data '{"currentTarif":"      BASE\u0000     "}' from endpoint 1 with groupID 0
debug 2022-11-21 19:58:08: Received Zigbee message from 'ZLinky', type 'readResponse', cluster 'liXeePrivate', data '{"currentTarif":"      BASE\u0000     "}' from endpoint 1 with groupID 0
debug 2022-11-21 19:58:08: Received Zigbee message from 'ZLinky', type 'readResponse', cluster 'liXeePrivate', data '{"currentTarif":"      BASE\u0000     "}' from endpoint 1 with groupID 0
debug 2022-11-21 19:58:08: Received Zigbee message from 'ZLinky', type 'readResponse', cluster 'liXeePrivate', data '{"currentTarif":"      BASE\u0000     "}' from endpoint 1 with groupID 0
debug 2022-11-21 19:58:08: Received Zigbee message from 'ZLinky', type 'readResponse', cluster 'liXeePrivate', data '{"currentPrice":"      BASE\u0000     "}' from endpoint 1 with groupID 0
debug 2022-11-21 19:58:08: Received Zigbee message from 'ZLinky', type 'readResponse', cluster 'liXeePrivate', data '{"currentTarif":"      BASE\u0000     "}' from endpoint 1 with groupID 0
debug 2022-11-21 19:58:08: Received Zigbee message from 'ZLinky', type 'readResponse', cluster 'liXeePrivate', data '{"currentTarif":"      BASE\u0000     "}' from endpoint 1 with groupID 0
debug 2022-11-21 19:58:08: Received Zigbee message from 'ZLinky', type 'readResponse', cluster 'liXeePrivate', data '{"currentTarif":"      BASE\u0000     "}' from endpoint 1 with groupID 0
debug 2022-11-21 19:58:08: Received Zigbee message from 'ZLinky', type 'readResponse', cluster 'liXeePrivate', data '{"currentTarif":"      BASE\u0000     "}' from endpoint 1 with groupID 0
debug 2022-11-21 19:58:08: Received Zigbee message from 'ZLinky', type 'readResponse', cluster 'liXeePrivate', data '{"currentIndexTarif":1}' from endpoint 1 with groupID 0
debug 2022-11-21 19:58:08: Received Zigbee message from 'ZLinky', type 'readResponse', cluster 'liXeePrivate', data '{"currentTarif":"      BASE\u0000     "}' from endpoint 1 with groupID 0
debug 2022-11-21 19:58:09: Received Zigbee message from 'ZLinky', type 'readResponse', cluster 'liXeePrivate', data '{"currentTarif":"      BASE\u0000     "}' from endpoint 1 with groupID 0
debug 2022-11-21 19:58:09: Received Zigbee message from 'ZLinky', type 'readResponse', cluster 'liXeePrivate', data '{"currentTarif":"      BASE\u0000     "}' from endpoint 1 with groupID 0
debug 2022-11-21 19:58:09: Received Zigbee message from 'ZLinky', type 'readResponse', cluster 'liXeePrivate', data '{"currentTarif":"      BASE\u0000     "}' from endpoint 1 with groupID 0
debug 2022-11-21 19:58:09: Received Zigbee message from 'ZLinky', type 'readResponse', cluster 'liXeePrivate', data '{"currentPrice":"      BASE\u0000     "}' from endpoint 1 with groupID 0
debug 2022-11-21 19:58:09: Received Zigbee message from 'ZLinky', type 'readResponse', cluster 'liXeePrivate', data '{"currentTarif":"      BASE\u0000     "}' from endpoint 1 with groupID 0
debug 2022-11-21 19:58:09: Received Zigbee message from 'ZLinky', type 'readResponse', cluster 'liXeePrivate', data '{"currentTarif":"      BASE\u0000     "}' from endpoint 1 with groupID 0
debug 2022-11-21 19:58:09: Received Zigbee message from 'ZLinky', type 'readResponse', cluster 'liXeePrivate', data '{"currentTarif":"      BASE\u0000     "}' from endpoint 1 with groupID 0
debug 2022-11-21 19:58:09: Received Zigbee message from 'ZLinky', type 'readResponse', cluster 'liXeePrivate', data '{"currentTarif":"      BASE\u0000     "}' from endpoint 1 with groupID 0
debug 2022-11-21 19:58:09: Received Zigbee message from 'ZLinky', type 'readResponse', cluster 'liXeePrivate', data '{"currentPrice":"      BASE\u0000     "}' from endpoint 1 with groupID 0
debug 2022-11-21 19:58:09: Received Zigbee message from 'ZLinky', type 'readResponse', cluster 'liXeePrivate', data '{"currentTarif":"      BASE\u0000     "}' from endpoint 1 with groupID 0
debug 2022-11-21 19:58:09: Received Zigbee message from 'ZLinky', type 'readResponse', cluster 'liXeePrivate', data '{"currentTarif":"      BASE\u0000     "}' from endpoint 1 with groupID 0
debug 2022-11-21 19:58:09: Received Zigbee message from 'ZLinky', type 'readResponse', cluster 'liXeePrivate', data '{"currentTarif":"      BASE\u0000     "}' from endpoint 1 with groupID 0
debug 2022-11-21 19:58:09: Received Zigbee message from 'ZLinky', type 'readResponse', cluster 'liXeePrivate', data '{"currentPrice":"      BASE\u0000     "}' from endpoint 1 with groupID 0
debug 2022-11-21 19:58:10: Received Zigbee message from 'ZLinky', type 'readResponse', cluster 'liXeePrivate', data '{"currentTarif":"      BASE\u0000     "}' from endpoint 1 with groupID 0
debug 2022-11-21 19:58:10: Received Zigbee message from 'ZLinky', type 'readResponse', cluster 'liXeePrivate', data '{"currentTarif":"      BASE\u0000     "}' from endpoint 1 with groupID 0
debug 2022-11-21 19:58:10: Received Zigbee message from 'ZLinky', type 'readResponse', cluster 'liXeePrivate', data '{"currentTarif":"      BASE\u0000     "}' from endpoint 1 with groupID 0
debug 2022-11-21 19:58:10: Received Zigbee message from 'ZLinky', type 'readResponse', cluster 'liXeePrivate', data '{"currentDate":"H221121195754"}' from endpoint 1 with groupID 0
debug 2022-11-21 19:58:10: Received Zigbee message from 'ZLinky', type 'readResponse', cluster 'liXeePrivate', data '{"currentTarif":"      BASE\u0000     "}' from endpoint 1 with groupID 0
debug 2022-11-21 19:58:10: Received Zigbee message from 'ZLinky', type 'readResponse', cluster 'liXeePrivate', data '{"currentTarif":"      BASE\u0000     "}' from endpoint 1 with groupID 0
debug 2022-11-21 19:58:10: Received Zigbee message from 'ZLinky', type 'readResponse', cluster 'liXeePrivate', data '{"currentTarif":"      BASE\u0000     "}' from endpoint 1 with groupID 0
debug 2022-11-21 19:58:10: Received Zigbee message from 'ZLinky', type 'readResponse', cluster 'liXeePrivate', data '{"currentPrice":"      BASE\u0000     "}' from endpoint 1 with groupID 0
debug 2022-11-21 19:58:10: Received Zigbee message from 'ZLinky', type 'readResponse', cluster 'liXeePrivate', data '{"currentTarif":"      BASE\u0000     "}' from endpoint 1 with groupID 0
[... cut ...]
debug 2022-11-21 19:58:14: Received Zigbee message from 'ZLinky', type 'readResponse', cluster 'liXeePrivate', data '{"currentTarif":"      BASE\u0000     "}' from endpoint 1 with groupID 0
debug 2022-11-21 19:58:14: Received Zigbee message from 'ZLinky', type 'readResponse', cluster 'liXeePrivate', data '{"currentTarif":"      BASE\u0000     "}' from endpoint 1 with groupID 0
debug 2022-11-21 19:58:14: Received Zigbee message from 'ZLinky', type 'readResponse', cluster 'liXeePrivate', data '{"currentTarif":"      BASE\u0000     "}' from endpoint 1 with groupID 0
`

Z2M redevient utilisable, avec un CPU entre 0.2% et 2% dès que je débranche le ZLinky (v11 limited).

fairecasoimeme, 2022-11-22:
De ce que je vois, il y a un rapport sur le cluster liXeePrivate / currentTarif (à minima) qui est mal réglé.
je pense que vous devriez remettre à zéro le ZLinky en appuyant 10 sec sur le bouton, tout en étant en mode appairage. Cela remettra à zéro vos rapports.

fairecasoimeme, 2022-11-23:
Pouvez-vous copier ici le paramétrage des rapports et paramètres spécifiques ?

Fred

BRSS73, 2022-11-26:
Config :
'0x00158xxxxxxxxxxx':
  friendly_name: ZL
  measurement_poll_interval: -1
  linky_mode: standard
  energy_phase: single_phase
  production: 'false'
  tarif: Standard - BASE
  tic_command_whitelist: all
  measurement_poll_chunk: 0

Reporting : (un seul pour mon usage, j'enlève les autres)
  haElectricalMeasurement  apparentPower  60  900  1

antoinelibert, 2022-11-26:
Je recommande de mettre `SINSTS` dans `tic_command_whitelist` puis d'appuyer sur Reconfigure puis de faire une action "Apply" sur le reporting et enfin redémarrer Z2M.

BRSS73, 2022-11-26:
Pas mieux, malheureusement.
Quelques secondes après le branchement du ZLinky, flood de Z2M.

antoinelibert, 2022-11-26:
Et avec `measurement_poll_interval` à 120 ? (redémarrage de Z2M nécessaire)

mathieu-mp, 2022-11-26:
Idem ici: Z2M prend progressivement du retard dans l'injection des données sur le MQTT jusqu'à planter.
J'ai appliqué un debounce de 30s sauf pour apparent_power, current_summ_delivered, rms_current, warn_d_p_s
La charge CPU reste élevée, mais Z2M ne plante plus.