fairecasoimeme/Zlinky_TIC #137: MAJ v12 Limited OTA impossible par ZHA.

WarC0zes, 2022-11-23:
Bonjour @fairecasoimeme,
j'ai voulu mettre a jour le nouveau firmware v12 limited par OTA. Comme d'habitude je modifie mon configuration.yaml en ajoutant:
```
zha:
  zigpy_config:
    ota:
      otau_directory: /config/zigpy_ota
      ikea_provider: false
      ledvance_provider: false

logger:
  default: info
  logs:
    homeassistant.components.zha: debug
    zigpy: debug
```
Dans mon dossier /config/zigpy_ota/ j'ai le fichier ZLinky_router_v12_limited.ota

Je reboot, mais le démarrage de la mise a jour ce lance pas.
```
2022-11-23 07:58:04.140 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0xD98B](ZLinky_TIC): Device seen - marking the device available and resetting counter
2022-11-23 07:58:04.141 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0xD98B](ZLinky_TIC): Update device availability - device available: True - new availability: True - changed: False
```

En théorie je peut mettre a jour par OTA comme je suis en firmware v11 Limited.
Es ce que d'autre utilisateur de ZHA on le même problème ou j'ai mal fait quelque chose ?

fairecasoimeme, 2022-11-23:
Les requêtes OTA se font toutes les 2-4 heures. Pour déclencher rapidement la mise à jour, il faut débrancher et rebrancher l'appareil.
Fred

WarC0zes, 2022-11-23:
D'accord, je ferais ca. merci du tuyau.

WarC0zes, 2022-11-24:
Bonjour Fred,
La mise a jour c'est faite sans problème en débranchant et rebranchant le Zlinky, ca c'est lancer directe.
Je vais tester plusieurs jour, mais ca l'air bien plus stable chez moi. 
Je te ferais un retour.

Merci encore Fred.

closed 2022-11-24 by WarC0zes