fairecasoimeme/Zlinky_TIC #138: OTA toujours disponible après flash v12_limited

DamGranger, 2022-11-23:
Bonjour,
J'ai flashé avec un module CP2102 le firmware v12_limited (j'essaie de voir si j'aurai moins de déconnexion, car là il arrête de répondre toutes les 24h ce qui est frustrant). Je n'ai pas encore de retour d'expérience sur ce firmware à part celle-ci.
Je constate que Z2M continue de me proposer un update OTA pour le device. Si je clique dessus, je vois qu'il essaie d'installer la version router et fail avec un "Bad image format". Y a-t-il une manipulation à faire quelque part pour indiquer le type de firmware à utiliser dans Z2M afin que lors de la prochaine MAJ je ne sois pas obligé de la faire manuellement ?

DamGranger, 2022-11-24:
Bonjour, je ne pense pas qu'il s'agisse d'un problème d'alimentation pour plusieurs raisons :

- Je vous avais déjà renvoyé le module (fin décembre 2021) et vous l'aviez modifié, je crois avec un superviseur de tension (et une tropicalisation car il est situé dans une montée d'escalier ouverte il peut donc y faire froid), ça avait résolu les soucis de déconnexion d'ailleurs, il est étrange que ceux-ci réapparaissent.
- Le module est bien alimenté quand je vais voir dans le Linky (la LED clignote)
- la console dév me donne pour le moment un voltage correct et je vois peu de variation et, quand bien même, avec le superviseur il devrait en principe redémarrer : 
```
Info 2022-11-24 00:39:16Read result of 'genPowerCfg': {"mainsVoltage":3304}
Info 2022-11-24 00:52:55Read result of 'genPowerCfg': {"mainsVoltage":3307}
Info 2022-11-24 00:57:06Read result of 'genPowerCfg': {"mainsVoltage":3305}
Info 2022-11-24 01:03:14Read result of 'genPowerCfg': {"mainsVoltage":3307}
```

J'ai donc pensé à une saturation du module liée à mon réseau Zigbee (82 devices, dont 9 routeurs) ou un problème de distance.

Les routes ont l'air plutôt stables cela dit sachant que "PC COU R" se situe à l'étage en-dessous du ZLinky_TIC soit à environ 3-4m, après il y a environ 16-18m entre ces deux devices et la maison contenant l'ensemble du réseau mais pas grand chose pour mettre un routeur dans cet espace malheureusement. "PC CHA F" étant le second plus proche puisque situé sous une fenêtre de la maison à même hauteur que le ZLinky_TIC et à proximité du coordinateur (Sonoff Zigbee Dongle-P).
[image]
[image]

Pour l'instant avec le firmware limited, il est trop tôt pour dire s'il sera plus stable mais j'ai quand même pas mal de trous dans les remontées... peut-être des redémarrage liés à des chutes de tension TIC, je pense que je vais donc reflashé en firmware normal, pour à nouveau bénéficier de l'OTA, on verra ce que ça donne.

Je suis de très près ma consommation en ce moment car j'ai une PAC qui doit être installée dans les jour qui viennent et je cherche à savoir si un délesteur sera nécessaire ou pas (puisque Enedis refuse de dépasser 12kVA en monophasé).

fairecasoimeme, 2022-11-24:
Bonjour,
En effet, vu les données remontées, ce n'est probablement pas un souci d'énergie fournie par le Linky. Avec les détails que vous me fournissait, il est possible qu'avec la version limitée, il y est des soucis de portée et du coup, des trous dans les remontées. (D'après le schéma, le ZLinky passe uniquement par PC CHA F et PC COU R Pour atteindre le coordinateur (les chemins avec 2 LQI x/x)).
Dans votre cas, je pense qu'il est préférable d'être sur la version "Legacy",

DamGranger, 2022-11-24:
C'est parti pour reflash en legacy alors, je vais essayer de suivre si le problème de déconnexion se reproduit et voir si je peux contrôler régulièrement les voltages.

plouf34, 2022-11-25:
Bonjour,
Vous faites comment pour passer et repasser d'un firmware limited<>router sans USB TTL, le tout tournant sous Zigbee2mqtt HA?

DamGranger, 2022-11-26:
> Bonjour, Vous faites comment pour passer et repasser d'un firmware limited<>router sans USB TTL, le tout tournant sous Zigbee2mqtt HA?

Bonjour, je le fais avec un module USB to TTL CP2102.

Sinon mon problème de connectivité ne s'est pas amélioré avec le firmware legacy. Je n'ai qu'une route pour le module. 
Le routeur en question a un LQI de 3 vers le coordinateur et donc je n'ai que des routes à 3 ou 4 sauts de viables. Clairement il va falloir que je trouve une astuce pour étirer mon mesh vers cette zone.

Dans tous les cas, je marque cette issue solved puisque ça ne relève pas d'un problème avec le module, ni le firmware.

closed 2022-11-26 by DamGranger