fairecasoimeme/Zlinky_TIC #139: Couleur du lendemain non disponible (Z2M sur HA)

BBESINET, 2022-11-23:
Bonjour.

J'ai ouvert une issue là: [https://github.com/Koenkk/zigbee2mqtt/issues/14481](url) car j'ai le champ "couleur du lendemain" qui ne change jamais de la valeur "----".

J'ai ouvert une issue dans Z2M car j'ai l'impression que le problème est dans Z2M plus que dans le Zlinky, mais j'ai tout de même le doute.

Des possesseurs de tarif tempo sur d'autres plateformes que la mienne (HomeAssistant OS et Zlinky intégré par Zigbee2MQTT v1.28.2-1) arrivent ils à avoir la couleur du lendemain?

Si oui cela signifie bien que le problème se trouve au niveau du décodage de l'info fait par Z2M...

Je fais mes mises à jour de firmware régulièrement donc a priori pas de soucis de ce coté là.. 

Bien cordialement.

BBESINET

fairecasoimeme, 2022-11-23:
Bonjour, 
la couleur du lendemain, en mode historique, sous l'étiquette DEMAIN est disponible uniquement s'il y a une annonce de la part d'Enedis. S'il y a "----" c'est qu'il n'y a pas d'annonce de changement de couleur pour le lendemain.

Pour avoir plus d'infos comme la couleur du jour et celle du lendemain, il faut passer en mode standard et faire un contrôle du registre des statuts.

vous pouvez retrouver ces informations en cherchant le document sur Internet: Enedis-NOI-CPT_54E.pdf

Fred

BBESINET, 2022-11-23:
Bonjour Fred.

Merci pour les précisions.
Pour ma part je suis en mode historique, Et je retrouve bien ce que je savais de la téléinfo.

Concernant Zlinky:
- J'ai bien (et ce depuis l'installation du Zlinky) la PTEC qui s'affiche tout a fait correctement dans Home assistant.
` « HCJB » => Heures Creuses Jours Bleus,
 « HCJW » => Heures Creuses Jours Blancs (White),
 « HCJR » => Heures Creuses Jours Rouges,
 « HPJB » => Heures Pleines Jours Bleus,
 « HPJW » => Heures Pleines Jours Blancs (White),
 « HPJR » => Heures Pleines Jours Rouges.`

- Je m'attendais a avoir tous les soirs entre 20h et 6h la couleur du lendemain qui s'affiche. Mais la valeur reste à "----" tous les jours depuis l'installation du Zlinky il y a presque un mois. Le petit boitier "CPL" avec des diodes, ainsi que le Linky me remontent pourtant bien l'info que demain sera "BLEU" tous les soirs (Nous n'avons pas encore eu de jour blanc ni rouges depuis l'installation du Zlinky). Je m'attendais donc à avoir "----" entre 6h et 20h (couleur du lendemain inconnue avant 20h environ), puis "BLEU", BLAN" ou "ROUG" entre 20h et 6h en fonction de la couleur du lendemain.

Avec Zlinky, la valeur ne changerai que s'il y a un changement de tarif ?
Donc étant actuellement en jour bleu, on n'aurait une valeur à 20h que si celle ci est "BLAN" ou "ROUG" ?

D'avance merci pour ces précisions.

Bernard

BBESINET, 2022-11-23:
Par acquis de conscience je suis allé vérifier sur le compteur et en effet DEMAIN est indiqué ---- !

Il faudra vérifier si cela bascule au prochain jour blanc ou rouge...

Il n'y a peut être pas de soucis en fait...

fairecasoimeme, 2022-11-24:
Je ne pense pas qu'il y ait un souci, le ZLinky ne fait que retranscrire ce que le Linky donne comme info. (même si on est pas à l'abri d'un PB de décodage mais je ne pense pas)

BBESINET, 2022-11-24:
Merci du support.

Concernant le mode, je suis un peu coincé en mode historique car j'utilise par ailleurs depuis plus de 15 ans un boitier delta dore connecté à la TIC qui fait du délestage et la gestion des chauffages (fil pilotes) et du cumulus, y compris en fonction du tarif TEMPO. 
Vu l'âge de mon boitier antérieur au LINKY, ça m'étonnerai qu'il continue à marcher avec une TIC en mode standard...

Au passage le ZLINKY en parallèle de ce boitier sur la Téléinfo se comporte comme un charme...

Plus de news lors du prochain passage en jour blanc ou rouge (l'hiver arrive, ça ne devrait pas être long...)

fairecasoimeme, 2022-11-24:
Oui l'ancien boitier ne saura pas décoder le mode standard. Je crois que l'on peut mettre en parallèle jusqu'à 5 appareils sur la TIC.

BBESINET, 2022-11-28:
Bon c'est confirmé. Aujourd'hui la couleur du lendemain est bien BLAN...

Cette info remonte bien via Zlinky dans zigbee2mqtt. Pas de soucis. c'est juste l'info ENEDIS qui n'est pas dispo tous les jours, mais seulement lors des changement de couleur...

Je ferme l'issue. Merci à tous pour le support !

closed 2022-11-28 by BBESINET

BBESINET, 2022-12-02:
Je corrige un peu pour ceux qui viendraient avec la même question...

Apres un passage de plusieurs jour Blancs successifs, le comportement semble être le suivant (à confirmer pour les jours rouges...) :

Couleur du lendemain bleue       => zlinky renvoie "----" entre 22h et 6h
Couleur du lendemain blanche   => zlinky renvoie "BLAN" entre 22h et 6h
Couleur du lendemain rouge      => zlinky renvoie "ROUG" entre 22h et 6h
Couleur du lendemain inconnue => zlinky renvoie "----" entre 6h et 22h

BBESINET, 2022-12-09:
Suite au premier jour rouge hier... Je confirme le comportement du Linky ( et du Zlinky) suivant:
Couleur du lendemain bleue => zlinky renvoie "----" entre 22h et 6h
Couleur du lendemain blanche => zlinky renvoie "BLAN" entre 22h et 6h
Couleur du lendemain rouge => zlinky renvoie "ROUG" entre 22h et 6h
Couleur du lendemain inconnue => zlinky renvoie "----" entre 6h et 22h