fairecasoimeme/Zlinky_TIC #140: Cluster Power Configuration ajouté en v12.0 non déclaré ?

pdecat, 2022-11-23:
Bonjour @fairecasoimeme,

j'ai ouvert une PR https://github.com/zigpy/zha-device-handlers/pull/1962 pour ajouter le support du nouveau cluster Power Configuration ajouté en v12.0.

Et j'ai été surpris de constater que ce cluster n'est apparemment pas déclaré par le ZLinky_TIC : https://github.com/zigpy/zha-device-handlers/pull/1962/commits/f05d54651776347c28aa480495648c56ee7be2a4

C'est volontaire ?

WarC0zes, 2022-11-24:
Bonjour  @pdecat,
je viens de faire la maj en v12 limited et je vois bien le cluster dans la gestion du Zlinky.
[image]

fairecasoimeme, 2022-11-24:
> Et j'ai été surpris de constater que ce cluster n'est apparemment pas déclaré par le ZLinky_TIC : [zigpy/zha-device-handlers@f05d546](https://github.com/zigpy/zha-device-handlers/commit/f05d54651776347c28aa480495648c56ee7be2a4)
> 
> C'est volontaire ?

Non, le cluster est normalement "discoverable"

pdecat, 2022-11-24:
Hmm, j'ai eu des soucis également lors du passage de v11 à v12 en OTA, et j'ai dû ré-appairer également, mais je n'ai pas supprimé côté ZHA pour ne pas perdre l'historique. C'est peut-être à de ça.

pdecat, 2022-11-24:
> Bonjour @pdecat, je viens de faire la maj en v12 limited et je vois bien le cluster dans la gestion du Zlinky. [image]
> 
> Par contre j'ai supprimer le Zlinky et je les réappairer après la maj en v12.

Du coup, le quirk qui permet de requéter le cluster _manufacturer specific_ ne s'applique plus à ton ZLinky semble-t-il.

WarC0zes, 2022-11-24:
@pdecat  c'est exacte j'avais même pas remarquer, sa utilise plus le quirk.
[image]

pdecat, 2022-11-24:
Oui, car la signature change dans ton cas mais pas le mien 🤔

pdecat, 2022-11-24:
C'est bien un problème de cache côté ZHA.

pdecat, 2022-11-24:
@WarC0zes tu peux aussi voir si le quirk s'applique directement sur le device :

[image]

closed 2022-11-24 by pdecat

WarC0zes, 2022-11-24:
@pdecat
je savais mais comme sa utilise plus de quirk, je voulais te montrer mon tableau flex ou on voit les quirks utiliser ou pas.
[image]