fairecasoimeme/Zlinky_TIC #141: Ne fonctionne plus en V12

mducommun, 2022-11-24:
Bonsoir,
Depuis mon passage en V12, mon module ne voit plus aucun changement de valeurs coté linky...
Il est resté bloqué avec le même index de base et une conso PAPP de 510VA.

Je l'ai débranché/rebranché, reboot de HomeAssistant, reboot de ma zigstar rien n'y fait.

Je voulais procéder a un downgrade mais malgré le tuto, je ne trouve pas le fichier lixee.js.

Avez vous une idée?
Merci d'avance

closed 2022-11-24 by mducommun

mathieu-mp, 2022-11-26:
Bonjour @mducommun,

Je rencontre le même problème, comment l'as-tu résolu ?

Merci !

mducommun, 2022-11-26:
Bonsoir,
Il a fallu que je supprime le zlinky de z2m puis que je fasse l’appairage de nouveau et il a re fonctionner.

mathieu-mp, 2022-11-26:
Merci pour ce retour