fairecasoimeme/Zlinky_TIC #142: Power Cluster and Configure reporting

pipiche38, 2022-11-25:
It looks like v12 is bringing an additional cluster 0x0001 with 3 values reported. Could you please provide if those can be configure reported or not, as I didn't see them in the documentation

fairecasoimeme, 2022-11-25:
You can find this infos here : https://github.com/fairecasoimeme/Zlinky_TIC#powerconfiguration-cluster-0x0001----from-v12

RO= READ only - RP= Reportable - RW = Read/Write

pipiche38, 2022-11-25:
So 0x0001/0x0001 is not reportable , and is then requesting the host to poll the value, in addition ,  I do not see the Alarm Cluster is that something you plan to do ? As I don't understand the purpose of the ThreshodMin/Max without an alarm cluster. which then would be preventing the fact to poll such value

And from the Zigbee standard

```
The alarm settings in this table require the Alarms cluster to be implemented on the same device - see De-
pendencies```

closed 2022-11-25 by pipiche38