fairecasoimeme/Zlinky_TIC #143: Z2M convert VA / W

syl2042, 2022-11-28:
Hi,
How convert PAPP VA data in Watts data with Z2M? 
I tried fex things, but without success. 
Thanks per advance.

pipiche38, 2022-11-28:
VA and Watts are 2 differents things. Not sure what you want to do.
https://en.wikipedia.org/wiki/Power_factor

Now under certain assumption you can assume that W = VA, but if for instance you have induction this will have big deviation.

syl2042, 2022-11-29:
Thx for your answer. 
I saw that. But i saw too that some people have this (eg : https://forum.hacf.fr/t/cle-zlinky-lixee-avec-zha-conbee2-remonter-dans-energie-hp-hc/9058)

But i don't undestand why we don't have from the linky the instant consumption power in watts. 

In the Energy Table, we have the consumption by hours, in KWH. 

Why don't have this data for the instant moment? 

If the linky could give the KWH consumption for one hour, it have so the consumption for each moment. 

May be we could extract a spécific data from this way? (from the hourly consumption in kwh)

Thanks per advance.

DamGranger, 2022-11-30:
kWh can't be instant. It's the energy calculated by doign the integral of effective power during a certain time (hence the 'h' at the end). Linky measures the apparent power in VA which is a basic operation: current times voltage. If you want the effective power (in W, which is apparent power times power factor) you'd need to calculate the power factor which can go from 0 to 1 depending on equipments on the line. For the sake of a simple home with a single phase, you can assume this is very close to 1, so 1VA=1W (ish).