fairecasoimeme/Zlinky_TIC #145: PAPP et IINST ne remontent plus

DamGranger, 2022-11-30:
Bonjour,

j’ai régulièrement ce genre de souci, le module ne remonte plus les valeurs IINST et PAPP alors que les autres valeurs remontent bien. 

[image]

Je reçois régulièrement des payload du module avec des valeurs variant pour les index, le linkquality, ainsi que pour la période HC/HP en cours (active tier registered). 

La solution est souvent de supprimer et réappairer mais je me demande s’il n’y aurait pas une solution plus pérenne et/ou un moyen plus simple de « relancer la machine »

Pour info la configuration semble s’être effectuée correctement
[image]

[image]

DamGranger, 2022-11-30:
En première réponse : en reappliquant le reporting sur apparentpower ça ne change rien, mais en supprimant et recréant le-dit reporting j’ai un update…

DamGranger, 2022-11-30:
Second test, sans reporting sur apparentPower je remonte bien des update, beaucoup d'ailleurs (trop ?). Dois-je supprimer les 3 reportings qui ont été configurés automatiquement par l'inclusion dans Z2M ? ou bien est-ce un bug et puis-je fournir des éléments permettant de le résoudre ?

BrunoSpy, 2022-12-31:
Idem ici depuis la V12
Je suis en mode standard et j'ai le même souci : SINSTS ne se met pas à jour alors que toutes les autres valeurs remontent correctement.

Dernière version de Z2M.

BrunoSpy, 2023-01-06:
Merci pour l'astuce : la suppression du reporting dans Z2M corrige le problème mais cela revient à chaque redémarrage de HA. Le souci est arrivé depuis le passage de V5 à V12, sans changement de version de Z2M. Je suis passé en 1.29.0 pour Z2M, mais pas d'amélioration. Je ne m'y connais pas assez pour comprendre ce qu'il se passe.

VonOx, 2023-02-26:
Salut, V13 idem plus de SINSTS

VonOx, 2023-11-13:
Salut , il y'a du nouveau sur ce sujet ?