fairecasoimeme/Zlinky_TIC #146: constant fast blinking after OTA

toxic0berliner, 2022-12-01:
Hello, not sure if french or english is better ==> je peux redemander en français si besoin

A few days ago I upgraded my lixee ZLinky TIC with the OTA upgrade that Z2M proposed to me (v12). After that, it stopped reporting the kWh I consumed, I played a little with the reporting option, pushed the precision to 3, the min report value to 0.000001, report intervals to 60 seconds... It seemed not to work, so I wanted to downgrade to v11.
I updated my z2m config file : 
<details><summary>configuration.yaml</summary>

```
homeassistant: true
mqtt:
  server: mqtt://core-mosquitto:1883
  user: addons
  password: XXX
serial:
  port: >-
    /dev/serial/by-id/usb-ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2_20220816084617-if00
  adapter: ezsp
frontend:
  port: 8099
advanced:
  homeassistant_legacy_entity_attributes: false
  legacy_api: false
  legacy_availability_payload: false
  log_level: debug
  log_syslog:
    app_name: Zigbee2MQTT
    eol: /n
    host: localhost
    localhost: localhost
    path: /dev/log
    pid: process.pid
    port: 123
    protocol: tcp4
    type: '5424'
  last_seen: ISO_8601
  pan_id: 18277
  channel: 15
  network_key:
    - X
    - X
    - X
    - X
    - X
    - X
    - X
    - X
    - X
    - X
    - X
    - X
    - X
    - X
    - X
    - X
device_options:
  legacy: false
availability:
  active:
    timeout: 4
  passive:
    timeout: 1500
ota:
  zigbee_ota_override_index_location: index-custom-ota.json
devices:
  <some devices>
groups:
  <some groups>

```
</details>

Added the `/config/zigbee2mqtt/index-custom-ota.json` file : 
<details><summary>/config/zigbee2mqtt/index-custom-ota.json</summary>

```
[
    {
        "fileVersion": 12,
        "fileSize": 245710,
        "manufacturerCode": 4151,
        "imageType": 1,
        "sha512": "5d7e0a20141b78b85b4b046e623bc2bba24b28563464fe70227e79d0acdd5c0bde2adbd9d2557bd6cdfef2036d964c35c9e1746a8f1356af3325dd96f7a80e56",
        "url": "https://github.com/fairecasoimeme/Zlinky_TIC/releases/download/v12.0/ZLinky_router_v12.ota"
    },
    {
        "fileVersion": 11,
        "fileSize": 244142,
        "manufacturerCode": 4151,
        "imageType": 1,
        "sha512": "d7996ef7495b7c98cf6adb49530db29f3210c0e097f938da433f13e5b13446e72f0ece2205b126ee8525a900f1fa0e574fdf4aff07a9013945ea216d83fafc0d",
        "url": "https://github.com/fairecasoimeme/Zlinky_TIC/releases/download/v11.0/ZLinky_router_v11.ota",
        "force": true
    }
]

```
</details>

Rebooted, clicked the search OTA update button, and I believe it's at that time that the Lixee went offline. I tried a few things then, always recieving an error that the lixee was offline or timed out to any command. I then went down in the garage and saw the lixee was blinking very fast. 

I tries a lot of things since then : 
- unplug/replug the lixee
- push and hold the pair button
- forcibly removed the device from z2M
- all of the above, while haveing pairing enabled in z2m
Pushing the pair button stops the blinking for a bit, but as soon as released it starts blinking fast again.

Seems the lixee is "dead". I have ordered this https://www.amazon.fr/gp/product/B00LODGRV8/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1 
It should arrive on monday and I hope it will let me flash a fresh v10 or v11 firmware and solve my issue...

But if you have some other way meanwhile to "reset" and get it working, that would be nice !

fairecasoimeme, 2022-12-01:
Hi,
If the ZLinky is blinking fast, It's because the device is left the network and it is in pairing mode. With Z2M, after an update, you have to left the network, reboot Z2M and re pair the zlinky.

Fred

toxic0berliner, 2022-12-01:
wow, you are correct, sorry for the hassle !
Turns out enabling paring on all devices was a bad idea. Enabling it only on the nearby router worked better but interview failed, I had to enable paring on the coordinator, that was far, so it failed a few times but now it seems ok.
I do believe the network will heal itself and lixee will soon have it's messages relayed by the router nearby and possibly other devices in my garage, let's see.
Closing this then

Thanks a lot Fred !

closed 2022-12-01 by toxic0berliner