fairecasoimeme/Zlinky_TIC #147: Plus aucune remontée avec la V12.

neotwix, 2022-12-02:
Bonjour,
Depuis que j'ai fait l'OTA vers la v12. je n'ai plus aucune remonté qui ce fait entre le module et Z2MqTT.
Mon champs base reste bloqué a sa derniere valeur et ne bouge plus depuis.
J'ai redemarrer un peu tout le monde (Home Assistant et Z2MQTT) le module (enlevé remis du Linly. 
mais rien n'y fait je n'ai plus de remontée.

rhcpdu13, 2022-12-02:
Bonjour,
Je me joins à toi, j'ai eu le même problème, voir pire. J'ai tenté tout ce que tu as dit, mais en plus : 
Suppression forcée sur z2mqtt, tentative de ré appairage, mais il me sort tout le temps en non supporté. 
J'ai tenté aussi un reset du lixee (en mode appairage, appuyer pendant 10 sec sur le bouton...) rien n'y fait. Dans les logs j'ai uniquement ça : Failed to interview '0x00158d0005d2a937', device has not successfully been paired

@neotwix as tu pu t'en sortir ?

ellmout, 2022-12-02:
Bonjour,
Idem même problème depuis hier, mais après passage en v11, dans Z2m j'ai le message d'erreur suivant :

`Zigbee2MQTT:error 2022-12-02 22:05:39: Failed to configure 'Zlinky', attempt 1 (Error: Bind 0x00158d0005d284c1/1 seMetering from '0x00212effff075e84/1' failed (waiting for response TIMEOUT)
    at DeconzAdapter.bind (/app/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:768:19)
    at Endpoint.bind (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:567:13)
    at Object.bind (/app/node_modules/zigbee-herdsman-converters/lib/reporting.js:40:9)
    at Object.configure (/app/node_modules/zigbee-herdsman-converters/devices/lixee.js:598:9)
    at Configure.configure (/app/lib/extension/configure.ts:115:13)
    at Configure.start (/app/lib/extension/configure.ts:68:13)
    at Controller.callExtensions (/app/lib/controller.ts:316:17)
    at Controller.start (/app/lib/controller.ts:153:9)
    at start (/app/index.js:107:5))`

Si je vais au compteur, que je débranche et rebranche le module, il recommunique mais juste un fois et plus de nouvelle.

WarC0zes, 2022-12-02:
Bonjour, quand vous flasher en v12 supprimer et ré-appairer le Zlinky. Je vais dire mieux quand vous flasher une mise a jour, vaut mieux supprimer et ré-appairer le Zlinky. J'ai eu le même soucis en v12 Limited sous ZHA et tout fonctionne au top ( je dirais que même plusieurs personnes ont eu ce soucis).  
Penser aussi a la distance entre la clef zigbee et le Zlinky ca joue beau, rapprocher le de la clef pour l'appairer.

https://forum.hacf.fr/t/zlinky-lixee-ota-4000-0012-zigbee2mqtt-probleme/17783

chryss44, 2022-12-03:
Bonjour, je confirme les galère de la V11 à la V12, il m'a fallut faire plusieurs suppressions dans Z2M et plusieurs débranchements retranchements sans compter un redémarrage de Z2M.
Tout est reparti mais c'était effectivement un peu lourd.

klemsgithub, 2022-12-03:
Bonjour, 
Idem, je suis passé de la V5 à la V12 la semaine dernière et j'ai perdu la remontée de certaines infos. J'ai du forcer la suppression dans zigbee2mqtt puis ré-appairer (avec difficulté, au bout de 4 ou 5 essais) pour un retour à la normal. Depuis ce matin j'ai de nouveau perdu la connexion (j'ai le lixee depuis plusieurs mois et je n'avais jamais eu un pépin jusque ici...). Je verrai demain pour tenter de ré-appairer (il commence à faire nuit, le compteur est à 10m de la maison). J'ai l'impression que la qualité du signal est moins bonne qu'avant (et je ne peux pas déplacer mon serveur qui se trouve physiquement dans une baie 19" dans la maison, je passe par un prise philips hue branchée au plus proche du linky dans la maison pour faire relais...) Est-il envisageable que le firmware joue sur la qualité du signal ?  ou alors c'est juste une coïncidence lié a la météo un peu fraîche?

vingerha, 2022-12-04:
J'avais plusieurs soucis moi même V11>V12, reappairage Z2M très difficile et j'avais rallongé entre Linky et Zlinly pour retourner au normal...mon compteur est dehors, à 8m d'un repeteur zigbee.

DamGranger, 2022-12-04:
Quand vous avez le problème, vous n'avez plus rien qui remonte ou c'est comme dans #145 ?

neotwix, 2022-12-05:
Rien de rien.
J'ai suivi votre manip. (remove de Z2MQTT et reset du module) ca amarché

pabaillon, 2022-12-09:
Plus de remonté depuis cette nuit également en V11 et Z4D..

klemsgithub, 2022-12-09:
Pour en revenir à la déconnexion que j'évoquais en début de semaine, je n'ai rien eu besoin de faire pour que ça remarche. 
En effet j'ai maintenant des coupures de temps en temps qui durent de quelques minutes à 30/40 minutes puis ça revient tout seul. Encore une fois j'ignore si il y a un lien avec le firmware V12 :/.

rhcpdu13, 2022-12-09:
Hello,
Je confirme que, j'ai pu tout refaire fonctionner correctement, et faire un appairage sans problème en déplaçant le serveur à proximité du linky. Depuis, aucun problème, même après rebascule du serveur à son emplacement. Je sais que j'ai aussi un bon maillage autour.. ça peut expliquer aussi les problèmes que vous avez.. en tout cas plus de soucis pour ma part en v12 :)

ER503, 2022-12-12:
Bonjour,

J'ai aussi des problèmes depuis le passage en V12, effectué il y a deux jours.
Mon ZLinky est hors ligne depuis que j'ai fait cette mise-à-jour.

J'ai beau essayer de le supprimer de Z2M et de tenter un réappairage, il apparait toujours comme "Non supporté", et je ne sais plus quoi faire...

fairecasoimeme, 2022-12-13:
il faut aussi redémarrer Z2M

bdherouville, 2022-12-20:
Bonjour,

j'ai galéré à réappairer après l'upgrade. Mon modèle est celui avec antenne externe. J'ai remplacé l'antenne fournie par une antenne 2.4GHz plus grande et tout est rentré dans l'ordre. Mais j'ai mis du temps.

trollix, 2022-12-20:
Moi, après plusieurs semaines sans problèmes. après un passage à la V12, ça déconnecte toutes les 24h... Grrrrr

ericsche, 2022-12-29:
Pareil depuis le passage en v12 j'ai bcp de soucis de déconnections que je n'avais pas avant v10.

na-ji, 2023-01-06:
Salut ! 
Même problème de mon côté, il arrive souvent que le Zlinky_TIC s'arrête de mettre à jour les valeurs. Aucune erreur côté zigbee2mqtt, c'est comme si l'appareil renvoyait constamment les mêmes valeurs figées. La seule solution que j'ai trouvée pour l'instant, c'est débrancher et rebrancher l'appareil du compteur Linky. C'est assez gênant, vu que ça fausse toutes les stats dans HA. 
Est-il possible de downgrade le firmware?

Bazarov42, 2023-01-06:
Je n'arrive même plus à le connecter à z2m, il n'est plus reconnu...

billOu46, 2023-01-12:
Hello, même souci pour moi. J'ai pas fait attention lors de la maj 12, et je me rends compte que j'ai plus de stats depuis plusieurs jours.
Débrancher/rebrancher le Zlinky a pas suffit.
Supprimer du réseau et réappairer.
En effet, il a été capricieux au réappairage. Il a fallu que j'insiste.
J'attends que les infos soient a nouveau visible dans les stats, mais il a bien pris en compte l'index actuel depuis son réappairage, donc ça devrait le faire.

Serait-il possible de pouvoir faire des maj sans pour autant nécessiter de réappairer le Zlinky ?

bdherouville, 2023-01-12:
J'étais dans une configuration très contraignante avec un linky loin du premier routeur de mon installation, je ne parle même pas du coordinateur.
La première chose que j'ai faite est d'updater le Zlink_TIC à côté du coordinateur en l'alimentant avec du 5V entre A et L2.
Ensuite j'avais la version avec antenne externe. J'ai mis une antenne 2.4GHz plus grande, récupérée sur un ancien routeur wifi (en 2.4GHz aussi)
Enfin, j'ai mis en extérieur un module pour contôler une prise de courant. Ce module fait routeur donc il améliore la connectivité avec le coordinateur.
[module](https://fr.aliexpress.com/item/1005003637851837.html?spm=ae-sycm.sycm.0.0.654b4dce12OQ7a&spm=a2g0o.store_pc_home.singleImageText_2001462102935.0&gatewayAdapt=glo2fra)

fairecasoimeme, 2023-01-12:
> Serait-il possible de pouvoir faire des maj sans pour autant nécessiter de réappairer le Zlinky ?

@billOu46  Malheureusement, le redémarrage n'est pas dû au ZLinky mais de comment il est géré sur Z2M. Je ne sais pas pourquoi, quand il y a mise à jour, Z2M envoie pleins de requêtes au ZLinky sans arrêt et met le système sous l'eau.

fairecasoimeme, 2023-01-12:
Bin si, sauf que cela nécessite souvent un redémarrage de Z2M après la mise à jour.

WarC0zes, 2023-01-12:
Bonjour,
je suis passé de ZHA a Z2M et aucun soucis d'apparaige du Zlinky avec la v12 Limited.
@fairecasoimeme depuis la sortie de la v12,  j'ai augmenter mon réseau a 35 appareils (contre 17 a la sortie de la v12) et mon Zlinky est super stable ( plus de timeout ) 👍 
pour moi c'est le gold firmware :)

vingerha, 2023-01-12:
C'est clair qu'il y a probablement tant des gens qui n'ont rien de soucis... ceux qui l'ont sont dans le forum ou ici sur github. C'est juste intéressant que pour ceux qui ont les soucis c'est presque pareil: soucis d'appairer. Chez moi j; en suis presque sûre qu'il y a un lien avec Z2M car avec chaque appairage Zlinky, mqtt devient 'fou'

Bazarov42, 2023-01-12:
Depuis que je l'ai supprimer pour essayer de l'apareiller de nouveau, je ne le vois plus du tout, même tout à côté...

ylemoigne, 2023-01-13:
Je n'ai pas de soucis pour appareiller, jamais. Le dongle zigbee est a 3m du zlinky. Par contre, j'ai le soucis HASS(ZHA) choppe des données au démarrage, puis plus rien ne bouge.

Dites moi comment je peux aider a résoudre ce soucis d'update (fourniture de log, manip, test, etc).

fairecasoimeme, 2023-01-13:
> Je n'ai pas de soucis pour appareiller, jamais. Le dongle zigbee est a 3m du zlinky. Par contre, j'ai le soucis HASS(ZHA) choppe des données au démarrage, puis plus rien ne bouge. (firmware V12 limited)
> 
> Dites moi comment je peux aider a résoudre ce soucis d'update (fourniture de log, manip, test, etc).

Pour ZHA, il y a eu une mise à jour récente de HA qui devrait régler le problème. https://www.home-assistant.io/blog/2023/01/04/release-20231/#release-202312---january-8

https://github.com/home-assistant/core/pull/85355

Pofilo, 2023-01-15:
Bonjour,
Je suis passé hier de la V11 à la V12 (depuis Z2M 1.29.2).
Ce matin, je m'aperçois que je n'ai plus de valeur, BASE n'est plus incrémenté.
J'ai réussi à priori à corriger le problème avec cette manip:
+ _(Rebrancher le ZLinky sur le compteur Linky -> à priori, ça n'a rien fait)_
+ Suppression du ZLinky dans Z2M (cf capture)
+ Appui 10 secondes sur le bouton reset du ZLinky
+ Arrêt de Z2M
+ Remise des anciens paramètres (reset par la suppression dans Z2M) (voir + bas mes paramètres en question)
+ Redémarrage de Z2M
+ Et voilà!

```
'0xyyyyyyyyyyyyyyyy':
    friendly_name: Linky
    measurement_poll_interval: 360 
    linky_mode: historique
    energy_phase: single_phase
    production: 'false'
    kWh_precision: 3
    measurement_poll_chunk: 1
    legacy: false
![Capture](https://user-images.githubusercontent.com/7527594/212538347-4116298e-ec4b-4c8b-ab54-357ad23d3904.PNG)

    optimistic: true
```

Si chacun ayant eu ce problème peut décrire sa procédure de fix, peut-être qu'on découvrira qqchose :)

KipK, 2023-01-16:
J'ai voulu tester la v12, j'étais en V9 depuis Septembre,
Autant je n'avais aucun soiucis avec la V9 en mode no routeur ( sur Deconz/Conbee ) , autant avec la v12,j'ai essuyé 3 deconnections en 4 jours avec obligation de debrancher/rebrancher le module.

Repassage en v9 ( no router )  et tout refonctionne correct.
Je n'ai pas encore testé de flasher le firmware no route v12 , il faut que je retrouve mon FTDI avant .

ylemoigne, 2023-01-16:
> Pour ZHA, il y a eu une mise à jour récente de HA qui devrait régler le problème. https://www.home-assistant.io/blog/2023/01/04/release-20231/#release-202312---january-8
> 
> [home-assistant/core#85355](https://github.com/home-assistant/core/pull/85355)

Toujours pas. J'ai fais maj HASS, supprimé le zlinky des devices, reboot hass, reset 10s zlinky, réassocié, reboot hass.

Home Assistant 2023.1.4 
Supervisor 2022.12.1
Operating System 9.4
Interface utilisateur : 20230110.0 - latest 

J'ai un abonnement Base.

[image]

Pas d'autre config particulière.

A noter, si j'interroge manuellement la valeur la, elle se met bien a jour :
[image]

Comme pour mon message précédent : Dites moi comment je peux aider a résoudre ce soucis d'update (fourniture de log, manip, test, etc).

WarC0zes, 2023-01-16:
@ylemoigne 
c'est toujours pas fixer ce problème, la dernière maj de ZHA change rien.
le PR pour fix, faut patienter encore https://github.com/home-assistant/core/pull/82602

ylemoigne, 2023-01-30:
@WarC0zes j'ai du mal a suivre pour le coup dans quel version ce sera disponible.

WarC0zes, 2023-01-31:
> @WarC0zes j'ai du mal a suivre pour le coup dans quel version ce sera disponible.

C'est dispo dans la version core 2023.2.0 qui sort demain (01/02/2023) ou maintenant dans la beta.

pipiche38, 2023-02-10:
je pense que si vous avez un coordinateur autre qu'une Zigate, il y a une grande probabilité de perte du ZLinky lors d'une mise-à-jour de ZHA et notamment des couches zigpy

ylemoigne, 2023-02-12:
De mon coté c'est bon depuis la 2023.2.0 comme l'avait annoncé @WarC0zes (que je remercie pour le suivi ❤️ )

Bazarov42, 2023-02-12:
C'est tombé en marche tout seul et la c'est  de nouveau pété ^^
Pas l'impression que ça soit une perte de connexion le Zlinky est toujours reconnu et présent mais les remontées d'info sont sporadiques

fairecasoimeme, 2023-02-13:
si les remontées sont sporadiques, c'est surement qu'il y a des pertes de trames. Avez-vous des logs ? quel est le LQI ?

fairecasoimeme, 2023-02-13:
Pouvez-vous générer une carte du réseau avec les LQI ?

Bazarov42, 2023-02-13:
C'est de nouveau en rade.
Le schéma a été plus long à faire que d'habitude 
[image]

au niveau des logs le zlinky à disparu, j'ai essayé de le reconfigurer voici ce que disent les logs ; 

`error 2023-02-13 14:59:40Failed to configure 'Linky', attempt 1 (Error: Bind 0x00158d0005d2a8ad/1 seMetering from '0x00212effff07f7e2/1' failed (waiting for response TIMEOUT) at DeconzAdapter.bind (/app/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:768:19) at Endpoint.bind (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:567:13) at Object.bind (/app/node_modules/zigbee-herdsman-converters/lib/reporting.js:40:9) at Object.configure (/app/node_modules/zigbee-herdsman-converters/devices/lixee.js:598:9) at Configure.configure (/app/lib/extension/configure.ts:117:13) at Configure.onMQTTMessage (/app/lib/extension/configure.ts:55:21))
info 2023-02-13 14:59:40MQTT publish: topic 'zigbee2mqtt/bridge/response/device/configure', payload '{"data":{"id":"Linky"},"error":"Failed to configure (Bind 0x00158d0005d2a8ad/1 seMetering from '0x00212effff07f7e2/1' failed (waiting for response TIMEOUT))","status":"error","transaction":"kuval-7"}'`

fairecasoimeme, 2023-02-13:
Les logs montrent que le coordinateur ne reçoit pas la réponse du Bind (timeout). Sur le schéma, on voit que le lien entre le ZLinky et le routeur a une qualité (LQI) de 1.

Bazarov42, 2023-02-13:
J'ai refais un schéma alors que je n'ai toujours pas de remontée en temps réelle 

[image]

[image]

j'ai une machine qui tourne, ça devrait pas être à 2 A  ^^

ericsche, 2023-02-14:
> @Bazarov42<https://emea01.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2FBazarov42&data=05%7C01%7C%7Cdbe52d7792454fa2085808db0eb4b329%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C638119941479979172%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C3000%7C%7C%7C&sdata=eSZ4oeRBHyuUxd%2Fhzu4iIkq6j0eipsPVa%2FvoMcbEvpg%3D&reserved=0>
> C'est bizarre, votre Zlinky a pas de lien directe au coordinateur. C'est quoi le routeur cuisine comme matériel ?
>
> Chez moi il y a un lien directe au coordinateur ( trait droit ) et les liens vers d'autres routeurs ( trait pointiller )
> [Zlinky z2m]<https://emea01.safelinks.protection.outlook.com/?url=https%3A%2F%2Fuser-images.githubusercontent.com%2F91471397%2F218818566-17452199-5363-4f06-b662-064e16ef5c13.png&data=05%7C01%7C%7Cdbe52d7792454fa2085808db0eb4b329%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C638119941480135412%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C3000%7C%7C%7C&sdata=eFYGgA5MXldN3wq1ps7mj1YF9OYGHgAr2I5MD7UYA2s%3D&reserved=0>

Oui je pense que le maillage est un peu faible.
D'expérience 3 lien c'est quand même mieux.
Pour ma part je n'ai pas de lien direct mais le linky a 5 voisins.

Envoyé à partir de Outlook pour Android<https://aka.ms/AAb9ysg>

Bazarov42, 2023-02-14:
Je suis trop loin pour qu'il se connecte au coordinateur en direct, c'est pour ça que j'ai pris un petit routeur : https://www.amazon.fr/gp/product/B08438TKYG/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1

Je n'ai pas d'autres appareils sur lequel le Zlinky peut se connecter.
Si je ne mets pas se routeur/répéteur la l’interrupteur de la cuisine se connecte directement au coordinateur sans problème mais le Zlinky n'y arrive pas.

WarC0zes, 2023-02-15:
Faut rajouter d'autre routeur, avec le lqi a 65 ( vu l'ancien Screenshot a 1, ca doit yoyo le lqi ), je pense que c'est pas assez pour qu'il soit stable.

ericsche, 2023-02-15:
Prends plutôt des prises qu'un répéteur au moins cela t'apporte une fonction domotique plutôt que de brancher un élément juste pour faire routeur.
Perso j'ai cela

Envoyé à partir de Outlook pour Android<https://aka.ms/AAb9ysg>

ericsche, 2023-02-15:
Perso j'ai deux trois  https://amzn.eu/d/gOo6jf4
Comme cela tu peux piloter une prise et mesurer la conso.
Marche très bien derrière mon sèche linge 😊

Envoyé à partir de Outlook pour Android<https://aka.ms/AAb9ysg>

vingerha, 2023-02-15:
J'ai mon linky et donc zlinky à l'extérieur et ajouté un prise extérieur Lidl qui maintenant couvre aussi mon gazpar

ericsche, 2023-02-16:
Pour info même soucis avec la v13 et zigbee2mqtt j'ai  lancé la maj il y a 24h et depuis plus rien 😒 je suis bon pour un reset je pense. C'est dommage car on perd l'historique à chaque fois.

ericsche, 2023-02-22:
> Perte de l'historique: c'est que quand ton sensor est viré de HA.
> Option 1....pas beau, est d'installer une sensor a côté et utiliser l'ancien +neuf
> Option 2: ... pas facile, copier les données dans la bdd, il faut être a l'aise dans SQL

Merci Arjan !

Au final en lisant des blog l'autre soir j'ai trouvé une option 3 qui est d'utiliser un helper de type utility métier : https://www.home-assistant.io/integrations/utility_meter/
Du coups si tu change de fournisseur ou tu dois reset le zlinky tu ne perds pas l'historique.

Du coups je vais tester cela 😊

Envoyé à partir de Outlook pour Android<https://aka.ms/AAb9ysg>

vingerha, 2023-02-23:
Ç'est comme option 1, une sensor à côté

mycanaletto, 2023-02-26:
Comment expliquer que depuis cette MAJ on perde le le contact, sinon par le code de cette mise à jour puisqu'aucune autre modification n'a eu lieu et que tout fonctionnait à merveille depuis plus d'un an ?

Y a t'il possibilité de repasser sur une vieille version ?

fairecasoimeme, 2023-02-26:
> Comment expliquer que depuis cette MAJ on perde le le contact, sinon par le code de cette mise à jour puisqu'aucune autre modification n'a eu lieu et que tout fonctionnait à merveille depuis plus d'un an ?
> 
> Y a t'il possibilité de repasser sur une vieille version ?

Ça n'a rien à voir avec le firmware mais avec le processus de mise à jour OTA. Il arrive souvent qu'après une mise à jour OTA, selon le coordinateur ou le plugin, il faut RAZ le ZLinky et refaire un appairage.

Pour faire un downgrade, il faut un module USB TTL et suivre la procédure de mise à jour Non OTA : https://github.com/fairecasoimeme/Zlinky_TIC#non-ota

Bazarov42, 2023-03-07:
> 

J'ai trouvé une utilité à mettre une prise zigbee et en effet ça fonctionne beaucoup mieux. (merci du coup de main)
Par contre je n'arrive plus à mettre les relevés dans le tableau énergie ^^

KipK, 2023-03-07:
> Ça n'a rien à voir avec le firmware mais avec le processus de mise à jour OTA. Il arrive souvent qu'après une mise à jour OTA, selon le coordinateur ou le plugin, il faut RAZ le ZLinky et refaire un [...]

Pour le downgrade vous pouvez aussi éditer le numéro de version dans le fichier OTA avec le client Deconz ( onglet Otau ) et mettre une version supérieure. l'OTA va se faire, et le firmware installé aura qd meme la bonne version après installation. Du coup bien penser à virer le fichier ensuite ou ça va refaire la mise à jours en boucle ( vu qu'il verra une version supérieure )

Bazarov42, 2023-03-07:
> J'ai trouvé une utilité à mettre une prise zigbee et en effet ça fonctionne beaucoup mieux. (merci du coup de main) Par contre je n'arrive plus à mettre les relevés dans le tableau énergie ^^

J'ai parlé trop vite, ça a fonctionné un temps et la c'est re-denouveau hs, je n'arrive même plus à l'appairer.
Je vais attendre des fois ça revient tout seul

cynnfx, 2023-07-30:
Hello first and thank for all work accomplished. idk if this is the right place but you guys seems to have a lixee version with antenna. I'm with the 1.4 version without antenna and even a connector for it. If somekind of hero can send a picture of the pcb and the antenna connection it would be wonderful 👍 i actually try as you probably guess to add an antenna. Thanks everyone!