fairecasoimeme/Zlinky_TIC #151: Paramétrage non trouvé pour mon abonnement: Standard "HC SEM WE LUNDI"

trollix, 2022-12-13:
Bonjour,
Home assistant avec zigbee2mqtt dernière version.

J'ai le lixee avec la dernière version du firmare installée. Tout à l'air de fonctionner normalement sauf mon abonnement qui apparemment n'est pas pris en compte.
Mon Linky est en standard, mais mon abonnement qui est spécial: 

[image]

C'est HP/HC semaine et HC le WE + le lundi. J'ai pas trouvé la bonne option dans le paramétrage du module 👍 

[image]

Dans home assistant , du coup, ça doublonne en semaine:

[image]

Alors que ça marche bien le week-end...

[image]

Merci de votre aide. Je pense que les paramétrage possibles ne correspondent pas avec mon abonnement

su2626, 2022-12-14:
Même abonnement et même problème que toi. J’imagine que tous les abonnements Zen WE + lundi ou vendredi ont les mêmes soucis.  

Ça serait top d’avoir une solution

trollix, 2022-12-15:
De quel plugin vous parlez. Sur Home Assistant y'a pas de plugin

fairecasoimeme, 2022-12-15:
D'après vos screenshot vous utilisez Z2M. Pour que le ZLinky soit compatible avec Z2M, un composant ou plugin a été développé.
https://gist.github.com/ralmn/02de46e193bf111062f15be07d3e9bd2?permalink_comment_id=4018011

puis a été intégré directement dans Z2M : https://github.com/Koenkk/zigbee-herdsman-converters/pull/3493

Sur ce github, personne ne pourra vous aider sur votre problème car votre problème n'a aucun lien avec le matériel ou le firmware. Vos informations de tarifs ne concerne que le traitement des données donc du côté du plugin