fairecasoimeme/Zlinky_TIC #152: Abonnement Zen

su2626, 2022-12-15:
Bonjour à tous, 

J'ai quelques problèmes depuis mon changement de fournisseur. Je suis passé chez EDF sur une offre Zen HC Weekend + Lundi et depuis je ne parviens pas à récupérer les bonnes valeurs sur mon Lixee. 

Je suis sur Zigbee2mqtt. Voici la conf du lixee : 

[image]

Voici les données que j'obtiens : 
[image]

Le journal : 
MQTT publish: topic 'zigbee2mqtt/Linky', payload '{"active_enerfy_out_d01":0,"active_enerfy_out_d02":0,"active_power":-32768,"active_power_max":-32768,"active_power_ph_b":-32768,"active_register_tier_delivered":"TH..","apparent_power":190,"available_power":45,"average_rms_voltage_meas_period":0,"current_date":"","current_index_tarif":0,"current_price":"","current_summ_delivered":19133.943,"current_tarif":"BASE","current_tier1_summ_delivered":19133.943,"current_tier2_summ_delivered":0,"current_tier3_summ_delivered":0,"current_tier4_summ_delivered":0,"current_tier5_summ_delivered":0,"current_tier6_summ_delivered":0,"drawn_v_a_max_n1":0,"linkquality":126,"message1":"","message2":"","meter_serial_number":"022061736895","power_threshold":null,"relais":0,"rms_current":0,"rms_current_max":90,"rms_voltage":65535,"schedule_h_p_h_c":0,"site_id":"","software_revision":null,"start_notice_e_j_p":0,"status_register":"","tomorrow_color":"","update":{"installed_version":"4000-0012","state":"idle"},"update_available":false}'

Si quelqu'un a des pistes ? :/

fairecasoimeme, 2022-12-15:
Votre linky est en BASE pas en Zen HC Weekend + Lundi

su2626, 2022-12-15:
Pourtant sur le Linky je vois bien "HC Semaine", "HP Semaine", "HC WE", "HP WE", "HC Lundi" et "HP Lundi".

Je dois me tourner vers EDF du coup?

su2626, 2022-12-15:
Très bien, je vais donc les contacter et je ferai un retour ici au besoin. 

Merci pour votre réactivité en tout cas. 

J'en profite juste pour une seconde question, va-t-on disposer du choix "Standard - Sem WE Lundi" et "Standard - Sem WE Vendredi" à l'avenir ?

fairecasoimeme, 2022-12-15:
> J'en profite juste pour une seconde question, va-t-on disposer du choix "Standard - Sem WE Lundi" et "Standard - Sem WE Vendredi" à l'avenir ?

je ne sais pas, il faut poser la question aux développeurs du plugin

SilentT-FR, 2023-07-20:
@su2626 il existe depuis 3 mois maintenant
https://github.com/Koenkk/zigbee-herdsman-converters/blame/master/src/devices/lixee.js#L323

fairecasoimeme, 2023-07-20:
merci pour le retour, je clôture.

closed 2023-07-20 by fairecasoimeme