fairecasoimeme/Zlinky_TIC #153: Presque pas de métriques

kimizz59, 2022-12-16:
Bonjour,

J'ai connecté un ZLinky_TIC à mon HA via Z2M mais je n'ai jamais réussi à avoir des informations intéressantes. 

Version du firmware : 4000-0012
Zigbee2MQTT version: 1.28.4-1 
Mode linky : Standard (j'étais en histo et suis passé en standard en espérant un changement, hélas non)

J'ai uniquement ce genre d'info
[image]
[image]

Je n'ai que ce genre de log actuellement : 
`Zigbee2MQTT:info  2022-12-16 23:18:39: MQTT publish: topic 'zigbee2mqtt/Lixee', payload '{"active_enerfy_out_d01":0,"active_power":-32768,"active_power_max":-32768,"active_power_ph_b":-32768,"active_register_tier_delivered":"","apparent_power":65535,"available_power":null,"average_rms_voltage_meas_period":0,"current_date":"","current_index_tarif":0,"current_price":"","current_summ_delivered":0,"current_tarif":"BASE","current_tier1_summ_delivered":0,"current_tier2_summ_delivered":0,"drawn_v_a_max_n1":0,"linkquality":164,"message1":"","message2":"","meter_serial_number":"","power_threshold":null,"relais":0,"rms_current":65535,"rms_current_max":65535,"rms_voltage":65535,"site_id":"","software_revision":null,"status_register":"","update":{"installed_version":"4000-0012","state":null},"update_available":null,"warn_d_p_s":0}'`

J'ai tenté de passer les paramètres spécifique en manuels, j'ai tenté de delete le lixee pour le réappairé à multiple reprises mais je n'ai jamais plus que ça.

Assez nouveau sur HA et Z2M, je ne vois pas trop quoi tester de plus et où je pourrais choper des infos intéressantes pour résoudre mon souci.

josephpage, 2022-12-17:
Avez-vous essayé de jouer avec les différents paramètres de l'onglet "Paramètres (spécifiques)" dans Zigbee2Mqtt ? Si j'étais dans votre situation c'est ce que je ferais.

Mais c'est vrai que le fait que même le Serial Number ne remonte pas est très mauvais signe.
Si vous ne trouvez pas de solution avec différents réglages et après plusieurs ré-appairage, il vous reste à demander de l'aide au fabricant : https://lixee.fr/nous-contacter

kimizz59, 2022-12-17:
Yes j'ai essayé un peu de jouer un peu avec les options.
Je ne maitrise pas les métrique "poll" en revanche, avec quelles valeurs je pourrait faire des essais à ce niveau là ?
[image]

Si avec ça je n'arrive toujours rien, je vais contacter le fabricant directement dans ce cas.

CladStriffe, 2022-12-18:
Bonjour,

Deja on voit une bizarrerie : les valeurs dans l'onglet "expose" sont celles du mode historique. 
Essayez dans l'onglet "paramètres spécifiques" de mettre le mode à "standard". 
Puis single phase et false si vous n'avez qu'une phase et pas de production.

CladStriffe, 2022-12-18:
Les valeurs dans "expose" sont celles du mode historique. 
C'est toujours ça que vous avez même après avoir changé manuellement dans paramètres spécifiques sur "standard" ?

kimizz59, 2022-12-18:
Yes, ma config est celle sur le screen juste au dessus. J'ai également testé en production "false".
J'ai vérifié, je suis bien en mode standard sur la TIC chez Total, elle était en histo au début, lors de mes premières mises en place.
Je pense qu'un réappairage suffit lors d'un changement de mode de TIC (je ne compte même plus le nombre de réappairages que j'ai fait) mais je vais vérifier s'il n'y a pas de manipulation particulières dans ce genre de cas.

Il est possible qu'au début j'ai vu une maj disponible et l'ai lancé. Après pas mal de recherche je vois que cela pourrait poser des soucis (je ne sais pas si c'est la cause du miens).

Mardi je testerais cette procédure (je n'ai pas un accès physique constant à mon installation)
[image]
source : https://www.domo-blog.fr/test-lixee-teleinformation-zigbee-avec-home-assistant-jeedom/

r0m1CH, 2023-03-10:
Bonjour,
J'ai un souci similaire il me semble.
Je suis passé en standard récemment (y'a un ou deux jours) et depuis je n'ai plus les valeurs "standards"
UMOY1, URMS1... mais j'ai pleins d'autres trucs.

J'ai tenté de réappareiller, de redémarrer Z2M... rien n'y fait.

Je suis en tempo mais je sais pas ce qu'il me manque.

[image]

[image]

Une idée ?

r0m1CH, 2023-03-10:
Bon je viens de relancer Z2M après avoir forcé les paramètres Standard + 1 phase + pas de production.
J'ai des infos en plus mais pas les bonnes:

[image]

Je sais vraiment pas ce que je fais de mal.

fairecasoimeme, 2023-03-13:
Quel est le LQI du ZLinky ?

r0m1CH, 2023-03-13:
Il oscille entre 127 et 180.

fairecasoimeme, 2023-03-13:
Pouvez-vous générer un schéma réseau et partager ?

r0m1CH, 2023-03-13:
C'est un peu brouillon mais ça donne ça:
[image]

r0m1CH, 2023-03-16:
Une idée de ce qui me manque ?

r0m1CH, 2023-03-16:
Bon je crois que j’ai trouvé mon souci.

Mon Linky était repassé en mode historique sans raison particulière (peut être le changement de provider)
Il a fallu donc rebasculer en mode standard puis:

Reconfiguration de la clé Lixee dans Zigbee2Mqtt
Redémarrage de Zigbee2Mqtt
Forcer le mode standard dans la conf Zigbee2Mqtt (le mode auto n’a semble-t-il pas fonctionné)
Redémarrage de Zigbee2Mqtt
Au bout de 4/5 minutes après le rédémarrage j’ai retrouvé mes valeurs habituelles (Tension, intensité…).

Je pense qu’il est possible de reconfigurer la clé + changer le mode standard en une action mais je vous décris ce que j’ai fait si jamais ça peut en aider un.

Bon courage !

kimizz59, 2024-11-14:
Je vois que cette issue est toujours open
De mon côté j'avais fait les MAJ OTA (je ne sais pas si c'est lié)
J'ai fait des tests avec le propriétaire du dongle, celui-ci a été changé et cela fonctionne maintenant correctement.
Je ne lance plus les MAJ OTA depuis, je ne sais pas si c'était lié

closed 2024-11-14 by kimizz59