fairecasoimeme/Zlinky_TIC #154: Graph statistique home assistant entité non répertoriée

xseth, 2022-12-17:
Bonjour,

Je souhaite faire des graph statistique avec Home assistant un peu comme ici : https://forum.hacf.fr/t/cle-zlinky-lixee-avec-zha-conbee2-remonter-dans-energie-hp-hc/9058
Cependant j'ai uniquement l'entité BASE qui remonte lors que je souhaite faire un graph statistique dans home assistant. 
Je n'ai pas l'entité PAPP et IINST par exemple... 

Home assistant indique d'ailleurs que si c'est le cas de faire suivre le lien : 
https://www.home-assistant.io/more-info/statistics/

Ou bien je dois faire peut être quelque chose de ma config que j'ignore pour débloquer cette possibilité ?

ericsche, 2022-12-29:
Oui je suis comme toi je cherche aussi à utiliser certaines variables et ce n'est pas possible.

misterbenj34, 2023-01-03:
Hello @xseth 
Dans mon cas, sur ZHA, PAPP correspond à "Apparent Power" et IINST à "RMS current".
Ce sont les capteurs remontés par défaut, sans bidouille de ma part.
ZHA + HA + ZLinky en v12

xseth, 2023-01-04:
@misterbenj34 oui, j'ai également ces entités PAPP, IINST... mais je ne peux pas utiliser ces entités pour les graph de statistique… Je peux uniquement faire des graph classique. 
Pour réaliser des graph statistique, j'ai uniquement l'entité PAPP qui est disponible dans ZHA

1300371, 2023-09-26:
C'est dû à l'attribut manquant: `state_class: measurement` pour l'entité `sensor...._papp` 😞