fairecasoimeme/Zlinky_TIC #156: Reporting config Z2M

PEGARifrouf, 2022-12-20:
Bonjour à vous,
J'étais entrain d'essayer de modifier le reporting du Zlinky comme on peut voir dans la capture d'écran :
[image]
Mon objectif est de ne remonter que les infos que j'utilise à une fréquence adaptée (il m'arrive d'avoir des déconnexion).

Seulement avec ça je n'ai plus de remontée d'info sous Home Assistant, les valeurs ne se mettent à jour que si j'interroge le ZLinky sur les données dans les cluster correspondant à mes infos dans l'onglet "Dev console".

Je sais que ça concerne plutôt Z2M, mais je cherche à voir si je peux augmenter la stabilité du Zlinky comme ça.

Quelqu'un a t'il une idée sur une erreur de config que j'aurais faite ?

Z2M à jour v1.28.4
ZLiny V12 (récemment de v5 à V12 limited)

antoinelibert, 2022-12-20:
Bonjour,

Vous avez essayé de simplement relancer Z2M et débrancher/rebrancher le ZLinky ?

PEGARifrouf, 2022-12-20:
Oui j'ai relancé Z2M et rebranché, dans cet ordre.

PEGARifrouf, 2022-12-20:
Pour l'instant je l'ai supprimé pour ajouté de nouveau à Z2M donc les paramètres par défaut sont revenus.
Si quelqu'un a des infos si ça peut effectivement alléger la conso électrique ou pas du tout je suis preneur.

Merci

KipK, 2022-12-28:
Salut

J'avais rencontré un soucis équivalent, cela venait de valeurs min/max qui étaient en dehors des clous et du coup ignorait les reporting.

PEGARifrouf, 2022-12-30:
Depuis que j'ai ré apairé le Zlinky je n'ai plus de coupure. Je ne sais pas si il y a des variations de qualité de l'alimentation du module sur la prise et du compteur et que ce serait donc passager mais je laisse comme ça.

Mais bon à savoir pour les valeurs hors plage.

closed 2022-12-30 by PEGARifrouf