fairecasoimeme/Zlinky_TIC #159: Non remontée index BBRHCJR

githubegins, 2023-01-01:
Je suis sous JEEDOM: 4.3.12 avec plugin Zigbee officiel: 2022-10-20 01:03:41
J'ai un compteur Linky avec un abonnement monophasé, 18kW en formule Tempo.

L'index BBRHCJR (1::1794::264) est remonté une fois au premier appairage, il y a environ 1 an (je ne me rappelle pas quelle version de firmware avait mon Z-Linky). Mais ne se met jamais à jour depuis.
J'ai réussi une fois à le faire se mettre à jour dans des conditions que je ne peux décrire précisément (mais aprés une mise à jour de firmware). Malgré tout depuis il ne se met toujours pas à jour. 
Je viens de la passer sur le firmware V12 (ZLinky_router_v12.bin). Malgré un ré-appairage, l'index BBRHCJR ne remonte toujour pas. 

Tous les autres index (y compris BBRHPJR, 1::1794::266) se mettent bien à jour régulièrement. Il n'y a vraiment que BBRHCJR qui ne remonte pas.

Phil2222, 2023-01-01:
Bonjour, 
J'ai de mon côté depuis une restauration de sauvegarde jeedom en fin de journée une non remontée de l'index BBRHCJR.
Comme ça me parait très proche de ce constat, je me permets de rebondir. 
Avez-vous trouvé votre solution ? 
De mon côté, tous les index remontent parfaitement sur mon lixee (qui est a priori en v5) mais impossible de remonter le BBRHCJR... 
JE n'ai pas en mémoire l'apparition de cette erreur avant la restauration jeedom, mais comme il n'y a que cet index qui ne remonte pas, je me demande si je ne loupe pas une astuce...
Le lixee communique avec une clé conbee2 connectée sur une freebox delta avec jeedom et le plugin zigbee... Jusqu'ici tout fonctionnait plutôt bien...

Merci d'avance de vos pistes...

Phil2222, 2023-01-01:
Bon et bien en creusant dans le détail du point dans jeedom, l'info brute remontait bien mais n'apparaissait pas dans jeedom. Une ébauche de solution a été de recréer une commande avec la même logical ID que l'index qui n'apparaissait plus et tout est revenu comme par magie...
Sujet clos de mon côté...

closed 2023-01-02 by fairecasoimeme

githubegins, 2023-01-02:
Effectivement. Le fait d'avoir recréer un nouvelle commande avec la même logic_id semble résoudre le problème.
A noter que ce n'est pas la nouvelle commande qui se met correctement à jour. Mais c'est bien l'ancienne commande qui s'actualise alors qu'elle ne s'actualisait plus.