fairecasoimeme/Zlinky_TIC #161: Registered delivered tier enum

piitaya, 2023-01-02:
La commande `PTEC` (cluster `0x0702` attribut `0x0020`) utilise un type `string` alors que la spécification de la ZigBee attend une `enum`.
Cela empêche une intégration avec zigpy (et donc ZHA dans Home Assistant). Je pense qu'il y aurait la possibilité de faire un quirk (https://github.com/zigpy/zha-device-handlers#what-the-heck-is-a-quirk) pour assurer la comptabilité avec ZHA mais si le firmware suit les specifications ZigBee, la compatibilité avec la plupart des systèmes sera facilitée.

[image]

[image]

[image]

Home Assistant PR pour l'ajout des tiers : https://github.com/home-assistant/core/pull/82602
Source : https://zigbeealliance.org/wp-content/uploads/2019/11/docs-07-5356-19-0zse-zigbee-smart-energy-profile-specification.pdf

fairecasoimeme, 2023-01-11:
Bonjour,

En effet, il y a une erreur et dans ce cas précis le data type ne correspond pas.
Cependant, cela va être compliqué de changer car il faudrait modifier tous les plugins / box qui utilisent cet attribut.
Dans tous les cas, pour le moment, je vous recommande d'ignorer cet attribut pour ZHA et passer par un quirks.

Je vais quand même essayer de réfléchir à une solution pour rendre compatible avec la norme.

Merci pour votre contribution

piitaya, 2023-01-11:
Peut-être pourrait-on dupliquer cette valeur dans le cluster Lixee Private (`0xFF66`) pour le laisser disponible avec ce format et utiliser la norme dans le cluster (`0x0702`). Mais il faudrait quand même que les plugins/box se mettent à jour pour utiliser l'autre cluster.

aauzi, 2024-08-18:
Bonjour,

Je viens de m'équiper d'un ZLinky_TIC, en début de semaine, et je suis très heureux de son intégration actuelle dans ZHA.

Je profite de cette occasion pour vous remercier de votre travail.

Néanmoins, je me permet de remonter ce ticket sur la pile pour apporter mon point de vue.

J'ai noté que le cluster Lixee Private (0xFF66) est conçu pour regrouper les "doublons" qu'il est susceptible d'y avoir entre les modes Historique et Standard.

Par exemple, l'étiquette OPTARIF du mode Standard et très justement regroupée avec l'étiquette NGTF du mode historique (attribut: 0x0000). Et cela relève du bon sens car ces deux étiquettes sont excluses suuivant le mode et reflètent toutes les deux le Contrat en cours.

De ce fait, la même information fonctionnelle se trouve dans le même attribut, quel que soit le mode de fonctionnement.

Si l'on suit cette logique, je verrais bien l'étiquette PTEC du mode Historique (actuellement attribut 0x0020 du cluster 0x0702) regroupée avec l'étiquette LTARF du mode Standard (attribut 0x0200 du cluster 0xFF66). car on aurait un seul attribut qui porterait l'information textuelle de tarif en cours.

Cette seule modif, qui je pense n'introduit pas de rupture de compatibilité, aurait l'avantage de rendre la lecture de la tarification courante commune aux mode Standard et Historique et permetterait d'implémenter un sensor la remontant de manière cohérente avec un hypothétique sensor remontant l'Option de tarification.

Les intégrations exploitant l'étiquette PTEC du cluster 0x0702 auraient tout intérêt à miger sur ce nouvel attribut car elles y gagneraient la disponibilité de l'information du mode Standard.

Et cela ferait probablement mieux passer la pillule de la mise en conformité avec la norme :)

Pour cet aspect, il me semble que l'étiquette NTARF du mode Standard est particulièrement propice à migrer vers l'énuméré de l'attribut ActiveRegisterTier de la norme.

En mode historique toutefois, les intégrations aimeraient probablement bénéficier du même service sur le même attribut, il me semble que le firmware pourrait effectuer le transcodage qu'il reflète déjà dans la mise à disposition des différents tiers d'énergie. 

Soit implémenter quelque chose qui ressemble à:

|Mode         |Etiquette|Valeur|ActiveRegisterTier|
|---------------|-------------|----------|--------------------------|
|Standard  |NTARF     |(n)        | (n)                           |
|Historique|PTEC       |TH..     |1|
|Historique|PTEC       |HC..     |1|
|Historique|PTEC       |HP..     |2|
|Historique|PTEC       |HN..    |1|
|Historique|PTEC       |PM..    |2|
|Historique|PTEC       |HCJB   |1|
|Historique|PTEC       |HPJB   |2|
|Historique|PTEC       |HCJW  |3|
|Historique|PTEC       |HPJW  |4|
|Historique|PTEC       |HCJR   |5|
|Historique|PTEC       |HPJR   |6|

source: [Enedis-NOI-CPT_02E, Version 6, 15/02/2017]  chapitre: 2.3.3. Liste des groupes d'information, § Description des informations (p 27/109)
source: https://github.com/fairecasoimeme/Zlinky_TIC?tab=readme-ov-file#synth%C3%A8se-d%C3%A9veloppeur

Note: j'ai implémenté cela grâce à l'excellent zha-toolkit assorti  de quelques templates.
Dans les limites de mon contrat (HC/HP), ça répond à mon besoin (d'autant mieux que je suis immédiatement passé en mode standard).

[sensors.yaml.txt](https://github.com/user-attachments/files/16650506/sensors.yaml.txt)
[automations.yaml.txt](https://github.com/user-attachments/files/16650509/automations.yaml.txt)

L'alternative, serait de carrément supprimer cet attribut du cluster 0x0702, après mise en commun avec LTARF , et de conserver NTARF comme il est dans le cluster 0xFF66. Charge à l'intégration d'implémenter le transcodage ci-dessus.

En conclusion, il apparait que la gestion des options tarifaires est un domaine spécifique (par rapport à la norme ZIgbee) qui a semble-t-il conduit à de nombreuses évolutions de protocole TIC dans la passé et qui en produira probablement d'autres.
C'est un service de valeur que rend le ZLinky_TIC en en fournissant une abstraction compatible de la norme et en conservant l'essentiel des aspects "exotiques" dans le cluster "Manufacturer".
C'est cool et  toute évolution allant dans ce sens, offrant en plus une réduction de payload, doit être saluée.

André

PS: désolé d'avoir été si long et merci encore :)

aauzi, 2025-01-04:
Parce que, parfois, la traduction en code vaut mieux que de longs discours.
Et parce que je note par ailleurs les efforts de mise au niveau du quirk ZHA permettant l'implémentation de nouveaux sensors.

Voici une maquette traduisant ce que je disais plus haut.

Le diff est basé sur:

```bash
[aauzi@vivopc Zlinky_TIC (master *)]$ git log | head -n 5
commit fc3160f546d6ebabcf6b28b3b5fd134cd17c0f05
Author: Fairesoimeme <akila@faire-ca-soi-meme.fr>
Date:   Mon Oct 21 15:51:54 2024 +0200

    Update README.md
```

<details><summary>PoC</summary>

```diff
diff --git a/ZLinky/Source/App_Linky.c b/ZLinky/Source/App_Linky.c
index 99f5cf5..2fe60f3 100644
--- a/ZLinky/Source/App_Linky.c
+++ b/ZLinky/Source/App_Linky.c
@@ -269,11 +269,19 @@ uint8 APP_vProcessRxDataStandard ( void )
 				{
 					DBG_vPrintf(1, "\r\nLTARF : %s",trim(au8Value));
 					memcpy(sBaseDevice.sLinkyServerCluster.au8LinkyLTARF, au8Value,16);
+#ifndef NOT_PTEC_ON_LTARF
+					sBaseDevice.sLinkyServerCluster.sLinkyLTARF.u8Length=16;
+#endif /* NOT_PTEC_ON_LTARF */
 
 				}else if(memcmp(au8Command,"NTARF",5)==0)
 				{
 					DBG_vPrintf(1, "\r\nNTARF : %s",trim(au8Value));
 					sBaseDevice.sLinkyServerCluster.au8LinkyNTARF = atoi(au8Value);
+#ifndef NOT_NTARF_ON_REGISTER_TIER_DELIVERED
+					sBaseDevice.sSimpleMeteringServerCluster.sActiveRegisterTierDelivered.pu8Data[0] =
+					    '0' + sBaseDevice.sLinkyServerCluster.au8LinkyNTARF;
+					sBaseDevice.sSimpleMeteringServerCluster.sActiveRegisterTierDelivered.u8Length=1;
+#endif /* NOT_NTARF_ON_REGISTER_TIER_DELIVERED */
 
 				}else if(memcmp(au8Command,"VTIC",4)==0)
 				{
@@ -986,8 +994,48 @@ uint8 APP_vProcessRxDataHisto ( void )
 				}else if(memcmp(au8Command,"PTEC",4)==0)
 				{
 					DBG_vPrintf(1, "\r\nPTEC : %s",au8Value);
+#ifndef NOT_PTEC_ON_LTARF
+					memcpy(sBaseDevice.sLinkyServerCluster.au8LinkyLTARF, au8Value, 4);
+					sBaseDevice.sLinkyServerCluster.sLinkyLTARF.u8Length=4;
+#endif /* NOT_PTEC_ON_LTARF */
+#ifndef NOT_PTEC_ON_NTARF
+					{
+					    static const struct {
+						uint8 au8ValuePTEC[5];
+						uint8 au8ValueNTARF;
+					    } tCvtPTEC2NTARF[] = {
+						{"TH..", 1},
+						{"HC..", 1},
+						{"HP..", 2},
+						{"HN..", 1},
+						{"PM..", 2},
+						{"HCJB", 1},
+						{"HPJB", 2},
+						{"HCJW", 3},
+						{"HPJW", 4},
+						{"HCJR", 5},
+						{"HPJR", 6},
+					    };
+					    int index;
+					    for(index=0 ; index < sizeof(tCvtPTEC2NTARF)/sizeof(tCvtPTEC2NTARF[0]) ; index++)
+					    {
+						if(memcmp(au8Value, tCvtPTEC2NTARF[index].au8ValuePTEC, 4)==0)
+						{
+						    sBaseDevice.sLinkyServerCluster.au8LinkyNTARF = tCvtPTEC2NTARF[index].au8ValueNTARF;
+#ifndef NOT_NTARF_ON_REGISTER_TIER_DELIVERED
+						    sBaseDevice.sSimpleMeteringServerCluster.sActiveRegisterTierDelivered.pu8Data[0] =
+							'0' + sBaseDevice.sLinkyServerCluster.au8LinkyNTARF;
+						    sBaseDevice.sSimpleMeteringServerCluster.sActiveRegisterTierDelivered.u8Length=1;
+#endif /* NOT_NTARF_ON_REGISTER_TIER_DELIVERED */
+						    break;
+						}
+					    }
+					}
+#endif /* NOT_PTEC_ON_NTARF */
+#ifdef PTEC_ON_REGISTER_TIER_DELIVERED
 					memcpy(sBaseDevice.sSimpleMeteringServerCluster.sActiveRegisterTierDelivered.pu8Data,au8Value,4);
 					sBaseDevice.sSimpleMeteringServerCluster.sActiveRegisterTierDelivered.u8Length=4;
+#endif /* PTEC_ON_REGISTER_TIER_DELIVERED */
 				}else if(memcmp(au8Command,"ADPS",4)==0)
 				{
 					DBG_vPrintf(1, "\r\nADPS: %s",au8Value);
```
</details>

Malheureusement, je n'ai pas d'environnement de développement croisé à ma disposition et je n'ai donc pas pu le compiler ni surtout en vérifier le bon fonctionnement :/  .

Et j'ai conservé la définition (erronée vis à vis de la norme) de l'attribut `ActiveRegisterTierDelivered` mais cette proposition montre comment il pourrait effectivement devenir  l'énuméré que la norme définit (si j'ai bien tout compris).

fairecasoimeme, 2025-01-06:
Bonjour,
et merci pour la contribution.
Dans la prochaine version, je gère la tarification dans le cluster 0xFF66 attribut 0x0010
Il aura pour spécificité d'être compatible avec les rapports et d'être transmit automatiquement à chaque changement de valeur.
Je prends note de vos changements dans le code pour pouvoir l'intégrer dans la prochaine version